pcengines/coreboot #80: bios update error on apu2c4

hdatma, 2017-11-30:
bios: http://pcengines.ch/file/apu2_v4.0.11.rom.tar.gz

```
root@box:/media/SYSLINUX# flashrom -w apu2_v4.0.11.rom -p internal
flashrom v0.9.8-r1888 on Linux 3.16.6-tinycore (i686)
flashrom is free software, get the source code at http://www.flashrom.org

Calibrating delay loop... delay loop is unreliable, trying to continue OK.
coreboot table found at 0xdffae000.
Found chipset "AMD FCH".
Enabling flash write... OK.
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI) mapped at physical address 0xff800000.
This coreboot image (PC Engines:PC Engines apu2) does not appear to
be correct for the detected mainboard (PC Engines:PCEngines apu2).
Aborting. You can override this with -p internal:boardmismatch=force.
```

pietrushnic, 2017-11-30:
@hdatma this is known issue. Unfortunately legacy BIOS with which boards were shipped contain different string in SMBIOS/DMI tables. Unfortunately there is not much we can do with that. You can force flashing as commend suggest. We also have [modified flashrom](https://github.com/pcengines/flashrom/tree/apu_support) which supress this warining by setting `boardmismatch=force` by default.

hdatma, 2017-12-01:
Looks good.

```
PC Engines apu2
coreboot build 20170724
BIOS version v4.0.11
4080 MB ECC DRAM

SeaBIOS (version rel-1.10.2.1)

Press F10 key now for boot menu

Select boot device:

1. USB MSC Drive VendorCo ProductCode 2.00
2. SD card SD04G 3796MiB
3. Payload [memtest]
4. Payload [setup]
```

hdatma, 2017-12-01:
There are two versions on site*: 4.0.11 and 4.6.0. Which one is best?

http://pcengines.ch/howto.htm#bios

pietrushnic, 2017-12-01:
@hdatma second round and all further will be good since SMBIOS/DMI table is not correct. Please take a look at [Binary releases section in README](https://github.com/pcengines/apu2-documentation#binary-releases). There is link to changelog, based on which you can decide which one fits your needs. If you like bleeding edge I would recommend 4.6.x. We will abandon improvements in 4.0.x soon.

hdatma, 2017-12-01:
I like stable and reliable.

Strykar, 2017-12-11:
@pietrushnic I just tried 4.6.0 on my APU2 and the BIOS no longer has the option to enable/disable UART C and D. Why was this removed?

pietrushnic, 2017-12-11:
@Strykar this option was never available in mainline coreboot, but was part of legacy version. Runtime configuration was added in recent release [4.6.3](https://github.com/pcengines/coreboot/blob/coreboot-4.6.x/CHANGELOG.md#v463---2017-10-30). `README.md` contain links only to firmware binaries verified by PC Engines. If you want to try different version you can build it from source. If you need that option now and don't want to build firmware by your own please try recent legacy version which is in [this section](https://github.com/pcengines/apu2-documentation#legacy)

pietrushnic, 2018-01-12:
@hdatma @Strykar If there are no other problems related I would like to close this issue.

hdatma, 2018-01-13:
Sure, go ahead! Thanks!

closed 2018-01-13 by hdatma

piu130, 2018-05-28:
@pietrushnic is 4.6.x now stable? Or still bleeding edge?

pietrushnic, 2018-05-28:
@piu130 it depends what you consider stable. There is no such thing like stable coreboot release - I was surprised also, but it happens that coreboot releases are arbitrary points in time. There are no extensive quality checks behind that. We try to do validation on our own but it is not an easy thing - we plan to publish our automated tests results soon, so people can see what are the things that were confirmed. Of course, we are not able to validate all configurations and have to rely on the community.

I also don't know what you mean by "bleeding edge". `coreboot-4.6.x` is quite far from upstream master branch, so for sure this is not the latest greatest coreboot version.

Anyone who tries to flash the firmware on apuX should have a very good reason behind that, if there is no reason then please do not update. If you want the update please make sure you have a fallback mechanism like [spi1a](http://pcengines.ch/spi1a.htm). We are not responsible for any damage caused by firmware upgrades.