pcengines/coreboot #91: Slow boot due to UDMA disabled

alexpro, 2018-01-11:
My English is not very good, excuse me if I said something wrong.
The initial boot stage in APU boadrs is extremely slow. This is due to disabled UDMA in SeaBios. I tried to add CONFIG_ATA_DMA and CONFIG_ATA_PIO32 options to seabios .config and got the load time of FreeBSD kernel four times faster. FreeBSD kernel and modules loads approx. 40 sec. without above parameters. With them load time is 9 sec. in my test case. Why these options are disabled in APU firmware? This is much better when the boot process is faster.

pietrushnic, 2018-01-11:
@alexpro I don't know if this option will interfere with anything, but we can enable UDMA in next release. We building SeaBIOS with default options, so far there was no requests about boot time.

We will let you know when new firmware will be released what should be end of January or beginning of February.

pietrushnic, 2018-01-11:
@miczyg1 please use:

```
CONFIG_ATA_DMA=y
CONFIG_ATA_PIO32=y
```

For SeaBIOS in next release.

miczyg1, 2018-03-16:
@alexpro UDMA has been enabled in SeaBIOS in v4.6.6 release and in v4.0.15 legacy release. We are sorry for late response.

closed 2018-03-20 by miczyg1