pcengines/coreboot #92: 4.0.14, APU2 no output to console

alexpro, 2018-01-12:
I tried to build firmware 4.0.14 from 4.0.x branch. All seems to work fine, but no output to console from SeaBios. Input works as usual, I can press F10 key, and boot from the desired source "in the blind". In more details, after BIOS version and RAM amount messages no output to console until OS begins to start.
You removed sgabios a week ago in last commit. I returned sgabios to build and console output started working fine again.

pietrushnic, 2018-01-12:
@alexpro thank you for reporting that issue. We still working on automatic regression testing. We will try to fix that issue in next release.

miczyg1, 2018-03-13:
@alexpro we have fixed that in 4.0.15 release. The default Kconfig option for SeaBIOS was not properly set.

alexpro, 2018-03-18:
Sorry, but I just tried to build and run APU2 with 4.0.15 and got the same behavior: no output to serial console. I see:

```
PC Engines apu2
coreboot build 20180301
BIOS version v4.0.15
4080 MB ECC DRAM

```
And nothing more until OS begins to start. I still can press F10 key and boot from the desired source "in the blind". I reverted changes in commits 63937b6 and 1b07673 (returned sgabios.rom to build), and can see output to console. Now I see:

`Press F10 key now for boot menu`

and when pressed, see:

```
Select boot device:

1. ata0-0: KINGSTON SMS200S330G ATA-8 Hard-Disk (28626 MiBytes
2. Payload [memtest]
3. Payload [setup]
```

Maybe I doing something wrong?
I set up docker environment as described in documentation, then I use pcengines-apu2.config from configs/ directory.

miczyg1, 2018-03-19:
@alexpro make a fresh config by using `make menuconfig` command. 
Selecting Mainboard > Mainboard vendor = PC Engines
and selecting Mainboard > Mainboard model = APU2 (or any other)
is enough. The configs present in `configs` directory are not used by us in the legacy releases and they may be outdated.

alexpro, 2018-03-19:
Thanks for hint.
I was wrong with config, you are right. I did everything as you told me and now all works fine, I see output on console, without sgabios.

Thanks again!

closed 2018-03-20 by miczyg1