pcengines/coreboot #108: How to build more recent ROM for APU1?

democles9308, 2018-03-14:
Are the main steps in the doch heer for building an APU2 ROM also applicable for building a more recent coreboot rom ?
What version is still usable 4.0.x or 4.6.x ?
I have the stock rom 140908 but it dus not enable the I2C port. I’ve build a rom from the coreboot git and got the I2C part working... but il lost ‘the F10 key and the loadorder’.. Would like build a complete (recent) ROM for my APU1.
Thx

pietrushnic, 2018-03-14:
@democles9308 thank you for reporting this issue. We recovered apu1 development recently and I think we can produce binary for it in next release which is planned for beginning of April. Have you got recovery procedure if upgrade went wrong ?

@miczyg1 you tested apu1 recently have you got anything to add ?

miczyg1, 2018-03-14:
@democles9308 I have pending work to bring apu1 to our development cycle (sortbootorder needs verification). You can expect binaries at the end of March or at the beginging of April. We will also add some notes about building the firmware for apu1. 

The official coreboot git has ESC key configured to enter bootorder by default. Also it is necessary to add a configuration file to coreboot image to make serial console work in SeaBIOS.

Releases 4.0.x are based on PC Engines 20160304 release and are a little bit more stable. Releases 4.6.x are the most up-to-date binaries. The development process is mainly focused around 4.6.x version.

democles9308, 2018-03-15:
@pietrushnic and @miczyg1  : I’ve got a Usb-stick with TinyCore and the stock rooms. I build my core (last Weekend) with the latest coreboot on their git (and their doc on how to build a coreboot). Also use their Seabios payload (.11 if I’m not misstaken). Because I did not included bootorder, nor the sgabios( if this is what is needed) it booted directly from my HDD on the SATA connector. On this drive I installed the latest IPFIRE 119 (with i2ctools included).  My I2c/serial backplate was recognized by i2c tools on the (correct) adres 0x3F on /dev/i2c-1. Everything seems to work well.
So if I understanding well:
- I’ll use coreboot from this git and build coreboot
- I’ll use the bootorder and sgabios ( do I need this). Also from this git and included them as well.
Where can I find the  configfile for coreboot IOT get serial console work ?
Many thanks already ! (Also for the upcomming binaries)

democles9308, 2018-03-15:
Sorry wrong button..

miczyg1, 2018-03-15:
@democles9308  sgabios isn't really needed with SeaBIOS 1.11.x. Just run this command on the coreboot build directory:
`./cbfstool coreboot.rom add-int -i 0x3f8 -n etc/sercon-port`
This configuration file allows to redirect output to console in SeaBIOS.

democles9308, 2018-03-17:
Thanks, I’ll try an report back...might take a week because I’m abroad next week for business

democles9308, 2018-03-17:
Found some time... some updates
1. tried to load coreboot from  pcengins/coreboot git : could not select APU1 under the boards. So I used the code from coreboot git (origin/master)
2. I build seabios from your hub and included it in coreboot as  elf file
3. could boot and use ESC to select medium to boot prom
4. I cant build sortbootorder
```
DIRK-PC sortbootorder # make
make: *** No rule to make target '/opt/sortbootorder/../../../..//util/xcompile/xcompile', needed by '.xcompile'.  Stop. 
```
I2C on ipfire 119 x86-64 : see my post on forum ipfire [here](https://forum.ipfire.org/viewtopic.php?f=86&t=20440#p115015 )

Any hints ?

miczyg1, 2018-03-17:
@democles9308 
1. Our default branch is coreboot-4.0.x (the one checked out after cloning). It is a legacy branch and we have not yet ported apu1 to legacy, that is why You simply can not find the board there. 

4. coreboot is very toolchain-sensitive. Every binary is built with cross toolchain and in the isolated docker environment. Building sortbootorder and any other payload/stage needs cross-compiler.

We have already finished porting features for apu1 and will announce the binary at the end of March. If You need the binary ASAP, we can provide a pre-release binary to You. But please take into consideration that pre-release binary is not tested yet and You are using it on Your own responsibility. We do not take any responsibility for damage caused by the binary.

democles9308, 2018-03-17:
@miczyg1 
Thanks for the additional info.
1. No problems with coreboot 4.x.x so far (ipfire works)
4 understood: as explained wiki coreboot, i'v build my crossgcc-X86-64 on al debian based (Mint 18.3) host (not in a docker Env). I'v got the impression that the error is more related to the path. I've cloned sortbootorder under /opt/coreboot/payloads/external ( as indidacted in the Makefile of sortbootorder) and modified the relative path in the file so it finds /opt/coreboot/util/xcompile.... but i have other errors.
Many thanks for your efforts but I'll wait for the binaries and do some testing during that time.. Ihope I can get the  lcdproc working on the apu1board  with an I2C/serial backplate. (cfr my pstings on ipfire forum)

democles9308, 2018-03-17:
Grrrr bricked my board... ordered a lpc1aapu tool to debrick it... wont be able to test after arrival of small tool.... trial and error it is called.

miczyg1, 2018-03-17:
Well, haste makes waste. Anyway @democles9308 if You have any Raspberry Pi board (or any board with spi running linux) on hand, You can use it to flash Your apu if You are enough experienced. You can use SPI header on apu to get it connected.

democles9308, 2018-03-18:
Hi @miczyg1,
Took a Raspi and compiled the latest flashrom (1.0.62) ... soldered a header on the SPI (J12) ( next to the NX25L1606E chip  and using the info on page 8/16 of the APU1 schematic, I connected the 6 pins to my raspberry... Helas it doesn't find the chip  ( flashrom -p linux_spi:dev=/dev/spidev0.0 or 0.1... I inversed DI/DO but no change.. I thinks I enable SPI/I2C during config of my Pi, the modules are loaded when I do an lsmod...
I'll have to look for an APU2 board in the near future I guess.....
Thanks for your help anyway...  I'll keep trying

democles9308, 2018-03-23:
@miczyg1. I've received the LPC1AAPUboard and could rewrite the firmware... I'll try it with my raspberry pi to see if this works like you said and I'll try to build a new firmware using the recent code. If I can help to test in the future, I will !

democles9308, 2018-03-31:
Any updates on the port  of recent bios to APU? 
I stil have some questions on i2c.  I2Cdetect on apu1 with ipfire 119 give me
`
[root@Thor ~]# i2cdetect -F 1
Functionalities implemented by /dev/i2c-1:
I2C                              no
SMBus Quick Command              yes
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               no
SMBus Block Write                yes
SMBus Block Read                 yes
SMBus Block Process Call         no
SMBus PEC                        no
I2C Block Write                  no
I2C Block Read                   no
`
So does this mean I cant use i2c ? Lcdproc does not work

miczyg1, 2018-04-03:
@democles9308 we have finished porting apu1 and will publish the binaries at the end of this week.
I assume You did run `i2cdetect` on apu1 with official coreboot right? Could You please provide dmesg log and coreboot log if possible? From the output above I cannot tell for sure if it is possible to use i2c or not.

democles9308, 2018-04-03:
@miczyg1 many thx for your support ! I'll reflash both official rom (080914) and my own complied coreboot a,d provide info. How do I get a coreboot log?

miczyg1, 2018-04-03:
@democles9308 there are two ways to obtain logs from coreboot. Either use a program like `minicom` which can capture the serial output to a file or compile `cbmem` in `util/cbmem` in the coreboot top directory (just run `make`). Copy the cbmem binary to the platform and run `cbmem -c > some_file.txt`.

democles9308, 2018-04-04:
@miczyg1  here are some files used with my personal build of coreboot (sources from coreboot.org) and SeaBIOS (from their github). I did the build process as explained  [here](https://pelican.craoc.fr/coreboot.html)
[coreboot.log](https://github.com/pcengines/coreboot/files/1877077/coreboot.log)
[ipfireBootlog.txt](https://github.com/pcengines/coreboot/files/1877029/ipfireBootlog.txt)
After inserting i2c-dev via modprobe i2c-dev you'll see the entries are created and the modules loaded.
[ipfire_lsmod.txt](https://github.com/pcengines/coreboot/files/1877014/ipfire_lsmod.txt)

i2c-detect -l gives : 
i2c-0	smbus     	SMBus PIIX4 adapter at 0b00     	SMBus adapter
i2c-1	smbus     	SMBus PIIX4 adapter at 0b20     	SMBus adapter

Hope this can help you/us further :-)

democles9308, 2018-04-06:
@miczyg1  OK, thanky for the support anyway. If the Roms for APU1 boards are released I'll give them a try and for the lcdprov, Il try the serial or USB way.

closed 2018-04-06 by democles9308