pcengines/coreboot #138: APU2 (4.6.8) - Occasionally, BIOS boots the second drive

Strykar, 2018-05-10:
This is an odd issue, and I would like to know how to troubleshoot it.

I recently updated to 4.6.8 and I now it will randomly boot into the second drive instead of the first. It once booted into the second drive with me manually selecting the first drive via F10 but I haven't been able to replicate that. I can replicate it booting the second drive if I reboot a few times.

[image]
It's set to boot to SD card, but will occasionally boot to mSATA.

pietrushnic, 2018-05-11:
@Strykar any chance you can catch both firmware boot logs? I think we can replicate similar configuration and test it, so hopefully, we can reproduce that problem.

Strykar, 2018-05-12:
@pietrushnic If you point me to some documentation to enable these logs you speak of, I'd be happy to. I don't see a BIOS logging option when I search the docs.

miczyg1, 2018-05-12:
@Strykar I suspect that SD card sometimes disappear from bootmenu and as a resut it boots from mSATA automatically. I have build a [v4.6.8 binary](https://cloud.3mdeb.com/index.php/s/9F4d0yCBrvCf3fh/download) for apu2 with verbose log output in SeaBIOS. Please flash it on Your board, set the bootorder as You attached above. Keep rebooting (no need to press F10) and gathering serial output to files until platform boots into undesired medium. Send us the log files from both cases and we should be able to determine what has gone wrong.

Strykar, 2018-05-20:
@miczyg That makes sense and would explain the odd behavior.

It went to the wrong drive at the first reboot after flashing the debug version.
I chose SD as the first device, wrote it to the BIOS but it rebooted to the mSATA.
Two logs pasted below:

```
Boot order - type letter to move device to top.

  b SDCARD
  c mSATA
  a USB 1 / USB 2 SS and HS
  d SATA
  e mPCIe1 SATA1 and SATA2
  f iPXE (disabled)

  r Restore boot order defaults
  n Network/PXE boot - Currently Disabled
  u USB boot - Currently Enabled
  t Serial console - Currently Enabled
  o UART C - Currently Enabled
  p UART D - Currently Enabled
  m Force mPCIe2 slot CLK (GPP3 PCIe) - Currently Disabled
  h EHCI0 controller - Currently Disabled
  w Enable BIOS write protect - Currently Disabled
  x Exit setup without save
  s Save configuration and exit
s

Erasing Flash size 0x1000 @ 0x6000
Writing 366 bytes @ 0x6000
Done

Exiting ...PC Engines apu2
coreboot build 20180406
BIOS version v4.6.8
4080 MB ECC DRAM

SeaBIOS (version rel-1.11.0.4-0-gfcbc9d7)
BUILD: gcc: (coreboot toolchain v1.44 March 3rd, 2017) 6.3.0 binutils: (GNU Binutils) 2.28
Attempting to find coreboot table
Found coreboot table forwarder.
Now attempting to find coreboot memory map
Add to e820 map: 00000000 00001000 2
Add to e820 map: 00001000 0009f000 1
Add to e820 map: 000c0000 cfede000 1
Add to e820 map: cff9e000 00062000 2
Add to e820 map: f8000000 04000000 2
Add to e820 map: 100000000 2f000000 1
Add to e820 map: 00000000 00004000 1
SeaBIOS (version rel-1.11.0.4-0-gfcbc9d7)
BUILD: gcc: (coreboot toolchain v1.44 March 3rd, 2017) 6.3.0 binutils: (GNU Binutils) 2.28
Found coreboot cbmem console @ cffde000
Found mainboard PC Engines apu2
malloc preinit
Add to e820 map: 000a0000 00050000 -1
Add to e820 map: 000f0000 00010000 2
Add to e820 map: cff5e000 00040000 2
phys_alloc zone=0x000ee9b8 size=40864 align=20 ret=cff53fc0 (detail=0xcff53f90)
Relocating init from 0x000e4aa0 to 0xcff53fc0 (size 40864)
malloc init
Found CBFS header at 0xff800138
phys_alloc zone=0xcff5ded8 size=156 align=10 ret=cff53e90 (detail=0xcff53e60)
Add romfile: cbfs master header (size=32)
phys_alloc zone=0xcff5ded8 size=156 align=10 ret=cff53dc0 (detail=0xcff53d90)
Add romfile: fallback/romstage (size=20940)
phys_alloc zone=0xcff5ded8 size=156 align=10 ret=cff53cf0 (detail=0xcff53cc0)
Add romfile: config (size=140)
phys_alloc zone=0xcff5ded8 size=156 align=10 ret=cff53c20 (detail=0xcff53bf0)
Add romfile: revision (size=608)
[... 1508 lines cut ...]
AHCI/0: ... intbits 0x1, status 0x50 ...
AHCI/0: ... finished, status 0x50, OK
ahci disk read, lba    4ad, count  13, buf 0x00068000, rc 0

                             GNU GRUB  version 2.02

 +----------------------------------------------------------------------------+
 |*OpenWrt                                                                    |
 | OpenWrt (failsafe)                                                         |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 +----------------------------------------------------------------------------+

      Use the ^ and v keys to select which entry is highlighted.
      Press enter to boot the selected OS, `e' to edit the commands
      before booting or `c' for a command-line. ESC to return
      previous menu.
```

miczyg, 2018-05-21:
@Strykar I thing that @miczyg1 should be referenced in your comment, as I have no idea what the topic is about ;)

pietrushnic, 2018-05-21:
@Strykar please use pastebin or similar service for logs since it trash conversation. Thanks is advanced.

Strykar, 2018-05-22:
@pietrushnic @miczyg1 Done. Do the logs indicate it's SD detection that's causing the issue?

miczyg1, 2018-05-22:
@Strykar i have performed an automated test utilizing following procedure:
1. Set SD card boot priority to highest
2. Boot apu2 (not pressing F10) and check if desired system is loading
3. Reboot platform and repeat step 2. twenty times
4. Set msATA boot priority to highest
5. Boot apu2 (not pressing F10) and check if desired system is loading
6. Reboot platform and repeat step 5. twenty times 

On mSATA I had installed Linux debian 4.9.0-4-amd64
On SD card I had installed pfSense 2.4.2

Unfortunately (or fortunately) each time apu booted into desired system. @Strykar what is the estimated frequency of booting into undesired media (x out of y times)? Maybe I should increase the repetition counter.

Strykar, 2018-06-08:
Every 4-5 reboots it boots the wrong device here.

pietrushnic, 2018-06-08:
@Strykar we have to be as close to your configuration as possible. Did you try any other combinations of SATA/SD devices? Maybe this is feature of your hardware?

@miczyg1 can you add screenshot/test framework logs of boot menu, just to compare how we are different with what @Strykar has.

miczyg1, 2018-06-08:
@pietrushnic see my comment above, I was not entering to boot menu, to not interfere bootorder in choosing boot storage. So i do not posses such screenshots.

pietrushnic, 2018-06-08:
@miczyg RTE should have logs

caadar, 2018-08-02:
Similar issue on APU1D with actual 4.8.0.2 or 4.6.10.

Test setup 1: SD card and mSATA disc. Can boot from SD card only by manual selection through F10, else it try mSATA card first.

Test setup 2: SD card only. All works as expected.

Reproducibility: always.

miczyg1, 2018-10-08:
@caadar apu1 had broken bootorder settings. It should work now with v4.8.0.5. I will link here the release page soon.

miczyg1, 2018-10-25:
@Strykar I have spent more time and tried again to reproduce Your problem. However bootorder works as expected always. This time I have done:

1. 50 boot cycles with SD as top priority - always booted to SD card (with Debian)
2. 50 boot cycles with mSATA as top priority - always booted to mSATA disk (with pfSense)

Tested firmwares: v4.6.8 and v4.8.0.5. 

By installing different systems I was always sure, that I booted from desired drive. Since I could not reproduce the issue even once on both firmware versions I have to close the issue.

@Strykar if the problem will still exist in the newest versions, feel free to reopen the issue. I will investigate it once again then.

closed 2018-10-25 by miczyg1