pcengines/coreboot #149: flashrom can't find chip in apu2

kjake, 2018-05-17:
apu2c4 running v4.6.1
I've updated the firmware on this board at least twice before

built v4.6.8 today and wanted to flash, but flashrom isn't able to find chip?  Any tips for debugging this?  Thanks!

```
$  flashrom -p internal -L | grep Winbond | grep SPI

Vendor        Device                 Test  Known   Size   Type
                                     OK    Broken  [kB]

(P = PROBE, R = READ, E = ERASE, W = WRITE, - = N/A)
Winbond       W25Q40.V               PREW            512  SPI
Winbond       W25Q80.V               PREW           1024  SPI
Winbond       W25Q16.V               PREW           2048  SPI
Winbond       W25Q32.V               PREW           4096  SPI
Winbond       W25Q64.V               PREW           8192  SPI
Winbond       W25Q128.V              PREW          16384  SPI
Winbond       W25Q20.W                               256  SPI
Winbond       W25Q40.W                               512  SPI
Winbond       W25Q80.W               PREW           1024  SPI
Winbond       W25Q16.W                              2048  SPI
Winbond       W25Q32.W               PREW           4096  SPI
Winbond       W25Q64.W               PREW           8192  SPI
Winbond       W25Q128.W                            16384  SPI
Winbond       W25X10                 PREW            128  SPI
Winbond       W25X20                 PREW            256  SPI
Winbond       W25X40                 PREW            512  SPI
Winbond       W25X80                 PREW           1024  SPI
Winbond       W25X16                 PREW           2048  SPI
Winbond       W25X32                 PREW           4096  SPI
Winbond       W25X64                 PREW           8192  SPI
```

```
$ flashrom -p internal -V
flashrom v1.0 on FreeBSD 11.1-RELEASE-p10 (amd64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.5.6, LLVM Clang 5.0.1 (tags/RELEASE_501/final 320880), little endian
Command line (3 args): flashrom -p internal -V
Using clock_gettime for delay loops (clk_id: 4, resolution: 2ns).
Initializing internal programmer
Found candidate at: 00000500-00000510
Found coreboot table at 0x00000500.
Found candidate at: 00000000-00000248
Found coreboot table at 0x00000000.
coreboot table found at 0xcffd6000.
coreboot header(24) checksum: 2e9f table(584) checksum: 3d58 entries: 19
Vendor ID: PC Engines, part ID: PC Engines apu2
Using Internal DMI decoder.
DMI string chassis-type: "Desktop"
DMI string system-manufacturer: "PC Engines"
DMI string system-product-name: "PC Engines apu2"
DMI string system-version: "1.0"
DMI string baseboard-manufacturer: "PC Engines"
DMI string baseboard-product-name: "PC Engines apu2"
DMI string baseboard-version: "1.0"
Found chipset "AMD FCH" with PCI ID 1022:780e.
Enabling flash write... SPI base address is at 0x0
ROM strap override is not active
OK.
The following protocols are supported: LPC, FWH.
Probing for AMIC A49LF040A, 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Atmel AT49LH002, 256 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Atmel AT49LH00B4, 512 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Atmel AT49LH004, 512 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Intel 82802AB, 512 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Intel 82802AC, 1024 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for PMC Pm49FL002, 256 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
[... 14 lines cut ...]
Probing for SST SST49LF160C, 2048 kB: probe_82802ab: id1 0x24, id2 0x41, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FLW040A, 512 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FLW040B, 512 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FLW080A, 1024 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FLW080B, 1024 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FW002, 256 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FW016, 2048 kB: probe_82802ab: id1 0x24, id2 0x41, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FW040, 512 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50FW080, 1024 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50LPW080, 1024 kB: probe_82802ab: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for ST M50LPW116, 2048 kB: probe_82802ab: id1 0x24, id2 0x41, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V040A, 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V040B, 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V040C, 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V040FA, 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V040FB, 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V040FC, 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V080A, 1024 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W49V002A, 256 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W49V002FA, 256 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V080FA, 1024 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V080FA (dual mode), 512 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash content
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

pietrushnic, 2018-05-17:
@kjake what looks suspicious is `ROM strap override is not active` and second is SPI base address which is `0x0`. Are you sure you didn't use `sudo` before?

kjake, 2018-05-17:
@pietrushnic I remembered that my version of flashrom updated - probably since the last time I flashed something.  So, I downgraded to a earlier release and it found the chip!  I thought that I might have a hardware issue, didn't think it would be an issue with the flashrom util itself.

```
$ flashrom -p internal:boardmismatch=force -w apu2_v4.6.8.rom
flashrom v0.9.9-r1955 on FreeBSD 11.1-RELEASE-p10 (amd64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
coreboot table found at 0xcffd6000.
Found chipset "AMD FCH".
Enabling flash write... OK.
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
This coreboot image (PC Engines:apu2) does not appear to
be correct for the detected mainboard (PC Engines:PC Engines apu2).
Proceeding anyway because user forced us to.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
```

pietrushnic, 2018-05-17:
@kjake great that you solved that. We use very recent version of flashrom and don't have problem, but we use Linux so probably some flashrom problem on FreeBSD.

Maybe it is worth to inform flashrom mailing list about regression?

kjake, 2018-05-18:
Good idea - I'm awaiting confirmation of my subscription to their mailing list; failing that, I can go to irc.  Thanks for listening!  I'll close this.

closed 2018-05-18 by kjake