pcengines/coreboot #180: Building FILO payload fails

stapelberg, 2018-06-12:
I ran `./apu2/apu2-documentation/scripts/apu2_fw_rel.sh build-ml menuconfig` and switched the payload from SeaBIOS to FILO, as I’d like to directly boot a vmlinuz image from a FAT partition.

Unfortunately, the build fails when building for pcengines/apu2 (but e.g. pcengines/apu1 succeeds):

```% ./apu2/apu2-documentation/scripts/apu2_fw_rel.sh build-ml 
build-ml
    GEN        generated/bootblock.ld
    CP         bootblock/arch/x86/bootblock.ld
    LINK       cbfs/fallback/bootblock.debug
    OBJCOPY    cbfs/fallback/bootblock.elf
    OBJCOPY    bootblock.raw.bin
    CREATE     build/mainboard/pcengines/apu2/cbfs-file.5CJdXr.out (from /release/apu2/coreboot/.config)
    GIT        FILO STABLE
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
Fetching origin
Switched to branch 'STABLE'
*** Default configuration is based on 'configs/defconfig'
#
# configuration written to .config
#
#
# configuration written to .config
#
    INSTALL    ../external/FILO/filo/build/libpayload/lib
    INSTALL    ../external/FILO/filo/build/libpayload/include
    INSTALL    ../external/FILO/filo/build/libpayload/bin
    CONFIG     FILO STABLE
Using binary libpayload, nothing to configure
*** Default configuration is based on 'configs/defconfig'
*
* FILO Configuration
*
Build for architecture
> 1. x86 (TARGET_I386)
  2. ARM (TARGET_ARM) (NEW)
choice[1-2]: 1
Include a MultiBoot header (MULTIBOOT_IMAGE) [Y/n/?] (NEW) y
*
* Interface Options
*
Use GRUB like interface (USE_GRUB) [Y/n/?] y
non-interactive interface (NON_INTERACTIVE) [N/y/?] (NEW) n
Command line prompt (PROMPT) [filo] filo
GRUB menu.lst filename (MENULST_FILE) [hda3:/boot/filo/menu.lst] hda3:/boot/filo/menu.lst
Timeout for loading menu.lst (MENULST_TIMEOUT) [0] 0
Use MD5 passwords in menu.lst? (USE_MD5_PASSWORDS) [Y/n/?] y
Support for parsing isolinux.cfg config files (ISOLINUX_PARSER) [Y/n/?] y
*
* Drivers
*
IDE DISK support (IDE_DISK) [N/y/?] n
  New IDE driver (IDE_NEW_DISK) [Y/n/?] y
Use libpayload's storage drivers (LIBPAYLOAD_STORAGE) [N/y/?] (NEW) n
USB Stack (USB_DISK) [Y/n/?] y
NAND Flash support (FLASH_DISK) [N/y/?] n
PCI support (SUPPORT_PCI) [Y/n] y
  Scan all PCI busses (PCI_BRUTE_SCAN) [N/y/?] n
  Sound Support (SUPPORT_SOUND) [N/y] n
Flash memory lockdown (FLASHROM_LOCKDOWN) [N/y/?] (NEW) n
Provide flashupdate command (FLASHUPDATE) [N/y/?] (NEW) n
*
* Filesystems
*
EXT2 filesystem (FSYS_EXT2FS) [Y/n] y
FAT (MSDOS) filesystem (FSYS_FAT) [Y/n] y
JFS (FSYS_JFS) [N/y] n
Minix filesystem (FSYS_MINIX) [N/y] n
ReiserFS (FSYS_REISERFS) [Y/n] y
XFS (FSYS_XFS) [N/y] n
ISO9660 filesystem (FSYS_ISO9660) [Y/n] y
  El Torito bootable CDROMs (ELTORITO) [Y/n/?] y
Compressed RAM filesystem (CRAMFS) (FSYS_CRAMFS) [N/y] n
Squash filesystem (FSYS_SQUASHFS) [N/y] n
CBFS ROM Image filesystem (FSYS_CBFS) [Y/n] y
*
* Loaders
*
Standard Linux Loader (LINUX_LOADER) [Y/n/?] y
Windows CE Loader (WINCE_LOADER) [N/y/?] n
Artec Loader (ARTEC_BOOT) [N/y/?] n
*
* Debugging & Experimental
*
Enable experimental features (EXPERIMENTAL) [Y/n/?] y
DEBUG_ALL (DEBUG_ALL) [N/y] n
DEBUG_ELFBOOT (DEBUG_ELFBOOT) [N/y] n
DEBUG_SEGMENT (DEBUG_SEGMENT) [N/y] n
DEBUG_SYS_INFO (DEBUG_SYS_INFO) [N/y] n
DEBUG_BLOCKDEV (DEBUG_BLOCKDEV) [N/y] n
DEBUG_VFS (DEBUG_VFS) [N/y] n
DEBUG_FSYS_EXT2FS (DEBUG_FSYS_EXT2FS) [N/y] n
DEBUG_PCI (DEBUG_PCI) [N/y] n
DEBUG_LINUXLOAD (DEBUG_LINUXLOAD) [N/y] n
DEBUG_IDE (DEBUG_IDE) [N/y] n
DEBUG_USB (DEBUG_USB) [N/y] n
DEBUG_ELTORITO (DEBUG_ELTORITO) [N/y] n
Developer Tools (DEVELOPER_TOOLS) [Y/n/?] y
#
# configuration written to .config
#
Using binary libpayload, nothing to configure
*
* FILO Configuration
*
Build for architecture
> 1. x86 (TARGET_I386)
  2. ARM (TARGET_ARM)
choice[1-2]: 1
Include a MultiBoot header (MULTIBOOT_IMAGE) [Y/n/?] y
*
* Interface Options
*
Use GRUB like interface (USE_GRUB) [Y/n/?] y
non-interactive interface (NON_INTERACTIVE) [N/y/?] n
Command line prompt (PROMPT) [filo] filo
GRUB menu.lst filename (MENULST_FILE) [hda3:/boot/filo/menu.lst] hda3:/boot/filo/menu.lst
Timeout for loading menu.lst (MENULST_TIMEOUT) [0] 0
Use MD5 passwords in menu.lst? (USE_MD5_PASSWORDS) [Y/n/?] y
Support for parsing isolinux.cfg config files (ISOLINUX_PARSER) [Y/n/?] y
*
* Drivers
*
IDE DISK support (IDE_DISK) [N/y/?] n
  New IDE driver (IDE_NEW_DISK) [Y/n/?] y
Use libpayload's storage drivers (LIBPAYLOAD_STORAGE) [N/y/?] n
USB Stack (USB_DISK) [Y/n/?] y
NAND Flash support (FLASH_DISK) [N/y/?] n
PCI support (SUPPORT_PCI) [Y/n] y
  Scan all PCI busses (PCI_BRUTE_SCAN) [N/y/?] n
  Sound Support (SUPPORT_SOUND) [N/y] n
Flash memory lockdown (FLASHROM_LOCKDOWN) [N/y/?] n
Provide flashupdate command (FLASHUPDATE) [N/y/?] n
*
* Filesystems
*
EXT2 filesystem (FSYS_EXT2FS) [Y/n] y
FAT (MSDOS) filesystem (FSYS_FAT) [Y/n] y
JFS (FSYS_JFS) [N/y] n
Minix filesystem (FSYS_MINIX) [N/y] n
ReiserFS (FSYS_REISERFS) [Y/n] y
XFS (FSYS_XFS) [N/y] n
ISO9660 filesystem (FSYS_ISO9660) [Y/n] y
  El Torito bootable CDROMs (ELTORITO) [Y/n/?] y
Compressed RAM filesystem (CRAMFS) (FSYS_CRAMFS) [N/y] n
Squash filesystem (FSYS_SQUASHFS) [N/y] n
CBFS ROM Image filesystem (FSYS_CBFS) [Y/n] y
*
* Loaders
*
Standard Linux Loader (LINUX_LOADER) [Y/n/?] y
Windows CE Loader (WINCE_LOADER) [N/y/?] n
Artec Loader (ARTEC_BOOT) [N/y/?] n
*
* Debugging & Experimental
*
Enable experimental features (EXPERIMENTAL) [Y/n/?] y
DEBUG_ALL (DEBUG_ALL) [N/y] n
DEBUG_ELFBOOT (DEBUG_ELFBOOT) [N/y] n
DEBUG_SEGMENT (DEBUG_SEGMENT) [N/y] n
DEBUG_SYS_INFO (DEBUG_SYS_INFO) [N/y] n
DEBUG_BLOCKDEV (DEBUG_BLOCKDEV) [N/y] n
DEBUG_VFS (DEBUG_VFS) [N/y] n
DEBUG_FSYS_EXT2FS (DEBUG_FSYS_EXT2FS) [N/y] n
DEBUG_PCI (DEBUG_PCI) [N/y] n
DEBUG_LINUXLOAD (DEBUG_LINUXLOAD) [N/y] n
DEBUG_IDE (DEBUG_IDE) [N/y] n
DEBUG_USB (DEBUG_USB) [N/y] n
DEBUG_ELTORITO (DEBUG_ELTORITO) [N/y] n
Developer Tools (DEVELOPER_TOOLS) [Y/n/?] y
#
# configuration written to .config
#
    MAKE       FILO STABLE
Found libpayload as /release/apu2/coreboot/payloads/external/FILO/filo/build/libpayload/lib/libpayload.a
  CC      build/main/grub/grub.o
/bin/sh: 1: -Ibuild: not found
make[2]: *** [/release/apu2/coreboot/payloads/external/FILO/filo/build/main/grub/grub.o] Error 127
make[1]: *** [filo] Error 2
make: *** [filo] Error 2
```

I tried making the following change in the filo repository:
```diff
diff --git i/Makefile w/Makefile
index 68338ba..d5ee468 100644
--- i/Makefile
+++ w/Makefile
@@ -151,7 +151,7 @@ endif
 
 $(obj)/filo: $(OBJS) $(LIBPAYLOAD)
 	printf "  LD      $(subst $(shell pwd)/,,$(@))\n"
-	CC=$(CC) $(LPGCC) $(OBJS) $(LIBS) -o $@
+	CC="$(CC)" $(LPGCC) $(OBJS) $(LIBS) -o $@
 
 $(obj)/filo.bzImage: $(TARGET) $(obj)/x86/linux_head.o
 	$(OBJCOPY) -O binary $(obj)/x86/linux_head.o $@.tmp1
@@ -171,11 +171,11 @@ $(KCONFIG_AUTOHEADER): $(src)/.config
 $(OBJS): $(KCONFIG_AUTOHEADER) $(obj)/version.h | libpayload
 $(obj)/%.o: $(src)/%.c
 	printf "  CC      $(subst $(shell pwd)/,,$(@))\n"
-	CC=$(CC) $(LPGCC) -MMD $(CFLAGS) $(CPPFLAGS) -c -o $@ $<
+	CC="$(CC)" $(LPGCC) -MMD $(CFLAGS) $(CPPFLAGS) -c -o $@ $<
 
 $(obj)/%.S.o: $(src)/%.S
 	printf "  AS      $(subst $(shell pwd)/,,$(@))\n"
-	AS=$(AS) $(LPAS) $(ASFLAGS) -o $@ $<
+	AS="$(AS)" $(LPAS) $(ASFLAGS) -o $@ $<
 
 $(obj)/%.map: $(obj)/%
 	printf "  SYMS    $(subst $(shell pwd)/,,$(@))\n"
```

While this fixes the immediate issue, I’m now getting errors when linking:
```
/release/apu2/coreboot/payloads/external/FILO/filo/build/libpayload/lib/libpayload.a(pdcdisp.libcurses.o):(.data.serial_acs_map+0x0): multiple definition of `serial_acs_map'
/release/apu2/coreboot/payloads/external/FILO/filo/build/libpayload/lib/libpayload.a(tinycurses.libcurses.o):(.data.serial_acs_map+0x0): first defined here
/release/apu2/coreboot/payloads/external/FILO/filo/build/libpayload/lib/libpayload.a(pdcdisp.libcurses.o): In function `PDC_gotoyx':
/release/apu2/coreboot/payloads/libpayload/curses/pdcurses-backend/pdcdisp.c:137: multiple definition of `fallback_acs_map'
/release/apu2/coreboot/payloads/external/FILO/filo/build/libpayload/lib/libpayload.a(tinycurses.libcurses.o):/release/apu2/coreboot/payloads/libpayload/curses/tinycurses.c:721: first defined here
/release/apu2/coreboot/payloads/external/FILO/filo/build/libpayload/lib/libpayload.a(pdcdisp.libcurses.o): In function `PDC_gotoyx':
/release/apu2/coreboot/payloads/libpayload/curses/pdcurses-backend/pdcdisp.c:137: multiple definition of `console_acs_map'
[…]
```

So I’m thinking I might not be on the correct path here.

Do you have any ideas as to why the FILO payload doesn’t build for the apu2?

Thanks,

pietrushnic, 2018-06-12:
@stapelberg it looks like names collision, this typically mean that incorrect compiler was used. Maybe libpayload during compilation didn't used xgcc or somehow fetched system libraries. Any reason why you try old build method?

@kmalkki any insights?

stapelberg, 2018-06-13:
I’m building 4.6.1 because all newer versions fail to PXE-boot on my apu2c4. I mailed support@ about it, but haven’t received a reply yet.

kmalkki, 2018-06-13:
It's a symptom of the .xcompile format in coreboot tree changing I believe. I don't quite see why/how you would have old coreboot/.xcompile but it sounds like it.

Try renaming that and see if new one is generated in compatible format.

Another choice would be to try revert followint in FILO repo, but it should not be necessary.
e7c85fe Sync with coreboot's latest xcompile script

stapelberg, 2018-06-13:
After identifying https://github.com/pcengines/coreboot/issues/181, I tried building FILO with the newer pce-fw-builder environment. The quoting patch I provided in my first comment is still required, but aside from that, FILO builds.

I modified FILO’s config like so:
```patch
diff --git i/configs/defconfig w/configs/defconfig
index 0fe3019..31c2dc1 100644
--- i/configs/defconfig
+++ w/configs/defconfig
@@ -10,7 +10,7 @@ CONFIG_TARGET_I386=y
 #
 CONFIG_USE_GRUB=y
 CONFIG_PROMPT="filo"
-CONFIG_MENULST_FILE="hda3:/boot/filo/menu.lst"
+CONFIG_MENULST_FILE="hda1:/menu.lst"
 CONFIG_MENULST_TIMEOUT=0
 CONFIG_USE_MD5_PASSWORDS=y
 CONFIG_ISOLINUX_PARSER=y
@@ -52,16 +52,16 @@ CONFIG_LINUX_LOADER=y
 # Debugging & Experimental
 #
 CONFIG_EXPERIMENTAL=y
-# CONFIG_DEBUG_ALL is not set
-# CONFIG_DEBUG_ELFBOOT is not set
-# CONFIG_DEBUG_SEGMENT is not set
-# CONFIG_DEBUG_SYS_INFO is not set
-# CONFIG_DEBUG_BLOCKDEV is not set
-# CONFIG_DEBUG_VFS is not set
-# CONFIG_DEBUG_FSYS_EXT2FS is not set
-# CONFIG_DEBUG_PCI is not set
-# CONFIG_DEBUG_LINUXLOAD is not set
-# CONFIG_DEBUG_IDE is not set
-# CONFIG_DEBUG_USB is not set
-# CONFIG_DEBUG_ELTORITO is not set
+CONFIG_DEBUG_ALL=y
+CONFIG_DEBUG_ELFBOOT=y
+CONFIG_DEBUG_SEGMENT=y
+CONFIG_DEBUG_SYS_INFO=y
+CONFIG_DEBUG_BLOCKDEV=y
+CONFIG_DEBUG_VFS=y
+CONFIG_DEBUG_FSYS_EXT2FS=y
+CONFIG_DEBUG_PCI=y
+CONFIG_DEBUG_LINUXLOAD=y
+CONFIG_DEBUG_IDE=y
+CONFIG_DEBUG_USB=y
+CONFIG_DEBUG_ELTORITO=y
 CONFIG_DEVELOPER_TOOLS=y
```

Unfortunately, when booting, I get only the following output, and then the system hangs:
```
PC Engines apu2
coreboot build 20180511
BIOS version v4.6.9
Missing or invalid mpcie2_clk knob, forcing CLK of mPCIe2 slot is not enabled .
Missing or invalid ehci0 knob, enable ehci0.
Missing or invalid uart knob, disable port.
Missing or invalid uart knob, disable port.
4080 MB ECC DRAM

coreboot: 4.6-280-gc3314b1c56-dirty
FILO version 0.6.0 (coreboot@698afad1a180) Wed Jun 13 20:19:05 UTC 2018
boot EAX = 0x0                                                         
boot EBX = 0x0
boot arg = 0x0
Current location: 0x100000-0x17a35f
Relocating to 0xcff23000-0xcff9d35f... 
```

(I did copy the bootorder file using `./release/coreboot/build/cbfstool release/coreboot/build/coreboot.rom add -f bootorder -n bootorder -t raw`, which addresses a bunch of warnings about knobs, but doesn’t help with the boot process, either.)

Do you have any idea as to why that is? Do the memory addresses look correct to you?

miczyg1, 2018-06-14:
@stapelberg adding `bootorder` to coreboot image is not as simple.  `bootorder` has to be properly aligned in the image i.e. 4k alignment. The special treatment for this file is needed because it is used to reflash its contents while changing it in sortbootorder (4k alignment to fit in erase chunk size). The command will look like this:
```
release/coreboot/build/coreboot.rom add -f bootorder -n bootorder -t raw -a 0x1000
```

I will try to build FILO and investigate the problem

miczyg1, 2018-06-14:
@stapelberg I managed to compile FILO, but also needed the double quotes as You mentioned above. Our payload development is mainly focused on SeaBIOS, so we do not have much experience with FILO codebase. Also FILO development is rather stall (few commits per year), it is likely to have many bugs (like the one with compiler).

I got to the same point with relocation and it freezes here. Did You try compiling grub as a payload maybe?

stapelberg, 2018-06-14:
I tried, but the GRUB payload doesn’t compile either :-/

miczyg1, 2018-06-14:
@stapelberg I managed to compile grub and boot Debian installed on mSATA disk. Here's short guide what to do:
1. Get `grub.cfg` file from installed system and copy it to coreboot rootdir
2. Change payload from SeaBIOS to GRUB2
3. Select `Include GRUB2 runtime config file into ROM image` option to include the `grub.cfg` to coreboot image
4. Save config and run `make`
If You will use pce-fw-builder there should be no problem. I hope this will help in some way.

stapelberg, 2018-06-14:
Thanks for looking into this! I tried to reproduce your steps, but unfortunately, for me, compilation fails with:

```
Making all in grub-core/gnulib
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in .
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c: In function 'do_setkey':
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c:229:9: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
         ;
         ^
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c:352:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (i = 0; i < keylen; i++)
                     ^
ar: `u' modifier ignored since `D' is the default (see `U')
In file included from ../grub-core/osdep/platform_unix.c:2:0:
../grub-core/osdep/unix/platform.c: In function 'grub_install_register_efi':
../grub-core/osdep/unix/platform.c:158:6: error: 'rc' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   if (ret)
      ^
cc1: all warnings being treated as errors
Makefile:9565: recipe for target 'grub-core/osdep/grub_mkrescue-platform_unix.o' failed
```

The full grub build log is at https://gist.github.com/stapelberg/ddbb0d41a2c6001187932c86a65f313c

Are you not building GRUB HEAD?

Anyway, when fixing the compilation error like so:
```diff
diff --git i/grub-core/osdep/unix/platform.c w/grub-core/osdep/unix/platform.c
index ca448bc11..55b8f4016 100644
--- i/grub-core/osdep/unix/platform.c
+++ w/grub-core/osdep/unix/platform.c
@@ -85,7 +85,7 @@ grub_install_remove_efi_entries_by_distributor (const char *efi_distributor)
   pid_t pid = grub_util_exec_pipe ((const char * []){ "efibootmgr", NULL }, &fd);
   char *line = NULL;
   size_t len = 0;
-  int rc;
+  int rc = 0;
 
   if (!pid)
     {
```

…I then get a new build error:
```
    MAKE       GRUB2 HEAD
Making all in grub-core/gnulib
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in .
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c: In function 'do_setkey':
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c:229:9: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
         ;
         ^
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c:352:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (i = 0; i < keylen; i++)
                     ^
ar: `u' modifier ignored since `D' is the default (see `U')
config.status: creating grub-kbdcomp
config.status: creating 00_header
config.status: creating 10_linux
config.status: creating 20_linux_xen
config.status: creating 30_os-prober
config.status: creating 40_custom
config.status: creating 41_custom
config.status: creating grub-mkconfig_lib
config.status: creating grub-shell
config.status: creating grub-shell-tester
config.status: creating grub-fs-tester
config.status: creating grub-mkconfig
config.status: creating grub-set-default
config.status: creating grub-reboot
Making all in grub-core
/usr/bin/ld: -r and -pie may not be used together
collect2: error: ld returned 1 exit status
Makefile:24791: recipe for target 'disk.module' failed
make[5]: *** [disk.module] Error 1
Makefile:24051: recipe for target 'all' failed
make[4]: *** [all] Error 2
Makefile:10926: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:3134: recipe for target 'all' failed
make[2]: *** [all] Error 2
Makefile:49: recipe for target 'grub2' failed
make[1]: *** [grub2] Error 2
payloads/external/Makefile.inc:178: recipe for target 'grub2' failed
make: *** [grub2] Error 2
```

I find it strange that you’re not running into these issues. I’m using pce-fw-builder git revision 22465e8e77e6021d54f57d88257de90c1b6ba2ec.

miczyg1, 2018-07-04:
@stapelberg sorry that I did not respond to You recently.

Regarding GRUB version it is a good approach to build stable versions. GRUB HEAD may be the latest version, but is less tested and some commits made on top may be related to other changes that were not merged yet. Big projects = big community = frequent commits. Often it leads to small bugs in code which are not always caught and cause problems in other projects like coreboot in this case. 

Summing it up, please try using stable version indicated as `2.02` in `GRUB2 version` Kconfig option

stapelberg, 2018-07-12:
Thanks for following up.

I gave GRUB2 2.02 a shot, but am running into the same error:
```
[…]
    MAKE       GRUB2 2.02
Making all in grub-core/gnulib
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in .
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c: In function 'do_setkey':
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c:229:9: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
         ;
         ^
../grub-core/lib/libgcrypt-grub/cipher/rijndael.c:352:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (i = 0; i < keylen; i++)
                     ^
ar: `u' modifier ignored since `D' is the default (see `U')
config.status: creating grub-kbdcomp
config.status: creating 00_header
config.status: creating 10_linux
config.status: creating 20_linux_xen
config.status: creating 30_os-prober
config.status: creating 40_custom
config.status: creating 41_custom
config.status: creating grub-mkconfig_lib
config.status: creating grub-shell
config.status: creating grub-shell-tester
config.status: creating grub-fs-tester
config.status: creating grub-mkconfig
config.status: creating grub-set-default
config.status: creating grub-reboot
Making all in grub-core
/usr/bin/ld: -r and -pie may not be used together
collect2: error: ld returned 1 exit status
Makefile:24260: recipe for target 'disk.module' failed
make[5]: *** [disk.module] Error 1
Makefile:23531: recipe for target 'all' failed
make[4]: *** [all] Error 2
Makefile:10903: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:3129: recipe for target 'all' failed
make[2]: *** [all] Error 2
Makefile:49: recipe for target 'grub2' failed
make[1]: *** [grub2] Error 2
payloads/external/Makefile.inc:178: recipe for target 'grub2' failed
make: *** [grub2] Error 2
```

damiankaras, 2021-10-18:
Available here: https://github.com/pcengines/apu2-documentation/issues/208

closed 2021-10-18 by damiankaras