pcengines/coreboot #185: vboot for APU

pkubaj, 2018-07-20:
You spoke about TPM and vboot on APU on ECC2017.

Could you upload the necessary FMD file to the repo and make a simple how-to for vboot-enabled coreboot?

That would probably make APU the 1st device with vboot that is not a Chromebook :)

pietrushnic, 2018-07-20:
@pkubaj yes, we getting this inquiry not only from you, but right now we are very busy with various other issues. Our code was note accepted that time to mainline [this was the review](https://review.coreboot.org/#/c/coreboot/+/21983/)/. Quite a lot changed since that time, but we plan to do something with that in Q3 2018 hopefully.

miczyg1, 2019-06-11:
We have added initial support of vboot and measured boot mode. It will be evolved and improved over time

https://github.com/pcengines/coreboot/pull/303

I believe we can close the issue.

miczyg1, 2019-06-11:
Building details: https://github.com/pcengines/apu2-documentation/blob/master/docs/apu2_vboot.md

closed 2019-06-11 by miczyg1

pkubaj, 2019-06-11:
Is it possible to use it on APU1?

miczyg1, 2019-06-11:
@pkubaj I am not sure about sufficient free space on SPI flash. 2MiB may be too small. But with 8MiB it will go for sure.