pcengines/coreboot #196: CPU clock stuck at low frequency (4.8.0.2)

NanoCaiordo, 2018-09-02:
After a short while CPU,  get stuck at 600Mhz and not even stressing all four cores.
This happens on pfSense with or without:
hint.acpi_perf.0.disabled
hint.acpi_throttle.0.disabled
hint.p4tcc.0.disabled
hint.hwpstate.0.disabled

NanoCaiordo, 2018-09-02:
Hello,

APU2C4, boot from mSATA, no USB or add-on cards. Plain wired connections.

On legacy it's all good no matter which FreeBSD tweaks I use, I was and then I went back on legacy because this issue.

To reproduce and monitor:
Install latest pfSense 2.4.3 from https://www.pfsense.org/download/

Reboot to enter vanilla install and press 8 (shell).

Install turbostat to monitor CPU:
`pkg add http://pkg0.isc.freebsd.org/FreeBSD:11:amd64/latest/All/turbostat-4.17_1.txz`
`rehash`

Save this script as 'watch' and make it executable:
`#!/bin/sh`
`while [ 1 -lt 2 ]`
`do`
`eval $@`
`sleep 5`
`done`

The call it as:
`./watch turbostat --quiet --show CPU,Busy%,Avg_MHz,Bzy_MHz,TSC_MHz sysctl dev.cpu.0.freq`

"dev.cpu.0.freq" row will show what FreeBSD think is the frequency, and it will always be 1000MHz

The "Bzy_MHz" columns will show real frequency and after few minutes, or as soon it gets to P-state 2 or a C-State (not sure here), frequency will be stuck at 600MHz. Not even a cpu stress test will make it raise.

Please let me know if you need any more details.

pietrushnic, 2018-09-02:
@NanoCaiordo thanks for extensive info. This sound like broken ACPI. We will take a look at it. @miczyg1 @krystian-hebel please put that on our roadmap.

madman2003, 2018-09-16:
I see the same/a similar phenomena on linux. Any idea when this will be looked into?

pietrushnic, 2018-09-17:
@madman2003 hopefylly v4.8.0.5 or v4.8.0.6.

miczyg1, 2018-09-18:
@madman2003 I can easily set manually the frequency to 1000MHz on Debian using:

```
apt-get install cpufrequtils sysfsutils
cpufreq-info
cpufreq-set -g userspace
cpufreq-set -f 1000MHz -c 0
cpufreq-set -f 1000MHz -c 1
cpufreq-set -f 1000MHz -c 2
cpufreq-set -f 1000MHz -c 3
cpufreq-info
```

That is one thing. Another approach I have tried was using stress command and simultaneously watch cpufreq:

```
sudo stress -c 4 -i 2 -m 2 &
sudo watch -n 1  cat /sys/devices/system/cpu/cpu*/cpufreq/cpuinfo_cur_freq
# dont forget to do fg and stop the stress
```
It showed me 1000MHz on each core. I will write some automated tests to verify it works for sure.

@NanoCaiordo in my case dev.cpu.0.freq is showing 800 Mhz all the time (watching the script for 10 minutes), however the Bzy_MHz is ~600MHz and sometimes reaches 1000MHz. Using `stress` utility I couldn't raise it to 1000MHz. My experience with *BSD systems is very low so it will take much time to check the issue. However after seeing the results is Linux, I would suspect that it is some BSD configuration stuff. Would appreciate Your help.

madman2003, 2018-09-18:
@miczyg1 I can reproduce your findings on 4.8.0.2, what's interesting the CPU MHz shown in /proc/cpuinfo is stuck at 600 MHz, or rather 598.8XX MHz. It used to be that /proc/cpuinfo would show changing frequency as well. Which one is right?

Can we somehow benchmark the cpu to confirm performance?

NanoCaiordo, 2018-09-18:
I did previously run some benchmark as follow:

```
# stress-ng` --cpu 4 --cpu-method matrixprod --timeout 30 --metrics

/* 4.8.0.2 after few minutes of system boot */
stress-ng: info:  [56392] cpu                1407     30.03    107.73      0.05        46.85        13.05

/* legacy */
stress-ng: info:  [80954] cpu                1987     30.04    116.20      0.05        66.15        17.09

```

NanoCaiordo, 2018-09-18:
I would also add that trusting any OS's sysutil would likely take you to wrong conclusions, bechmarking seems the only data to be trusted.

miczyg1, 2018-09-19:
@NanoCaiordo this stress-ng (or stress) seems not to be suitable for our case. From stress-ng man:

```
--metrics
              output  number of bogo operations in total performed by the stress processes.  Note
              that these are not a reliable metric of performance or throughput and have not been
              designed  to be used for benchmarking whatsoever
```

I have just used this tool to prove that frequency is not stuck at 600MHz, just for quick check. Performance is kind of different story. Feel free to suggest better methods. I will look for another approaches.

@madman2003 when running stress in the background, /proc/cpuinfo showed me ~940MHz on all cores:

```
stress -c 4 -i 2 -m 2 &
watch -n1 "cat /proc/cpuinfo | grep MHz"
```
Without stressing the cores it was ~700MHz according to /proc/cpuinfo.

madman2003, 2018-09-19:
On an APU2C4 I also tested against 4.8.0.4, and still get it the CPU MHz dropping to 598MHz shortly after booting. First 20/30 seconds it seems to behave ok'ish, except that it never goes to 1000MHz, but stays closer to 930MHz.

Are we all on the same hardware platform?

NanoCaiordo, 2018-09-21:
@miczyg1 
> ```
> --metrics
>               output  number of bogo operations in total performed by the stress processes.  Note
>               that these are not a reliable metric of performance or throughput and have not been
>               designed  to be used for benchmarking whatsoever
> ```
He might say so to stay on the safe side of the benchmark land, but surely it can provide a good guidance.

> I have just used this tool to prove that frequency is not stuck at 600MHz, just for quick check. Performance is kind of different story. Feel free to suggest better methods. I will look for another approaches.

I ran the test multiple times, also using a lighter --cpu-method, and bogo operations were consistent with method and available CPU frequency.

I let the system entirely boot and start stressing every few minutes or so and it's noticeable when the available frequency drops.
```
single core: 512,  513,  401,  404,  415,  414,  412,  411
quad core:  1824, 1838, 1324, 1471, 1402, 1535, 1550, 1537
```

As I noticed and pointed out earlier, at this stage I wouldn't personally trust any tool that just grab and show the reported frequency. Bogo ops, even if not spot on because it use available resources along every other running process and I guess this is the very same reason why man advices against it (in fact it's not "designed" for it), does return good enough answers, consistent between reboots and long run times.

>@madman2003
>Are we all on the same hardware platform?

I'm on APU2C4

NanoCaiordo, 2018-09-21:
@miczyg1 
> in my case dev.cpu.0.freq is showing 800 Mhz all the time (watching the script for 10 minutes), however the Bzy_MHz is ~600MHz and sometimes reaches 1000MHz. Using `stress` utility I couldn't raise [...]

It seems Debian/Ubuntu aren't effected, but FreeBSD and other Linux are by just updating bios/coreboot, I suspect different kernels and some change in bios/coreboot aren't compatible.
From what I can tell, the issue is between ACPI and the kernel as it is also possible to trick the kernel with 1/1.2/1.4GHz frequencies.

NanoCaiordo, 2018-09-26:
I cannot test v4.8.x for now, but what about using pmcstat to count actual CPU cycles:
```
kldload hwpmc
pmcstat -s cycles -w 1
             996942372            996997326            996996183            996863990
             998146885            998128892            998154678            998146000
             998124323            998420161            998396981            998347180
             998031773            997702326            997705445            997731932
             998117590            998174893            998248297            998222949
 ```

madman2003, 2018-09-27:
Anyone know a linux alternative to pmcstat?

NanoCaiordo, 2018-09-28:
@madman2003 which OS are you using

miczyg1, 2018-09-28:
@NanoCaiordo so pmcstat on the contrary shows 1GHz...

We have been running Memtest Pro recently on apus and report generated after tests showed ~800MHz CPU detected (but this was tianocore UEFI payload which might doing something nasty behind scenes). This is another proof that something may be wrong. I would also suspect firmware/ACPI to be at fault, as You suggested.

Anyway we will look closer on ACPI tables and try to locate the issue.

NanoCaiordo, 2018-09-28:
@miczyg1 that's great, let me know if I can be of any help because during weekends I have few hours windows where the firewall could be taken down for tests.

madman2003, 2018-09-29:
@NanoCaiordo archlinux, but just knowing what is available for linux kernel in general should help

krystian-hebel, 2018-10-24:
@NanoCaiordo I'm having problems trying `turbostat`:

```
#/root: turbostat --quiet --show CPU,Bus
y%,Avg_MHz,Bzy_MHz,TSC_MHz sysctl dev.cpu.0.freq                                
turbostat: /dev/cpuctl0 missing, kldload cpuctl: No such file or directory 
```
I believe I need a kernel module, but don't where to get it.

I did some preliminary tests and it seems that P-states transitions work OK, so I think the problem is with C-states, I'm investigating it further, but, well, lets just say that FreeBSD isn't the OS that I know good enough :)

NanoCaiordo, 2018-10-25:
> @NanoCaiordo I'm having problems trying `turbostat`:
>
> ```
> #/root: turbostat --quiet --show CPU,Bus
> y%,Avg_MHz,Bzy_MHz,TSC_MHz sysctl dev.cpu.0.freq
> turbostat: [...]

This is weird, FreeBSD comes with it.

`uname -rpm`
`stat -F /boot/kernel/cpuctl.ko`
`kload cpuctl`

miczyg1, 2018-11-28:
@soder10 this is our result of the research we have made till now: https://github.com/pcengines/apu2-documentation/blob/master/docs/debug/cpu_frequency.md

@krystian-hebel did You do more tests on BSD? Any news ?

krystian-hebel, 2018-11-28:
Unfortunately I can reproduce it any more, as mentioned in attached document. Nothing has changed in the hardware or firmware/OS version as far as I can tell, but the bug isn't triggered. I'll keep trying, but at the moment test platform is needed elsewhere.

miczyg1, 2018-11-28:
@soder10 we do have CPB feature on our roadmap, however we do face much more important issues which we focus mainly on now, for example: https://github.com/pcengines/apu2-documentation/issues/64

soder10, 2018-11-28:
Ok, I see your point, freeze and hang are more important. But having underpowered  cpu running below the factory specs is also an important issue in my view.

madman2003, 2018-11-28:
@krystian-hebel on which firmware version is the problem no longer reproducible?

NanoCaiordo, 2018-11-29:
@krystian-hebel while reading the debug notes I've noticed you missed to read an edited part of my post about the cpuctl kernel module.

You only need to manually load it with `kload cpuctl`

krystian-hebel, 2018-11-29:
@NanoCaiordo I did not. For some reason this module isn't present in our installation image. I keep this in mind and as soon as I'll find more time I'll try a fresh install from scratch using official images.

krystian-hebel, 2018-12-14:
I've prepared [test images](https://cloud.3mdeb.com/index.php/s/ssRQPSjYG8Ek6mD) which can help with debugging this issue (files starting with `05_`). I also managed to find some time to install pfSense with `cpuctl`, so we're starting tests on our side. We still may have some problems with reproduction, especially if this issue is connected with https://github.com/pcengines/apu2-documentation/issues/64, so any help is welcome.

NanoCaiordo, 2018-12-19:
As the main conversation is on the Reboot/Post Hang issue (apu2-documentation#64 which seems to be connected to this issue) I'm following up this report for other to read: with 05_ debug version, the issue is no longer present.

closed 2019-03-21 by miczyg1

thowe-switzerland, 2021-01-16:
I know: The problem with the CPU frequency stuck at 600MHz is an older one. But my APU2c4 still seems to suffer from this even with BIOS v4.10.0.0 from 08/08/2019.

I had the impression lately that my OPNsense firewall is very sluggish on the APU2c4. I then tried to take a look at the performance of the CPU.

**Turbostat**

> pkg add http://pkg0.isc.freebsd.org/FreeBSD:11:amd64/latest/All/turbostat-4.17_1.txz 
> rehash 
> kldload cpuctl 
> turbostat --interval 3

> turbostat version 17.06.23 - Len Brown <lenb@kernel.org>
> CPUID(0): AuthenticAMD 13 CPUID levels; family:model:stepping 0xf:30:1 (15:48:1)
> CPUID(1): SSE3 MONITOR - - - TSC MSR - -
> CPUID(6): APERF, No-TURBO, No-DTS, No-PTM, No-HWP, No-HWPnotify, No-HWPwindow, No-HWPepp, No-HWPpkg, No-EPB
> CPUID(7): No-SGX
> NSFOD /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver
> Core    CPU     Avg_MHz Busy%   Bzy_MHz TSC_MHz IRQ
>  \-      -        208     34.75   599     998     815
> 0       0       219     36.55   598     997     85
> 1       1       207     34.52   599     998     581
> 2       2       203     33.91   599     998     77
> 3       3       204     34.03   599     998     72

**Stress test also results in deep bogo ops:**

> # stress-ng --cpu 4 --cpu-method matrixprod --timeout 30 --metrics
> stress-ng: info:  [55320] dispatching hogs: 4 cpu
> stress-ng: info:  [55320] successful run completed in 30.09s
> stress-ng: info:  [55320] stressor       bogo ops real time  usr time  sys time   bogo ops/s   bogo ops/s
> stress-ng: info:  [55320]                           (secs)    (secs)    (secs)   (real time) (usr+sys time)
> stress-ng: info:  [55320] cpu                 988     30.06     75.13      0.05        32.87        13.14

**About my setup:**
- APU2c4
- BIOS v4.10.0.0 from 08/08/2019
- OPNsense 20.7.7_1-amd64
- FreeBSD 12.1-RELEASE-p11-HBSD

After a reboot, the CPU works normally for some time (hours) and also goes back to the nearly 1000MHZ. But after some time the maximum frequency is 600MHz again until the next reboot.

Since I already have a newer BIOS version: Is there anything I can do to make the CPU work normally? The performance loss is considerable.

thowe-switzerland, 2021-01-17:
To be sure that the problem is not related to the BIOS I have been using (BIOS v4.10.0.0 from 08/08/2019), I installed the latest recommended v4.11 BIOS v4.11.0.6 from 04/26/2020. I had the hope, that the stuck frequency problem could be solved. Unfortunately not. This morning the CPU is stuck at 600MHz again and the system noticeably slower.

To be clear: After applying flashrom I did a cold start. For some issue fixes in BIOS this is a must.

soder10, 2021-01-17:
> To be sure that the problem is not related to the BIOS I have been using (BIOS v4.10.0.0 from 08/08/2019), I installed the latest recommended v4.11 BIOS v4.11.0.6 from 04/26/2020. I had the hope, [...]

I think v4.13.x is the kost recent BIOS for APU

thowe-switzerland, 2021-01-17:
Yes this is true. But according to the introductory documentation above the all the BIOS downloads, the recommended version for production use is the last one of v4.11.x line.

As there have been "surprises" using very fresh BIOS versions in the past, I am normally sticking to the recommended versions. But if these latest versions fix the problem (and do not introduce new problems) then I would be willing to give v4.13.x a try.

thowe-switzerland, 2021-01-17:
Given the situation, I dared and installed the latest version v4.13.0.2, although it is still very young and I usually wait a bit before installing little used BIOS releases on production systems.

So far I haven't seen the problem with the 600MHz again. If the problem reappears even with this version, I would document it here.

thowe-switzerland, 2021-01-18:
**The problem is unfortunately not solved!**

After a few hours of operation, the problem with the limitation to a **maximum of 600MHz unfortunately reappeared** even with the latest firmware.

Hardware: APU2C4
BIOS: v4.13.0.2 (cold boot after flashrom)

The question is how the APU2 can be made to leave the economy mode of 600MHz again.

thowe-switzerland, 2021-01-20:
As the problem seems to still exist (at least under some circumstances), I have opened a new ticket: https://github.com/pcengines/coreboot/issues/457