pcengines/coreboot #198: Vanilla Coreboot on APU2?

mi2iec, 2018-09-13:
What are the current status of running vanilla mainline Coreboot (master branch) on APU2?

pietrushnic, 2018-09-14:
@mi2iec it should work without problems if you will configure it correctly, my minimum config I'm using on mainline:

```
# CONFIG_COLLECT_TIMESTAMPS is not set
CONFIG_VENDOR_PCENGINES=y
CONFIG_BOARD_PCENGINES_APU2=y
CONFIG_APU2_PINMUX_UART_C=y
CONFIG_APU2_PINMUX_UART_D=y
CONFIG_NO_GFX_INIT=y
CONFIG_SEABIOS_ADD_SERCON_PORT_FILE=y
CONFIG_PXE=y
CONFIG_BUILD_IPXE=y
CONFIG_PXE_ROM_ID="8086,157b"
# CONFIG_PXE_SERIAL_CONSOLE is not set
CONFIG_MEMTEST_SECONDARY_PAYLOAD=y
```

Please test and let us know if it works for you. Of course you will not have sortbootorder and modified SeaBIOS, so there will be some limitations. Platform also will not behave exactly the same since there is 2-3k SLOC difference between mainline and `v4.8.0.4`

mi2iec, 2018-09-17:
@pietrushnic Thanks. I'm still waiting for the system to arrive and I'm also getting a flash recovery board, so I guess that I can experiment without much danger.

closed 2018-09-17 by mi2iec

pietrushnic, 2018-09-17:
@mi2iec good luck and feel free to contribute either through bug reports or code.