pcengines/coreboot #199: apu5 doesn't detect iPXE on mainline

pietrushnic, 2018-09-15:
Please add that bug to release notes.

```
SeaBIOS (version rel-1.11.2-0-gf9626ccXHCI init on dev 00:10.0: regs @ 0xf7f22000, 4 ports, 32 slots, 32 byte contexts
XHCI    extcap 0x1 @ 0xf7f22500
XHCI    protocol USB  3.00, 2 ports (offset 1), def 0
XHCI    protocol USB  2.00, 2 ports (offset 3), def 10
XHCI    extcap 0xa @ 0xf7f22540
EHCI init on dev 00:12.0 (regs=0xf7f26020)
EHCI init on dev 00:13.0 (regs=0xf7f27020)
WARNING - Timeout at i8042_flush:71!
ATA controller 1 at 3010/3020/0 (irq 0 dev 88)
ATA controller 2 at 3018/3024/0 (irq 0 dev 88)
Found 1 lpt ports
Found 4 serial ports
Searching bootorder for: /rom@img/memtest
)
XHCI port #3: 0x00200e03, powered, enabled, pls 0, speed 3 [High]
Searching bootorder for: /pci@i0cf8/usb@12/hub@1/storage@1/*@0/*@0,0
Searching bootorder for: /pci@i0cf8/usb@12/hub@1/usb-*@1
USB MSC vendor='Multiple' product='Card  Reader' rev='1.00' type=0 removable=1
XHCI port #4: 0x00200e03, powered, enabled, pls 0, speed 3 [High]
Device reports MEDIUM NOT PRESENT
scsi_is_ready returned -1
Unable to configure USB MSC drive.
Unable to configure USB MSC device.
xhci_cmd_submit port #4: 0x000002a0, powered, pls 5, speed 0 [ - ]
xhci_alloc_pipe: address device: failed (cc -1)
XHCI port #1: 0x00001203, powered, enabled, pls 0, speed 4 [Super]
Initialized USB HUB (0 ports used)
Initialized USB HUB (0 ports used)
Searching bootorder for: /pci@i0cf8/usb@10/storage@1/*@0/*@0,0
Searching bootorder for: /pci@i0cf8/usb@10/usb-*@1
USB MSC vendor='Corsair' product='Voyager VEGA' rev='000A' type=0 removable=1
USB MSC blksize=512 sectors=30302208
WARNING - Timeout at xhci_event_wait:740!
xhci_send_pipe: xfer failed (cc -1)
All threads complete.
Scan for option roms

Press ESC for boot menu.

Select boot device:

1. USB MSC Drive Corsair Voyager VEGA 000A
2. Payload [memtest]
```

miczyg1, 2018-09-18:
@pietrushnic previously we had apu5a in our setup which had Intel I211 NICs. Currently apu5b have Intel I210 NICs (the same ones as on apu2). So basically it changes PCI ID between 8086:1539 and 8086:157b. I will ask vendor what is their plan with NICs

pietrushnic, 2018-09-24:
@miczyg1 didn't saw communication about that.

soder10, 2018-10-20:
wait, you guys even manufacture apu4/apu5 boards? Why no info about these on pcengines.ch site?

pietrushnic, 2018-10-20:
@soder10 this account is maintained by 3mdeb team not PC Engines. [apu4](http://pcengines.ch/apu4b4.htm) is available on the website and you can buy it. [apu5](http://pcengines.info/forums/?page=post&id=7D919FB8-C4BB-4C33-B47B-F8F53E01600D&fid=DF5ACB70-99C4-4C61-AFA6-4C0E0DB05B2A) is not available and behind the link, you can find an explanation from PC Engines owner.

soder10, 2018-10-24:
Thanks for the info. These APU board name selections are very confusing.

closed 2021-10-18 by damiankaras