pcengines/coreboot #205: APU1 v4.8.0.4 breaks serial console

ShinjiLE, 2018-09-24:
Flash to V4.8.0.4. But Serial output is broken (System and PXE) . Only Identification and Menu is printed out and after this no further output. But OS and PXE is booted.
Reflash with a SPI-programmer to V4.8.0.3. all is fine. After this i try V4.8.0.4. a second time but the same result.

Output of V4.8.0.4. ( yes it  prints out V4.8.0.3)
PC Engines apu1ress ENTER to boot or TAB to edit a menu entry
coreboot build 20183108
BIOS version v4.8.0.3

### PC Engines apu1 setup v4.6.9 ###
Boot order - type letter to move device to top.

  a USB 1 / USB 2 SS and HS 
  b SDCARD 
  c mSATA 
  d SATA 
  e mPCIe1 SATA1 and SATA2 
  f iPXE (disabled)

  r Restore boot order defaults
  n Network/PXE boot - Currently Disabled
  u USB boot - Currently Enabled
  t Serial console - Currently Enabled
  o UART C - Currently Enabled
  p UART D - Currently Enabled
  w Enable BIOS write protect - Currently Disabled
  x Exit setup without save
  s Save configuration and exit

pietrushnic, 2018-09-24:
@ShinjiLE thank you for reporting that issue.
@miczyg1 let's try to reproduce that and if reasonably simple to fix let's try to include that in `v4.8.0.5` if not we will have to schedule that for `v4.8.0.6`

miczyg1, 2018-09-25:
@ShinjiLE thank You for reporting the issue. There is a problem with serial console on apu1 and the BIOS version indeed. We will fix it in v4.8.0.5.

ShinjiLE, 2018-10-16:
Ahoi.
It seems that v4.8.0.5 is released now, should i try it again ?

miczyg1, 2018-10-16:
Hello, @ShinjiLE we kindly encourage You to verify if the bootorder works on apu1. Serial consoel and BIOS version should be fixed too. Please provide feedback, it will be appreciated.

ShinjiLE, 2018-10-17:
Good work ! All fine now.

closed 2018-10-17 by ShinjiLE

pietrushnic, 2018-10-17:
@ShinjiLE thanks for verification.