pcengines/coreboot #206: apu2 v4.8.0.4 breaks WLE200NX wifi cards

muzy, 2018-09-30:
Hi,

I just found out that by flashing the apu2 v4.8.0.4 bios to my APU2C4, my Compex WLE200NX wifi cards did no longer broadcast any SSID. Unfortunately there is no error in the kernel log or the hostapd debug messages.

When switching to apu2 v4.6.9 bios, the wifi cards did work again.

Any ideas on how to debug this?

miczyg1, 2018-10-01:
@muzy what operating system are You using? It would be better to stick to Your configuration as close as possible. Which slot was the card plugged in, mPCIe 1 or mPCIe 2?

muzy, 2018-10-01:
I was finally able to debug a more and conducted the following test matrix on multiple APU2C4 devices with two Compex WLE200NX in mPCIe 1 and mPCIe2, booted from SD card:

| Coreboot | Debian 9.5 | OpenWRT 18.06.1 |
| ------------- | ------------- |----------|
| 4.8.0.4  | not working  | working |
| 4.6.10| working  | working |

**Software Versions are as follows:**
**Debian 9.5**: hostapd 2.4
**OpenWRT 18.06.1**: hostapd 2.7

Output of _lspci_ for wifi cards:
```
01:00.0 Network controller: Qualcomm Atheros AR928X Wireless Network Adapter (PCI-Express) (rev 01)
05:00.0 Network controller: Qualcomm Atheros AR928X Wireless Network Adapter (PCI-Express) (rev 01)
```

While conducting the tests I noted a difference in the hostapd debug output on the Debian 9.5. The only thing changed between tests is Coreboot version. The hostapd configuration for reproducing the test result is as follows:

```
beacon_int=100
ssid=TEST
interface=wlp5s0
driver=nl80211
channel=6
ctrl_interface=/var/run/hostapd_wlp5s0
ctrl_interface_group=0
ignore_broadcast_ssid=0
ap_isolate=0
hw_mode=g
ieee80211n=1
ht_capab=[HT20]
wpa=2
wpa_passphrase=testtest123
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP
bridge=br0
```

This is the output from both test runs on Debian 9.5 (Click to toggle output). It appears that hostapd stops with the 4.8.0.4 Coreboot while waiting for some sort of command. 

<details><summary>

**hostapd with 4.8.0.4**
</summary>
<p>

```
random: Trying to read entropy from /dev/random
Configuration file: hostapd.conf
ctrl_interface_group=0
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
nl80211: TDLS supported
nl80211: TDLS external setup
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Using driver-based off-channel TX
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: interface wlp5s0 in phy phy1
[... 529 lines cut ...]
nl80211: send_mlme - da= da:a1:19:81:5b:5e noack=1 freq=0 no_cck=0 offchanok=0 wait_time=0 fc=0x50 (WLAN_FC_STYPE_PROBE_RESP) nlmode=3
nl80211: send_mlme -> send_frame
nl80211: send_frame - Use bss->freq=2437
nl80211: send_frame -> send_frame_cmd
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlp5s0
nl80211: RX frame sa=da:a1:19:81:5b:5e freq=2437 ssi_signal=-37 fc=0x40 seq_ctrl=0x3360 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=82
Ignore Probe Request due to DS Params mismatch: chan=6 != ds.chan=7
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlp5s0
nl80211: RX frame sa=da:a1:19:81:5b:5e freq=2437 ssi_signal=-38 fc=0x40 seq_ctrl=0x3370 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=82
Ignore Probe Request due to DS Params mismatch: chan=6 != ds.chan=7
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlp5s0
nl80211: RX frame sa=da:a1:19:81:5b:5e freq=2437 ssi_signal=-38 fc=0x40 seq_ctrl=0x3380 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=82
Ignore Probe Request due to DS Params mismatch: chan=6 != ds.chan=7
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlp5s0
nl80211: RX frame sa=ea:0e:14:21:d3:cb freq=2437 ssi_signal=-76 fc=0x40 seq_ctrl=0x2710 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=94
nl80211: send_mlme - da= ea:0e:14:21:d3:cb noack=1 freq=0 no_cck=0 offchanok=0 wait_time=0 fc=0x50 (WLAN_FC_STYPE_PROBE_RESP) nlmode=3
nl80211: send_mlme -> send_frame
nl80211: send_frame - Use bss->freq=2437
nl80211: send_frame -> send_frame_cmd
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlp5s0
```
</p>
</details>

If I can be of any assistance while debugging please let me know. I'd be happy to help.

miczyg1, 2018-10-02:
@pietrushnic it looks so. @muzy try the mPCIe2 CLK quirk in sortbootorder. We will try to figure something out on our side too.

muzy, 2018-10-02:
@miczyg1 @pietrushnic I tried the mPCIe2 CLK option and left everything else as it was before (Debian 9.5, hostapd config). The observable effects are the same (the log file output from hostapd is the same).

pietrushnic, 2018-10-03:
@muzy we have to reproduce that on our side any simple guide or maybe ready to use system images that we can use? @miczyg1 I'm not sure what would be best approach here, but maybe some debug build with spew logging and then trying to compare 2 boots?

muzy, 2018-10-04:
@pietrushnic I've uploaded a copy of the image here: https://ufile.io/2mmjv 

You need to unzip the image with gzip/gunzip and then copy the image to an SD card with at least 16GB using the _dd_ utility.

```
SHA1 hash of the image (unzipped): 4182a222f302dddef2bf81601a9567cf69cfc68e
User/PW: root/test123
```

pietrushnic, 2018-10-05:
@miczyg1 let's try that next week.

miczyg1, 2018-11-06:
@muzy I have finally found some free cycles to look into the issue. I have burned Your image onto SD card, but I face little problems from the start:
- I have only 1 WLE200NX (rather not significant)
- Although module is listed in lspci and dmesg says that wlan0 has been renamed to wlp1s0, it is not listed in ifconfig (hostapd fails to run the provided configuration wlp1s0.conf), this is rather unexpected right?
- I see 3 bridges and some docker over there, do they have some influence? What is their purpose?

I would appreciate any advices since I am not aware of all modifications and settings You have made to the system.

muzy, 2018-11-06:
@miczyg1 what you see is very unexpected. I'll check and reach out. The bridges should have no influence as they are used for docker only.

miczyg1, 2018-11-07:
@muzy I will try preparing a fresh installation with hostapd only and copy Your configs. Hope to get some results.

miczyg1, 2018-11-07:
@muzy I guess I have found the issue. The wireless interface is not going up automatically after boot. I have to explicitly call `ifconfig wlpXsX up`. I will workaround lack of second wifi card somehow. Could You please tell me how the bridges are created (which interfaces are bridged)? I would like to reproduce Your configuration to make it as close to Yours as possible.

miczyg1, 2018-11-07:
@muzy I managed to get SSID broadcasted on WLE200NX with v4.8.0.4 and v4.6.10. It looks like AMD IOMMU causes some malfunctions, it was introduced in v4.8.0.3. As a mere workaround You can add `amd_iommu=off` to kernel cmdline to turn it off. After that my phone started to discover the TEST wifi network on v4.8.0.4.

Please note that IOMMU is quite fresh feature on PC Engines platforms and may not yet be perfectly initialized.

miczyg1, 2018-11-08:
Related issue: https://github.com/pcengines/coreboot/issues/200

miczyg1, 2018-11-29:
@muzy did the `amd_iommu=off` help?

miczyg1, 2019-02-26:
@muzy if the quirk did not help for You, feel free to reopen the issue. Closing it for now.

closed 2019-02-26 by miczyg1

chr0mag, 2019-07-11:
I'm not sure this merits reopening (or even if it's something fixable by PC Engines) but after adding `amd_iommu=off ` to workaround this issue, I see the following stack trace in dmesg on boot.

My device recovers and boots normally and turning off iommu does allow me to see my WLE200NX SSID again.

```
WARNING: CPU: 0 PID: 0 at kernel/workqueue.c:2771 flush_workqueue+0x3ca/0x420
Jul 10 20:49:39 apu2c4 kernel: Modules linked in:
Jul 10 20:49:39 apu2c4 kernel: CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.1.16-arch1-1-ARCH #1
Jul 10 20:49:39 apu2c4 kernel: Hardware name: PC Engines apu2/apu2, BIOS v4.9.0.5 05/09/2019
Jul 10 20:49:39 apu2c4 kernel: RIP: 0010:flush_workqueue+0x3ca/0x420
Jul 10 20:49:39 apu2c4 kernel: Code: ff ff ff 4c 89 e7 48 89 0c 24 e8 51 f1 ff ff 41 8b 44 24 44 41 8b 74 24 40 48 8b 0c 24 e9 f8 fe ff ff 4c 89 fb e9 77 fe ff ff <0f> >
Jul 10 20:49:39 apu2c4 kernel: RSP: 0000:ffffffffac003d60 EFLAGS: 00010246
Jul 10 20:49:39 apu2c4 kernel: RAX: ffffffffac003d98 RBX: ffff8cdeea439980 RCX: 0000000000000000
Jul 10 20:49:39 apu2c4 kernel: RDX: ffffffffac67bc48 RSI: ffffffffabe98042 RDI: ffffffffac003d90
Jul 10 20:49:39 apu2c4 kernel: RBP: ffffffffac003de0 R08: 0000000000000000 R09: 0000000000000000
Jul 10 20:49:39 apu2c4 kernel: R10: ffffebc1c4a95608 R11: 0000000000000000 R12: 0000000000000000
Jul 10 20:49:39 apu2c4 kernel: R13: ffffffffac003d70 R14: 0000000000000501 R15: 0000000000010c01
Jul 10 20:49:39 apu2c4 kernel: FS:  0000000000000000(0000) GS:ffff8cdeeaa00000(0000) knlGS:0000000000000000
Jul 10 20:49:39 apu2c4 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 10 20:49:39 apu2c4 kernel: CR2: ffff8cdeeefff000 CR3: 000000003420e000 CR4: 00000000000406b0
Jul 10 20:49:39 apu2c4 kernel: Call Trace:
Jul 10 20:49:39 apu2c4 kernel:  kmem_cache_destroy+0x5b/0x200
Jul 10 20:49:39 apu2c4 kernel:  ? __free_pages_ok+0x156/0x2c0
Jul 10 20:49:39 apu2c4 kernel:  free_iommu_resources+0x5a/0x191
Jul 10 20:49:39 apu2c4 kernel:  iommu_go_to_state+0xafd/0x12ef
Jul 10 20:49:39 apu2c4 kernel:  amd_iommu_prepare+0x16/0x2a
Jul 10 20:49:39 apu2c4 kernel:  irq_remapping_prepare+0x36/0x5f
Jul 10 20:49:39 apu2c4 kernel:  enable_IR_x2apic+0x21/0x188
Jul 10 20:49:39 apu2c4 kernel:  default_setup_apic_routing+0x12/0x6f
Jul 10 20:49:39 apu2c4 kernel:  apic_intr_mode_init+0x15c/0x197
Jul 10 20:49:39 apu2c4 kernel:  x86_late_time_init+0x17/0x1c
Jul 10 20:49:39 apu2c4 kernel:  start_kernel+0x47c/0x530
Jul 10 20:49:39 apu2c4 kernel:  secondary_startup_64+0xa4/0xb0
Jul 10 20:49:39 apu2c4 kernel: ---[ end trace 1d30ed0ae73a58ff ]---
```

miczyg1, 2019-07-11:
@chr0mag I saw similar issues on different AMD platform running coreboot and kernel >= 4.19.x. It rather qualifies for a new issue so I would prefer You open a new one. Also provide some more detailed infornation: platform name, BIOS version, operating system version, extensuon cards used, which slots are occupied, the boot medium type (USB, SATA...) etc.

chr0mag, 2019-07-11:
I've opened issue #323 for this.