pcengines/coreboot #212: Wish: support BIOS output on alternate serial port

jernejs, 2018-10-18:
Since COM1 may be used for other things, it would be nice if it was possible to redirect BIOS output to COM2 (instead of just having an option to completely disable BIOS output). Would this be possible (I wouldn't mind even if a separate BIOS version would need to be flashed that would have its output hardcoded to COM2 header).

pietrushnic, 2018-10-18:
@miczyg1 isn't just configuration?

miczyg1, 2018-10-22:
@pietrushnic  it may be configuration, but it is not simple as it sounds. SeaBIOS serial console configuration is a file in CBFS, so the firmware has to be already prebuilt to support COM2. Runtime configuration of that feature is a lot of trouble or rather will require bunch of custom modifications. Serial output in coreboot will require digging into SuperIO code as well. However I will try to schedule it for v4.8.0.6.

miczyg1, 2018-10-22:
@pietrushnic it is not just configuration. It just sounds simple. SeaBIOS console output is configured via CBFS file generated during build. So the firmware must be built to support COM2 as output. Another thing is coreboot output will require digging in SuperIO code to enable it. However I will try to schedule it for v4.8.0.6.

michaelsteinmann, 2018-10-23:
> @jernejs <https://github.com/jernejs> unfortunately in light of above
> analysis the only way it can be addressed are:
>
>    1. Gather a bigger audience to push the features or convince PC
>    Engines that this is important features that firmware developers should
>    spend time on
>    2. Contribute patches
>    3. Sponsor development directly or through the company you work in. Of
>    course, if there is any business value in that feature.
>
> I'm sorry, but those are the only options I can offer now. Maybe in
> future, we will get back to this topic, but cannot promise anything.

Could you generate a binary with com2 for console output instead of com1?
A customer of ours is asking for that too.

miczyg1, 2018-10-23:
In this case I will prioritize implementing the feature, so it will be available in v4.8.0.6 release.

istamov, 2018-10-25:
I can also support the request for such feature, in my case I would like COM1 to be used for connecting external UPS via nut-tools on pfSense (since most UPS devices have RS232 levels COM2 can't be used directly).
I also don't mind having to use different BIOS binary for the COM2 output, and if we can help with something (testing/experimenting), please just say so. Thanks!

miczyg1, 2018-10-25:
@istamov change is already prepared. Will be available in v4.8.0.6 along with guide how to build the binary Yourself. I think we will distribute also the binaries with COM2 this time, because it is not yet ready for runtime configuration.

istamov, 2018-10-25:
@miczyg1 Thanks, I already saw pull request https://github.com/pcengines/coreboot/pull/214 and am looking forward to test it, when it is merged in v4.8.0.6.
Any chance to also backport it to the legacy v4.0.20 too? I find it quite stable so far but if it is too much of a hassle, this will be just one more reason to migrate to the mainline v4.8.

miczyg1, 2018-10-25:
@istamov yes, this is on our TODO list. v4.0.20 was already released, but will try to include it for v4.0.21.

miczyg1, 2018-11-09:
[apu2_com2_v4.0.21](https://cloud.3mdeb.com/index.php/s/ShUvqIZ2GFrwIOC/download)
[apu2_com2_v4.8.0.6](https://cloud.3mdeb.com/index.php/s/ETVv4IEllzv1bLy/download)
[apu3_com2_v4.0.21](https://cloud.3mdeb.com/index.php/s/5wY4z9pdOjVzstl/download)
[apu3_com2_v4.8.0.6](https://cloud.3mdeb.com/index.php/s/SLzrYP8sJw1Tcea/download)
[apu4_com2_v4.0.21](https://cloud.3mdeb.com/index.php/s/naCro8dKUe4pVEC/download)
[apu4_com2_v4.8.0.6](https://cloud.3mdeb.com/index.php/s/wrmJQO9x97VH9Lv/download)

These are binaries with console output redirection to COM2 in firmware. Please report any issues or suggestions, what can be improved. I already realize that it would be great to have it in runtime configuration.

There is also a guide how to build the binaries manually: https://github.com/pcengines/apu2-documentation/blob/master/docs/serial_console.md

istamov, 2018-11-09:
@miczyg1, thank you!
I have just flashed v4.0.21 on APU2C4 remotely, but unfortunately lost access to it and didn't have remote console attached. Will try to recover it today, test it over the weekend and will report back.

miczyg1, 2018-11-09:
@istamov of course, the console changed so there is no output on COM1 anymore (could be from OS only if console configured to ttyS0).

istamov, 2018-11-09:
@miczyg1: Sorry, I meant that I lost access to the whole device (it is a test pfSense router) - when rebooted the lights (NIC indicators) are blinking, but the pfSense is unreachable, so will have to check on site what is going on (it is possible to be something pfSense-related).

istamov, 2018-11-09:
@miczyg1: Under Ubuntu 16.04.3 it works just perfect - both BIOS and operating system terminal are working on COM2 and COM1 is "free" for another applications.

Under pfSense 2.4.4 it seems to hang at some point - the NIC LEDs are blinking, but the terminal is not responsive (please see below where it stops):

- Successful boot, BIOS v4.0.20, console on COM1 (comconsole_port="0x3F8" >> /boot/loader.conf.local)
```
/boot/kernel/kernel text=0x17bf9f0 data=0xb93c38+0x517b18 syms=[0x8+0x197280+0x8+0x197de9]
/boot/entropy size=0x1000
/boot/kernel/zfs.ko size 0x381f00 at 0x2f9c000
loading required module 'opensolaris'
/boot/kernel/opensolaris.ko size 0xa3a0 at 0x331e000
Booting...
KDB: debugger backends: ddb
KDB: current backend: ddb
Copyright (c) 1992-2018 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the University of California. All rights reserved.
FreeBSD is a registered trademark of The FreeBSD Foundation.
FreeBSD 11.2-RELEASE-p3 #17 e6b497fa0a3(RELENG_2_4_4): Thu Sep 20 09:04:45 EDT 2018
    root@buildbot3:/crossbuild/ce-244/obj/amd64/WvDslnYb/crossbuild/ce-244/pfSense/tmp/FreeBSD-src/sys/pfSense amd64
FreeBSD clang version 6.0.0 (tags/RELEASE_600/final 326565) (based on LLVM 6.0.0)
VT(vga): resolution 640x480
CPU: AMD GX-412TC SOC                                (998.16-MHz K8-class CPU)
  Origin="AuthenticAMD"  Id=0x730f01  Family=0x16  Model=0x30  Stepping=1
  Features=0x178bfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,MMX,FXSR,SSE,SSE2,HTT>
  Features2=0x3ed8220b<SSE3,PCLMULQDQ,MON,SSSE3,CX16,SSE4.1,SSE4.2,MOVBE,POPCNT,AESNI,XSAVE,OSXSAVE,AVX,F16C>
  AMD Features=0x2e500800<SYSCALL,NX,MMX+,FFXSR,Page1GB,RDTSCP,LM>
  AMD Features2=0x1d4037ff<LAHF,CMP,SVM,ExtAPIC,CR8,ABM,SSE4A,MAS,Prefetch,OSVW,IBS,SKINIT,WDT,Topology,PNXC,DBE,PTSC,PL2I>
  Structured Extended Features=0x8<BMI1>
  XSAVE Features=0x1<XSAVEOPT>
  SVM: NP,NRIP,AFlush,DAssist,NAsids=8
  TSC: P-state invariant, performance statistics
real memory  = 4815060992 (4592 MB)
avail memory = 4067495936 (3879 MB)
Event timer "LAPIC" quality 600
ACPI APIC Table: <CORE   COREBOOT>
FreeBSD/SMP: Multiprocessor System Detected: 4 CPUs
FreeBSD/SMP: 1 package(s) x 4 core(s)
ioapic1: Changing APIC ID to 5
ioapic0 <Version 2.1> irqs 0-23 on motherboard
ioapic1 <Version 2.1> irqs 24-55 on motherboard
SMP: AP CPU #3 Launched!
SMP: AP CPU #1 Launched!
SMP: AP CPU #2 Launched!
Timecounter "TSC" frequency 998159803 Hz quality 1000
ipw_bss: You need to read the LICENSE file in /usr/share/doc/legal/intel_ipw.LICENSE.
ipw_bss: If you agree with the license, set legal.intel_ipw.license_ack=1 in /boot/loader.conf.
module_register_init: MOD_LOAD (ipw_bss_fw, 0xffffffff80680430, 0) error 1
random: entropy device external interface
ipw_ibss: You need to read the LICENSE file in /usr/share/doc/legal/intel_ipw.LICENSE.
ipw_ibss: If you agree with the license, set legal.intel_ipw.license_ack=1 in /boot/loader.conf.
module_register_init: MOD_LOAD (ipw_ibss_fw, 0xffffffff806804e0, 0) error 1
ipw_monitor: You need to read the LICENSE file in /usr/share/doc/legal/intel_ipw.LICENSE.
ipw_monitor: If you agree with the license, set legal.intel_ipw.license_ack=1 in /boot/loader.conf.
module_register_init: MOD_LOAD (ipw_monitor_fw, 0xffffffff80680590, 0) error 1
wlan: mac acl policy registered
hn: tranparent VF mode, if_transmit will be used, instead of if_start
kbd0 at kbdmux0
netmap: loaded module
module_register_init: MOD_LOAD (vesa, 0xffffffff81209800, 0) error 19
nexus0
vtvga0: <VT VGA driver> on motherboard
cryptosoft0: <software crypto> on motherboard
padlock0: No ACE support.
acpi0: <CORE COREBOOT> on motherboard
acpi0: Power Button (fixed)
cpu0: <ACPI CPU> on acpi0
cpu1: <ACPI CPU> on acpi0
cpu2: <ACPI CPU> on acpi0
cpu3: <ACPI CPU> on acpi0
[... 232 lines cut ...]
module_register_init: MOD_LOAD (vesa, 0xffffffff81209800, 0) error 19
nexus0
vtvga0: <VT VGA driver> on motherboard
cryptosoft0: <software crypto> on motherboard
padlock0: No ACE support.
acpi0: <CORE COREBOOT> on motherboard
acpi0: Power Button (fixed)
cpu0: <ACPI CPU> on acpi0
cpu1: <ACPI CPU> on acpi0
cpu2: <ACPI CPU> on acpi0
cpu3: <ACPI CPU> on acpi0
atrtc0: <AT realtime clock> port 0x70-0x71 irq 8 on acpi0
atrtc0: registered as a time-of-day clock, resolution 1.000000s
Event timer "RTC" frequency 32768 Hz quality 0
attimer0: <AT timer> port 0x40-0x43 irq 0 on acpi0
Timecounter "i8254" frequency 1193182 Hz quality 0
Event timer "i8254" frequency 1193182 Hz quality 100
Timecounter "ACPI-safe" frequency 3579545 Hz quality 850
acpi_timer0: <32-bit timer at 3.579545MHz> port 0x818-0x81b on acpi0
hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Timecounter "HPET" frequency 14318180 Hz quality 950
acpi_button0: <Power Button> on acpi0
pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0
pci0: <ACPI PCI bus> on pcib0
```

miczyg1, 2018-11-09:
@istamov I am not FreeBSD/pfSense expert, but I do not see any reason it shouldn't work with pfSense. Try setting COM2 port whenerver possible. The console ioport should be 0x2f8. Regarding the hang on pci0 string, we noticed it in v4.0.19. It seems like legacy build is not reproducible, because in v4.0.20 the issue disappeared automagically without any intervention from our side... As a workaround, plug a USB stick, and it will boot probably. Try building the firmware Yourself and flash it, i have added a guide how to build it with COM2.

istamov, 2018-11-13:
@miczyg1: You are correct, when I plug in a USB stick it does boot (as described in this [thread](http://pcengines.info/forums/?page=post&id=4C472C95-E846-42BF-BC41-43D1C54DFBEA)).
I tried to build the firmware by myself and the result is the same - it does boot under Ubuntu but hangs on PCI initialisation under pfSense until USB stick is present. I will try to see if I can find any reason for that (but I doubt it, given the fact that you are much more experienced in coreboot/seaBIOS than me).
For the moment Ubuntu works just fine and I believe BIOS/console on COM2 will be useful for other people too, so thank you very much, Michal!

istamov, 2018-11-13:
Actually I just tried with another spare APU2C4 and updated it to pfSense 2.4.5-DEVELOPMENT (FreeBSD 11.2-RELEASE-p4) - and it boots without a USB stick attached! Did at least 8 consecutive reboots and all were successful 😃
So my guess is that the pci0 problem is caused by something in pfSense 2.4.4 (FreeBSD 11.2-RELEASE-p3) which is fixed in FreeBSD 11.2-RELEASE-p4. We will have to wait for the official 2.4.5 release (probably Q1 2019), but it seems to be working.

The only suggestion I can add you already know of - if there is a possibility to add runtime configuration of the console output redirection in the future releases, it would be great! But as it is at the moment is also very useful, so thank you once again!

miczyg1, 2019-02-26:
@istamov the problem with pfSense booting was related to EHCI0 controller. We have fixed the issue in v4.0.23: https://github.com/pcengines/coreboot/blob/coreboot-4.0.x/CHANGELOG.md#fixed

EHCI0 on apu2 is not routed to an external header on the PCB like it is done on apu3/4, thus we have permanently disabled EHCI0 for apu2 to avoid such issues in the future.

Regarding the COM2 console output, it has been implemented in runtime configuration since v4.0.22 and v4.8.0.7. The only issue is that memtest86+ works on COM1 (since it is compiled with hardcoded serial configuration). All limitations and how to build fully supported COM2 redirection is available here:

https://github.com/pcengines/apu2-documentation/blob/master/docs/serial_console.md

Visit also https://pcengines.github.io for newest releases and information about firmware. We have added binary signatures, blog and newsletter for PC Engines firmware.

closed 2019-02-26 by miczyg1