pcengines/coreboot #219: HUAWEI modem detection on pfSense

artur-rs, 2018-11-09:
Problem with HUAWEI modem detection (hardware suite) - coreboot regression v4.8.0.6 (apu3 with pfSense):
1st iteration (PASS): 
```
: dmesg | grep -A 1 HUAWEI
ugen1.5: <HUAWEI Technology HUAWEI Mobile> at usbus1
u3g0 on uhub2
u3g0: <HUAWEI Mobile Connect - Modem> on usbus1
u3g0: Found 5 ports.
[2.4.2-RELEASE][root@pfSense.localdomain]/root
```
2nd iteration (FAIL):
```
: dmesg | grep -A 1 HUAWEI
[2.4.2-RELEASE][root@pfSense.localdomain]/root
```

There is possibility that enabled IOMMU affects the detection of hardware:
https://github.com/pcengines/coreboot/issues/200
https://github.com/pcengines/coreboot/issues/206

pietrushnic, 2019-01-12:
@artur-rs  this looks like a regression, we should retest v4.0.22 to check if this is new or we had a problem with tests previously.

miczyg1, 2019-02-11:
@artur-rs has the issue been resolved?

artur-rs, 2019-02-11:
@pietrushnic @miczyg1 re-tests on the version `v4.8.0.6`, `v4.0.23` and newer doesn't reproduce the bug anymore, I think we should close the issue.

closed 2019-02-11 by miczyg1

theliquid, 2019-06-20:
Concerning Huawei Modem detection:
I have an apu3b4 with 2 Huawei ME909s-120 4G modems with pfsense 2.4.4-RELEASE-p3.
Mainline bios versions never worked for me.
I always had to use Legacy releases. Currently I am at version v4.0.18.

Today I wanted to try out the newer mainline releases.
I tried v4.9.0.6. With this version the second 4G modem connected to SIM2 doesn't get recognized.
Then I went back to Legacy releases. Sadly, the same issue with versions v4.0.26, v4.0.25, v4.0.23.
The problem is that pfsense doesn't even recognize the modem, it doesn't show up under usbconfig.

I didn't try v4.0.19, v4.0.21, v4.0.22 and v4.0.24 due the known issues. 

On v4.0.20 the second modem gets recognized (visible with usbconfig), but both modems are running with wrong USB speed (12M); the second one doesn't show up in the PPP logs.

So I went back to v4.0.18 which is the last working bios version for my setup :(

michaelsteinmann, 2019-06-20:
did you enable the EHCI0 controller in the BIOS settings?

theliquid, 2019-06-20:
> 
> […](#)

Thanks for the fast reply. No, do I have to?

michaelsteinmann, 2019-06-20:
yes, that's needed to enable the USB block inside the CPU which is
connected to the miniPCIe2 / J15.

theliquid, 2019-06-20:
Ok, thank you very much!
After flashing v4.0.26 again and manually enabling EHCI0 controller in BIOS settings both modems got recognized! :-)
v4.0.18 had EHCI0 controller enabled by default in BIOS settings.

miczyg1, 2019-06-20:
@theliquid does modem work with v4.9.0.6 an EHCI0 enabled?

theliquid, 2019-06-20:
@miczyg1 I just tried that. It is working for now - I also manually enabled EHCI0 controller. 
But sadly, most of the issues I always had with mainline BIOS versions occurred after some time. I'll stick with v4.9.0.6 and report back after some time.

miczyg1, 2019-06-20:
Ok, great. Thank You

theliquid, 2019-06-26:
After a couple of days no issues so far with this mainline BIOS version. The only issue, which I think is hardware related since I also had it on Legacy BIOS releases, is that after disabling a 4G modem it doesn't get recognized when it is enabled again in pfsense. To solve this a reboot is needed.