pcengines/coreboot #227: Upload releases on LVFS

paravoid, 2018-11-28:
[Linux Vendor Firmware Service (LVFS)](https://fwupd.org/) is a service for vendors to upload firmware for their hardware. The service goes hand-in-hand with (and is maintained by the author of) `fwupd`, a tool that scans your hardware and searches for firmware updates. A lot of the mainstream vendors use it and upload their firmware nowadays (e.g. Dell, Lenovo, Logitech etc.). fwupd is integrated iin desktop environments to provide "software updates" integration for firmware (e.g. in GNOME Software) but a command-line version exists and also works without any DE.

fwupd has a [flashrom](https://github.com/hughsie/fwupd/tree/master/plugins/flashrom) plugin, but not a lot of vendors that support it thus far (see hughsie/fwupd#805 for upstream's take). It'd be neat if PCEngines was one of the first ones there :) 

The software is still a bit clunky for headless systems, embedded systems and servers IMHO as it brings a lot of non-optional extra dependencies. For example the version in Debian sid has a dependency tree that wants to bring in libX11, but AIUI that has been addressed upstream. More broadly, fwupd is still not very useful nor popular in servers, but I think this may be a chicken-and-egg thing that also has to do with firmware availability in LVFS. Having a vendor like you -and especially one that is as open as you are!- maintain their firmware in LVFS may be a good step in the right direction, and be immediately useful to some users :)

(I'm not affiliated in any way with the fwupd project)

pietrushnic, 2018-11-29:
@paravoid thanks for your proactive action, but we are already on the same page:

1. PC Engines firmware maintaners are not PC Engines employees (we are [3mdeb](https://3mdeb.com))
2. There are some concerns about fwupd theory of operation that I have to discuss with PC Engines directly to push the effort, but your issue will definitely help me. Also if anyone feel the same please show your appreciation for the above issue and this will give me a better argument in a discussion.
3. If you will take a look [here](https://github.com/3mdeb/fwupd/pulls) and [here](https://www.mail-archive.com/flashrom@flashrom.org/msg14016.html) you will realize that we already working on this solution for [Librebox](https://libretrend.com/) - in that way 3mdeb will be probably first Open Source Firmware provider who can enable support for  LVFS.
4. Stay tuned and feel free to contribute.

paravoid, 2018-11-29:
This is awesome -- thank you so much!

soder10, 2018-11-30:
@pietrushnic: I think most visitors of this github site are not aware that it is not maintaned by pcengines company, but maintaned by a different company, yours.

I think this confusion comes from the fact, that
1) this site is called github.com/pcengines and not github.com/3mdeb
2) people bought their APU board from the company PCENGINES, or from a reseller/distributor, who represents PCENGINES. 3mdeb, as a company is normally not mentioned anywhere around time of purchase. PCENGINES should make it very clear, that they are selling a product, that itself is emptry hardware without any kind of burnt-in software at all! Including no BIOS. So their hardware is plain useless without BIOS, and they simply dont deal with that BIOS, but 3mdeb (you) do. And since most of the issues that endusers face is first blamed on software side (difficult to say on 1st look if its really a HW issue). So this site (run by 3mdeb) becomes the primary contact area.

Strange though, that PCENGINES company has so carefully removed themselves from activities happening here.

michaelsteinmann, 2018-11-30:
@Piotr Król <piotr.krol@3mdeb.com> perhaps we should add an 'about' - tab
to the top of https://pcengines.github.io/ like this:
| pfSense | Pxelinux | Regression Test Results | About |

It should mention that 3mbed is maintaining this page on behalf of PC
Engines.

Best regards,
Michael Steinmann

pietrushnic, 2018-11-30:
@soder10 I understand confusion, but it is easy to click profile and see what organization people belong to, if someone really wants. I don't think it matter much. I'm repeating that phrase mainly because of powers and influence on products, that people assign to us. To influence PC Engines, users should write directly to them or try to express things on forum.

3mdeb is paid by PC Engines to do firmware maintenance and validation. So it is our responsibility to provide quality firmware and if something in firmware doesn't work it is most probably 3mdeb fault, lack of resources, expertise, knowledge or binary blob issues caused by much bigger players.

To address your points:

1. Yes, it is called github.com/pcengines because we believe that great open hardaware deserve open firmware development and debugging community. Copyrights of work made in this repository belong to PC Engines since this is work for hire.
2. This is not true. PC Engines ships board with working firmware. It may be not the recent one, but still it is working firmware. Please consider that firmware validation is hard and consume a lot resources. Either you get reasonable good firmware with great hardware for decent price, either you pay 3-4x of that price and demand much higher firmware quality. To be honest we should be glad that PC Engines decide to have firmware maintainers, since there are hundreds of vendors that don't care about open or even maintaining their closed source stuff. We should try to look more how PC Engines compare to other vendors and terrible firmware quality other vendors provide.

@soder10 I invite you to join coreboot community conference like ELCE/OSFV to see how community look like and how it operates, maybe then judgment will be more realistic. I would be glad to meet and talk directly.

@michaelsteinmann Yes, we prepare productize service `coreboot maintainership` that we want to link on https://pcengines.github.io

pietrushnic, 2018-12-01:
@soder10 ok I understand your honest intention, but I can't understand your assumptions about delivering hardware without firmware. PC Engines hardware is delivered with firmware. Firmware delivered makes hardware usable out of the box. 3mdeb tries to improve PC Engines firmware using an iterative approach and keeping things close to mainline. If you will analyze other coreboot based hardware you also cannot find `vendor_name` firmware on board e.g. Purism firmware.

I wrote it somewhere, but truly IMO firmware should be updated very rarely - ideally never. Out of the box firmware should be enough for 95% of use cases. So if someone updates the firmware on PC Engines first take the risk by relying on 3mdeb validation capabilities and truly have some reason for that.

3mdeb cannot say it created whole PC Engines firmware since it is coreboot and other open sources projects involved. There were also other companies that managed PC Engines firmware before 3mdeb. All firmware 3mdeb developed has PC Engines copyrights.

I can ramble a lot about coreboot vs other firmware, but please analyze hardware you have in home and office. 95% of that hardware doesn't get firmware upgrades. Most x86 hardware vendors are offshore OEM who sell buggy firmware and doesn't support updates at all. PC Engines definitely has a different approach here. I hope to add fwupd/LVFS support to make upgrades of firmware even simple - probably it will not be the default option, but if someone wants it would be possible.

I will try to add correct `about` tab on https://pcengines.github.io website to clarify things and if anyone will claim that we are PC Engines employee and have some influence on hardware stuff we will redirect to read that tab.

pietrushnic, 2018-12-04:
@soder10 @michaelsteinmann https://pcengines.github.io/about.html

paravoid, 2019-08-14:
Setting the organizational entity meta-question aside, is there any progress on this on a technical level? Thanks :)

pietrushnic, 2019-08-14:
@paravoid, unfortunately, no, but we welcome contribution and can support by doing a review and provide advisory. Personally, I think *BSD systems support in LVFS would be more interesting for PC Engines hardware. If you can send gather some group of users or even individually send a request to PC Engines it may help placing that on the roadmap.

paravoid, 2019-08-16:
What would the request be? Is this blocked on PC Engines rather than you? I'd expect the  the folks that are handling the releases of firmware versions on pcengines.github.io right now to be the ones to upload them to LVFS, but maybe I'm misunderstanding something here :)

pietrushnic, 2019-08-16:
@paravoid I think the simple message "we the users of PC Engines want LVFS" is good enough. I believe 3mdeb would be the maintainers of PC Engines account in LVFS system if there would be any. Especially that we did initial public support for flashrom and coreboot in LVFS. To make that happen users have to convince PC Engines that LVFS is the top priority and make sense to spend maintainers time on that.

I believe long term there would be support for PC Engines in LVFS, but this is probably 3-6 months perspective. Sorry that it takes so long, but customer bugs and some important features like keeping coreboot guidelines is more important then fwupd/LVFS.

paravoid, 2020-11-10:
Hey, sorry for the bump but I'm wondering if there is any progress here? Do you happen to have any news from PC Engines on this? Given 3mdeb is already authorized to use PC Engines' names (including here on GitHub), I wouldn't have imagined managing their LVFS account to be that controversial! :)

pietrushnic, 2020-11-10:
@paravoid we don't have a budget from PC Engines, but we work on https://dasharo.com, which definitely has LVFS/fwupd support on the roadmap. The good thing we did in the realm of LVFS/fwupd recently is that we get the grant from NLNet Foundation to develop and contribute support for LVFS/fwupd to a couple of major BSD systems.