pcengines/coreboot #256: Checking integrity of apu2c4 firmware

Pferdebockwurst, 2019-01-10:
Hello everyone. I've bought a second hand apu2c4. How can I check and verify the integrity of the firmware (bios/coreboot)?

pietrushnic, 2019-01-12:
Hi @Pferdebockwurst, there is no easy way to do that. Mainly because we had no reproducible builds for coreboot - we will address that in the recent release which will be published this weekend (if I will find enough time).

Each firmware version that you can find [here](https://pcengines.github.io/) in the package has an MD5 sum file. So you can check the version of your firmware `dmidecode -t bios` or similar way, read SPI using flashrom and compare MD5 sum. There can be many problems even with that, first, there is bootorder file and various options in sortbootorder which are saved to SPI, your MD5 may be different. More to that you can use `cbfstool` to extract pieces and then compare those with one that you will get from the mentioned website.

This, of course, is not correct handling of the problem, because of that we will change the way we provide binary, then we will include SHA256 and signed SHA256 file, so anyone can have reasonable trust in what is downloaded.

pietrushnic, 2019-01-12:
@Pferdebockwurst yes, you are safe if you choose correct file for your platform. New firmware overwrite whole SPI flash, so nothing from previous firmware will persist.

Pferdebockwurst, 2019-01-12:
@pietrushnic Great, that's what I was hoping to hear. If flashing should fail I will be the recovery dongle.

Shall I close this issue?

pietrushnic, 2019-01-12:
@Pferdebockwurst great I could help. I will close this issue.

closed 2019-01-12 by pietrushnic