pcengines/coreboot #264: apu5: Fake (doubled) USB entry in the bootorder menu after warmboot/reboot

artur-rs, 2019-02-06:
On apu5 tests setup we have attached 2 x USB 3.0 Corsair storages. Reproduced issue:
```
Select boot device:

1. USB MSC Drive Corsair Voyager VEGA 000A
2. USB MSC Drive Corsair Voyager VEGA 000A
3. USB MSC Drive Corsair Voyager VEGA 000A
4. iPXE
5. Payload [setup]
6. Payload [memtest]

t. TPM Configuration

```
Choosing `1. USB MSC Drive Corsair Voyager VEGA 000A` brings up iPXE setup/console.

miczyg1, 2020-08-04:
@lwcislo does  the problem still persist?

lwcislo, 2020-08-04:
@miczyg1 with the latest firmware we are not reproducing this issue, but we have changed USB sticks in hardware matrix:

     1. USB MSC Drive Kingston DataTraveler 3.0 PMAP
     2. USB MSC Drive Kingston DataTraveler 3.0 PMAP
     3. SD card SB16G 15193MiB
     4. ata0-0: SATA SSD ATA-10 Hard-Disk (15272 MiBytes)
     5. iPXE
     6. Payload [memtest]
     7. Payload [setup]

closed 2020-08-04 by miczyg1