pcengines/coreboot #284: MemTest can't use multiple cores

PeterFalken, 2019-03-12:
I tried running the memtest utility from the bios image and configuring the SMP option both at startup and through the configuration menu (c) option. Even when selecting 'Parallel (All)' it goes back to SMP Disabled.

Has anybody been able to activate SMP for the MemTest??

[image]
[image]
[image]

miczyg1, 2019-03-20:
IIRC SMP is not enabled by default in memtest86, because it caused much trouble on many platforms. I will look deeper into it and provide more precise answer.

PeterFalken, 2019-03-20:
Thanks for looking into this, Michał.

532910, 2019-06-20:
In case of trouble it must not ask to enable it on startup:
```==> Press F2 to force Multi-Threading (SMP) <==```

miczyg1, 2019-06-24:
The reason why enabling it is not possible is that serial console in memtest does not handle the multibyte sequences produced by Fx keys, only the printable ASCII keycodes. Since USB keyboards are not supported, but PS2 are, only pressing F2 on PS2 keyboard can possibly turn SMP on.

Tried to enable SMP manually and it works, however when switching to parallel tests, the output gets garbled (lots of terminal control sequences pop up in few places). Serial is accessed by all cores for a brief moment.

I may upload a patch to memtest to fix the serial console.

soder10, 2019-06-30:
@miczyg1 Reading your post I start thinking that the "official" memtest payload provided with the APU2 firmware is not really tested to fit this platform.

miczyg1, 2019-06-30:
@soder10 what makes You think so? I do not recall having better memtest for this platform. SMP is not necessary for the payload operation. The reason why SMP is not enabled by default, is that memtest does not get that much focus from community ond us ourselves. SMP is quite complex and requires significant amount of time to fix. Why still not fixed? There are much more valuable and critical things to do in other field, and memtest payload is not widely used anyway. Besides You are free to use what You want, i.e. if memtest does not fit Your needs, do not use it until it reaches Your satisfaction.

PeterFalken, 2019-07-05:
Thanks for looking into this Michał.
So in the ends it seems that a patch is necessary for memtest to handle the new USB keyboards.

miczyg1, 2019-07-05:
Indeed. Sending complex escape sequences via serial console is not trivial. It will be better to have the USB support so other platforms could control the memtest too.

soder10, 2019-07-05:
@miczyg1: what I mean is that there is a built-in payload (e.g. memtest) that contains no attached guide or warning or known issues. Naive myself would imagine, that at least the options 3mdeb compiled into the blob and I can activate or select in the memtest config, ...well.. they should just work?! Or do I see things again wrong?

I know there is similar problem with the ECC, memtest compiled into the firmware does not test ECC. But at least "ECC not supported" was written clearly in a well hidden 3mdeb blog post, when the ECC support for the 4.9.x mainline was introduced a couple of months ago. But that also had to be located (would require a detective if you ask me), nothing suggests anything about non-working memtest features on pcengines.github.io

Are you telling me, that I can go ahead and modify the main page of pcenginea.github.io as I like? Seriousl? As I cannot believe thats true.

miczyg1, 2019-07-09:
@soder10 memtest was included as a payload even before 3mdeb started maintaining PC Engines firmware. We basically keep the state, but improve quality, upstream patches, resolve bugs etc. Facing troubles with OS issues is more important to us than memtest, that said, we cannot possibly focus on everything all the time.

The history behind ECC support is much longer and more detailed than You think. On the beginning of apu2 product, PC Engines official product site announced ECC support on the platform. However, there were two issues: 

1. AMD proprietary silicon initialization code called AGESA had a bug that excluded whole memory from ECC. That is why ECC was not working at all.
2. PC Engines was unable to prove that ECC works and had to remove the information about ECC support from the product site. And this is where 3mdeb came, resolved the problem and has written a blog post about ECC.

We were announcing the blog post via many social media channels: twitter, facebook, linkedin, reddit, various forums. It's a pity that it could not reach You, despite our efforts. Now we also have the pcengines.github.io page which serves as a container for all information related to PC Engines firmware. The so-called "hidden" blog post is linked to the v4.8.0.5 release notes on the pcengines.github.io page ...

Yes, I am telling You that You can go ahead and modify the page. Set up a pull-request with any known issues You are experiencing and let's improve the site quality. This is the idea of open-source where community engagement helps the project grow and anybody can contribute.

soder10, 2019-08-12:
@miczyg1 Thanks for the updated pcengines.github.io page.

Please add somewhere notes that the multi-core and ECC not supported for the built-in memtest payload.
If you can addd some small glossary at the bottom of the page, describing what does SOL mean (I know it is resolved to Sign Of Life, but what this SOL string really means in detail 2 simple sentence etc.)

miczyg1, 2019-08-12:
@soder10 okay, will do.