pcengines/coreboot #299: Feature request: ACPI GPIO device (AMD0030) for FCH GPIO block

avg-I, 2019-05-23:
It would be nice to have such a device to make it easier and more uniform to work with the FCH GPIO-s that are connected to the APU LEDs and S-Button and possibly other functions (like the SIM selector).

On top of that, it would be great if the firmware configured an interrupt (e.g., IRQ 7) for the GPIO function and declared it in the ACPI device. That would be especially useful for the button events.

And example of such an ACPI device can be found in coreboot:
https://review.coreboot.org/c/coreboot/+/17218
https://review.coreboot.org/c/coreboot/+/17218/3/src/mainboard/amd/gardenia/acpi/carrizo_fch.asl

Also, it seems like there was a similar ACPI configuration for APU1:
https://github.com/coreboot/coreboot/blob/master/src/mainboard/pcengines/apu1/acpi/gpio.asl
https://github.com/coreboot/coreboot/blob/master/src/mainboard/pcengines/apu1/acpi/leds.asl

miczyg1, 2019-05-24:
Thank You for submitting the issue. We are currently working on that, we had similar requests from OpenWRT community. Appreciate the attached links.

miczyg1, 2019-05-24:
@krystian-hebel hope these will help You with Your task

miczyg1, 2019-07-26:
@avg-I got it working with ACPI.  Will be available in next release

miczyg1, 2019-10-14:
https://github.com/pcengines/coreboot/pull/343
https://github.com/pcengines/coreboot/pull/324
Associated documentation: https://github.com/pcengines/apu2-documentation/blob/master/docs/gpios.md

closed 2019-10-14 by miczyg1