pcengines/coreboot #309: APU4: frequent connection loss using BIOS 4.9.0.5

ruggerio, 2019-06-11:
Actual, non working version: 4.9.0.5
Working Version:                 : 4.0.23

Problem: Frequent loss of connections
Protocols: unspecific, SSH, IMAP, IMAPS...

OS: Opnsense 19.1.9

As soon as a longer transmit (e.g a mail with a large attachment) is sent, the whole connection is blocked. Eventually openend ssh-conns are closed too. No logenty neither in systemlogs, nor in dmesg.

miczyg1, 2019-06-11:
@artur-rs could You try reproducing the issue?

@ruggerio I assume sending a large file via scp will suffice to emulate a long transmission.

ruggerio, 2019-06-11:
@miczyg1: might be sufficient. The mailattachment had about 1 MB, which is not that much. Apparently, the connection broke in evolution, when writing back the mail to the sent-folder via imap (stucked between 35% and 67%)

miczyg1, 2019-06-12:
@ruggerio we have problems with reproducing the issue. What we did:

1. Installed OPNSense 19.1.4 on apu4 with v4.9.0.5 firmware.
2. Connected a laptop to the LAN port of apu4 and enabled SSH via web GUI.
3. Further upgraded OPNSense to 19.1.9 via web GUI.
4. Launched youtube video, `ping google.com` and did scp a file from the WAN network to the laptop disk. All processes were running simultaneously. 

The connection has not been broken even once. 

Any more details on the frequency? How long do we have to stress the link to trigger the connection loss? The OPNSense configuration may also influence the results, since we do not replicate Your configuration, our environment is a little bit different and unfortunately we have an APU4A only.

ruggerio, 2019-06-12:
OK, so updated again to 4.9.0.5 - the problem came  back. 

- Sending a Mail on a virtual server (lxc)
- ssh'ing to the same server
= block.

meanwhile, all traffic to internet on other ports flows without
problems.

going back to 4.0.23 again - no problem at all. I have no explanation
for this, but i am on a apu4c4.

i would appreciate, it wouldn't break.

ruggerio, 2019-06-12:
before i forget: this goes within seconds, when sending the mail
(problem will be saving the outgoing mail in imap-folder on the lxc)
and having a ssh-conn to the same server open.

miczyg1, 2019-06-13:
The issue trigger looks very specific to me. I am not sure whether it is firmware related or software related. Have You tried posting a similar problem elsewhere? Maybe someone had a similar issue.

What I can offer is:
1. To setup exactly the same installation of OPNSense and try the virtual server approach. However it would require Your OPNSense configuration file.
2. Try v4.9.0.4 and v4.8.0.2 firmware and get back to me if the issue is present there also.

ruggerio, 2019-06-13:
Today, 4.9.0.6 made its way, as it seems with a new version of seabios.
I upgraded a few hours ago, til now no more problems. Strange.

I posted this also to QNAP, as the main problems occured when
connecting to one of the virtual ips of the device.

Please close the case, sorry for your time. Thanks!

Roger