pcengines/coreboot #313: Feature request: assign I/O port range to GPIO function of NCT5104D

avg-I, 2019-07-01:
As documented in section 10.2, there are two ways to control GPIO pins. One is through configuration registers and the other is through a more direct access to I/O registers.
In the first mode an indirect access through HEFRAS has to be used. That may mean many I/O reads and writes. For instance:
- enabling the configuration mode
- selecting the required LDN
- reading or writing the CR
In the second mode we just or write a single I/O port.

So, the second mode is substantially faster.
That can make a big difference when using a GPIO pin for a protocol such as 1-wire where some timing are required to be pretty short.

The following setup is need to enable the second mode:
- program CRs 0x60 and 0x61 of LDN 8 with a base I/O port
- set bit 1 (mask 0x02) of CR 0x30 in LDN 8

In my tests this worked rather well.
I picked 0x290 as the base port as it didn't seem to conflict with any other device on APU2.

miczyg1, 2019-07-10:
@avg-I first thing is that the pins have to be configured for GPIOs not UARTs. We have the runtime configuration for that. We were considering adding a better support for NCT5104D in ACPI, which could help in interfacing GPIOs/UARTs. AFAIK there is no Linux kernel module to handle that, however FreeBSD has:
https://reviews.freebsd.org/rS297443 with some further fixes to device IDs.

The problem here is that the driver uses the first method. Can those two methods cooperate in parallel (I do not remember whole Nuvoton datasheet)? If yes, we could add the second method.

avg-I, 2019-07-10:
As far as my testing shows, both methods do coexist without problems.

miczyg1, 2019-10-14:
Implemented for both legacy and mainline:

https://github.com/pcengines/coreboot/pull/343
https://github.com/pcengines/coreboot/pull/340

closed 2019-10-14 by miczyg1