pcengines/coreboot #314: Feature request: perform a soft reset of NCT5104D before configuring it

avg-I, 2019-07-01:
I observe some configuration settings of NCT5104D, such as GPIO configuration, for instance, survive reboots, resets and even soft power-off. The only thing that fully resets the chip is a hard power off. I guess that that's because some functions use the standby power well.
In my opinion this is inconsistent and may even be dangerous if a GPIO pin remains in a non-default (input, high impedance) state before software can take control.

My proposal is to reset the chip by writing to CR 0x02 with value of 1 as documented in section 12.1. This needs to be done quite early, before any configuration is applied.
`enable_dev()` in src/superio/nuvoton/nct5104d/superio.c might be a good place to do that.

miczyg1, 2019-07-10:
@avg-I I think we have analyzed that some time ago and it occured that LPCRESET# signal does not reset the NCT5104D. Due to limited Nuvoton datasheet and SoC datasheet we had very little luck to make things work as intended. BTW we also thought about CR 0x2 and soft reset. Will approach the bug again in the next release

miczyg1, 2019-10-14:
Reset via CR 0x2 did not work. Setting the input and open drain does the trick. Fixed in mainline and legacy:
https://github.com/pcengines/coreboot/pull/338
https://github.com/pcengines/coreboot/pull/339

closed 2019-10-14 by miczyg1