pcengines/coreboot #315: NCT5104D watchdog function does not seem to work on APU2C4

avg-I, 2019-07-01:
APU2C4 user here.

APU2 user manual mentions that NCT5104D has a watchdog function and that pins 5 & 6 of J2 header need to be shorted to enable it. According to the APU2C schematic that should connect WDTO pin of the NCT5104D to the reset signal (via Schottky diodes). So, I guess that a low signal on WDTO should reset the system. The problem is that I never see  a low signal on pin 6.

I am using FreeBSD wbwd driver. And I can inspect the relevant configuration registers in LDN 8. If the watchdog daemon is killed, I can see how the watchdog timer runs down to zero (register 0xF1) and jumps back to its starting value and then I see that the watchdog timeout status gets set (bit 4 in 0xF2). However, there is no low pulse on pin 6 of J2 when that happens. I checked that by using a logic analyzer. I don't know what the problem is. Either that pin is incorrectly wired or NCT5104D does not pulse its WDTO pin (pin 22).
(A side note, I took a look at APU2D schematic and there is a note near the the Schottky diodes "Rev 2D: depopulate").

Another observation is that if I configure one of multi-function pins to carry WDTO signal, then I can see a low pulse of approximately 100 ms at that pin.  For example, I configured GP01 (which is routed to pin 4 of J17 GPIO header) in that fashion by setting bit 1 of configuration register 0xE0 in LDN 8. After that I got the low pulse on that pin.

Could you please clarify what is going on with respect to the watchdog functionality and WDTO signal of NCT5104D on APU2 ?
Thank you!

michaelsteinmann, 2019-07-01:
The NCT5104D watchdog doesn't work - the Nuvoton part generates a reset
pulse,
which gets cut short by the reset coming from the AMD part. And then the
AMD part gets all confooooooosed.

avg-I, 2019-07-01:
@michaelsteinmann I observed something like that when I reconfigured GP01 for WDTO function and connected that GPIO pin to the reset pin on J2 header...
But as I said, I do not see any low pulse at all on pin 6 of J2 header. The pin is not connected to reset. I just connected a signal analyzer to it , so that I can watch it without disturbing anything.

michaelsteinmann, 2019-07-01:
Forget the watchdog on the NCT5104, instead make use of the watchdog of the
CPU.
See https://wiki.polaire.nl/doku.php?id=pc_engines_apu_-_watchdog_centos_6.5

avg-I, 2019-07-01:
That one does work without any issues. It's amdsbwd in FreeBSD.
But I am exploring the hardware (and the documentation too: "_Pins 5-6 = enable NCT5104D watchdog reset (requires additional software support)_").

michaelsteinmann, 2019-07-01:
I repeat and this is an official PC Engines statement:
The NCT5104D watchdog doesn't work on apu boards.

avg-I, 2019-07-01:
This is very clear. Thank you!

miczyg1, 2019-07-10:
I assume the issue is resolved then. Closing.

closed 2019-07-10 by miczyg1