pcengines/coreboot #316: Feature request: configure an interrupt for FCH GPIO

avg-I, 2019-07-01:
My current ultimate goal is to be able to get an interrupt when S1 is pressed.
I believe that it should be sufficient to put a value at position 0x62 in `mainboard_intr_data[]`.
For example, interrupt 7 seems to be common choice. It should work for APU-s as traditionally that interrupt was used for a parallel port which is missing, of course.

dankm, 2019-07-05:
This looks related (but not the same as) #299, and this should probably have an ACPI entry added to advertise the interrupt to the OS.

miczyg1, 2019-07-10:
@avg-I we are working on the ACPI GPIO support for APU devices. When done it should be possible to configure interrupt on S1 button in user-space.

miczyg1, 2019-07-26:
@avg-I got it working with interrupts. Binary for APU2 for testing:

https://cloud.3mdeb.com/index.php/s/mRJ8pz6BcHSdk3Q

Let me know if something requires improvements (like debounce-interval etc.) or in the documentation: https://github.com/pcengines/apu2-documentation/pull/157

miczyg1, 2019-10-14:
Added both to legacy and mainline: https://github.com/pcengines/coreboot/pull/343 https://github.com/pcengines/coreboot/pull/324

closed 2019-10-14 by miczyg1

avg-I, 2019-10-15:
Thank you very much!