pcengines/coreboot #321: apu3 legacy v4.0.27: watchdog cannot be enabled

artur-rs, 2019-07-09:
After setting various values, watchdog is still in the state: `Currently Disabled`

miczyg1, 2019-09-13:
Fixed in v4.10.0.1 and v4.0.28:
https://github.com/pcengines/coreboot/pull/326
https://github.com/pcengines/coreboot/pull/327

closed 2019-09-13 by miczyg1

soder10, 2019-11-15:
Can you guys at least write a couple of lines of description, how this watchdog works? Similar to what other docs are already found on: 

https://github.com/pcengines/apu2-documentation/blob/master/README.md

Does Opnsense / Freebsd support this watchdog, or any other Linux distribution, is it safe to enable it, or will it cause any reboot loops if the OS doesnt handle the watchdog signals properly etc.?

miczyg1, 2019-11-15:
@soder10 https://github.com/pcengines/sortbootorder#settings-description
There is an entry that describes the watchdog. If you do not know what watchdog is, you should not probably use it.

Debian 10.0 stable supports it and it is safe to use it.

soder10, 2019-11-15:
> @soder10 https://github.com/pcengines/sortbootorder#settings-description
> There is an entry that describes the watchdog. If you do not know what watchdog is, you should not probably use it.
> 
> Debian 10.0 stable supports it and it is safe to use it.

I know what a watchdog supposed to do. At least in theory. But since I learned in the past 1,5 yrs that things normal enduser people (a.k.a not developers) understand about COREBOOT and APU2 and things 3mdeb devs understand about the same subject, is mist of the time the opposite sides if a line. So I tried to discover what this watchdog feature should give me, and therefore I ask the people who added this feature into COREBOOT. I mean, you guys.

Sortbootorder doc indeed talks about this watchdog, but only in a very minimalistic way:

"Watchdog" - enables/disables hardware watchdog. Each time toggled, the prompt will pop out asking to provide a timeout value in seconds (0 to disable). Set reasonably high value so it will be possible to disable the watchdog later. Too low value may result in a reset loop!

-> maybe its just me again, but what is a reasonable timeout value here? I mean some starter value would be more than welcome. 60 seconds? Or 3600 seconds (a.k.a 1 hour)? Is there an upper limit of this value by the way, eg. 16bit number 65535 or  32bit number 4.1 billion?

Does it require jumper shortcircuitting on the board as well, or this a pure software-only feature?

Regarding the OS support: Debian 10 is the only working example, or do you know any list of other OS as well?

pietrushnic, 2019-11-15:
@soder10 we can be as precise as the budget allows us, more to that we should both assume goodwill. I'm sorry if the explanations we provide are confusing to you. We would appreciate it if you can point exact places in the documentation or our conversation which we can improve (ideally would be in the form of issue). We also accept pull requests if you can contribute to documentation.

The reasonable watchdog time value can be understood by reading google results. Value highly depends on your application. We cannot educate you about every single feature and all hardware details (schematics are there as well as source code) we simply don't have time for that. We don't have magic powers we have the same resources as you in this case.

I'm not sure where you get information about the jumper, @miczyg1 is there anything like that needed?
We tested that only on Debian, we know some customers used those patches with older firmware on OpenWRT, we are not sure how OpenWRT works with the recent firmware versions.

I'm really sad we have this repeating tone in discussion with you. PC Engines is a manufacturer with the most frequently updated firmware continuously providing very good quality per price. If you can point us to the vendor that meets your quality criteria we will try to follow. I'm also curious who in your opinion does a better job.

Please note you can support us in various ways including paying us to meet your expectations.

soder10, 2019-11-16:
@pietrushnic Well, what to say. I asked a simple question: what does the watchdog do in APU2? As if I assume it does SOMETHING, and turns out my expectations were too high, or incorrect, you will come and tell me the 3mdeb translation of watchdog is a flash-and-blood animal, and not what an average router builder will think. Joking, of course, before you get mad at me. 
Guys, you dont have to write a 20 page essay. Just pls drop 5 sentence, what this watchdog can do, and what system can use it for the moment. Before we get starteed, dont take this literally, it can be even 6 or 7 sentences.

It is also a valid question, how big the timeout can possibly be in sortbootorder? Just to give an idea to your users.

Jumpers and watchdog, found here:
http://www.pcengines.info/forums/?page=post&id=F1D33B75-58E7-4A5A-9A80-1E8CA35AC84D&fid=DF5ACB70-99C4-4C61-AFA6-4C0E0DB05B2A&pageindex=5

Dont get me wrong guys, its great that you add new features, and I appreciate that. But at least put there a small summary for all these things on pcengines.github.io (as you have already done with flashrom, CPB, warm reboots etc).

miczyg1, 2019-11-18:
@soder10 the jumper was designed to enable the watchdog present on Nuvoton NCT5104D, but it is not supported in coreboot. Most BSD and Linux distro also don't support each SuperIO chip.

The watchdog implemented in coreboot uses the CPU watchdog. The reasonable value depends on the hardware configuration. It should be set at least to such high value, that OS will be able to take over and start feeding the watchdog with its own system drivers. The reason we implemented that was the possibility to enable watchdog in the early boot phase, so that the platform would reset if for any reason it would hang during bootup.

soder10, 2019-11-18:
> @soder10 the jumper was designed to enable the watchdog present on Nuvoton NCT5104D, but it is not supported in coreboot. Most BSD and Linux distro also don't support each SuperIO chip.
>
> The [...]

Thanks @miczyg1 that is all I really wanted to know, and most of your users will also be informed well with this. Keep up the good work guys!

miczyg1, 2019-11-18:
Great. I will add a similar description to pcengines.github.io/apu2-documentation and some background to other runtime configuration options as well then.