pcengines/coreboot #322: Is hyperthreading supported and available in APU2?

soder10, 2019-07-10:
Hello everyone,

I tried to figure out if Hyperthreading is supported in APU2 or not. Seems its not an easy task. The AMD CPU in APU2 is a quad core type, so there are at least 4 CPUs reported to any OS. But I can also see the CPU feature flag HTT mentioned in the logs of the OS I am using. I have enabled the setting ftir Hyoerthreading in the OS (in fact, there is a setting that can DISABLE the HTT, which I havent set, so HTT should be still enabled).  So in theory(!) the OS should report total of 8 virtual CPU. 

In theory. 

Unless there is again something in the background, why this isn't true in reality. But I cannot seem to find anything about this. Please enlighten me, if I found another thing in the APU2 product that is not working (e.g because of AMDs fault etc.)

soder10, 2019-07-10:
Even the 3mdeb blog about the turbo clock mentions hyperthreading un one of the benchmarks, althoufh the context is not clear:

https://blog.3mdeb.com/2019/2019-02-14-enabling-cpb-on-pcengines-apu2/

So the 1 million dollar question: the AMD GX-412TC SoC has 4 physical cores and no hyperthreading capability, or on fact it has only 2 physical cores and hyperthreading makes it look like 4 virtual cores?

miczyg1, 2019-07-25:
@soder10 that is interesting... I didn't even notice such capability in the first place. Hope to find something useful in the SoC documentation.

PS: I have always believed that Hyperthreading is Intel's feature. Anyway good reason to research it.

miczyg1, 2019-07-25:
@soder10 ok that was pretty easy to find:

https://www.amd.com/system/files/TechDocs/52740_16h_Models_30h-3Fh_BKDG.pdf
Section 2.4.4 (page 36 and 37):

```
• The cores of a L2 complex may be software downcored by D18F3x190[DisCore]. 
...
• D18F3x190[DisCore] affects the value of CPUID Fn0000_0001_EBX[LogicalProcessorCount], CPUID
Fn0000_0001_EDX[HTT], CPUID Fn8000_0001_ECX[CmpLegacy], CPUID
Fn8000_0008_ECX[NC], D18F5x80[Enabled, DualCore, TripleCore, QuadCore]. D18F3x190[DisCore] does not affect the value of D18F5x84[CmpCap].
```

And CPUID Fn0000_0001_EDX Feature Identifiers on page 531:

```
HTT: hyper-threading technology. Value: CPUID Fn8000_0008_ECX[NC]!=0. 1=Multi core
product (CPUID Fn8000_0008_ECX[NC] != 0). 0=Single core product (CPUID
Fn8000_0008_ECX[NC]==0).
```

So the HTT bit indicates whether this is a single core or multi-core system. If we downcore the SoC to a single core, the HTT should disappear. So the HTT does not mean Intel's hyperthreading introducing additional set of virtual cores. But I have found this:
https://www.quora.com/Why-is-the-hyper-threading-absent-in-AMD-processors

It might be the Clusterfuck (SMT) :) So quad core may be a dual core with two threads on each. If You are interested in checking the downcoring capability as a proof I may try it.

soder10, 2019-07-25:
Thanks for this detailed post @miczyg1  greatly appreciated! Seriously!

Yes, this topic is very confusing: AMD advertises this SoC as being 4-core
Reference: https://www.amd.com/Documents/AMDGSeriesSOCProductBrief.pdf
Additional reference: https://www.amd.com/en/products/specifications/embedded/ --> search for GX-412TC

 I would say, if AMD says its 4-core, that MUST mean 4 physical cores. But then, HTT is a completely different topic, enabling HTT on a 4-physical core should result in a 8 logical core setup. Which is not the case as I can see. So clearly, I dont know how to interpret the so-called HTT support in this SoC. Being multi-core does not mean being HTT-capable or vice-versa.

soder10, 2019-08-02:
...cirp, cirp, cirp...

miczyg1, 2019-08-02:
Okay, I assumed my explanation was fair enough. @soder10 I know that documents very well. However, we should not mistake the HTT here with Intel hyperthreading technology. For the Intel platforms, HTT is located under the same location, but the meaning is slightly different:
https://www.intel.com/content/www/us/en/architecture-and-technology/64-ia-32-architectures-software-developer-instruction-set-reference-manual-325383.html
(see Figure 3-8 and Table 3-11). It serves the purpose of identifying if hyperthreading is supported and/or enabled. Given that the bit is under the same location, AMD put something similar there to be compatible. So for AMD it basically means HTT=multi-core, at least for this particular SoC.

There are a mutli-threaded processors from AMD:
https://www.amd.com/en/product/8546 (4 cores, 8 threads)
To compare with GX-412TC:
https://www.amd.com/en/product/7426 (4 cores)

Another interesting thing is that GX-412TC is an SoC and has a security processor, but this product catalogue reports it is a CPU type product without security processor. Bullshit...
In the light of attached evidence and misleadign informations from the silicon vendor, I find it almost impossible to prove whether it has the capability to be a 8-threaded CPU or not or any other combination.

soder10, 2019-08-05:
@miczyg1 Well, thanks once again. My concern against AMD business practice has not been addressed (that would be AMDs fault dont get me wrong), you did all what you could do. If you dont see any further chance to get the official final one and true answer, you can close this "issue".

miczyg1, 2019-08-05:
All I can do is to try disabling all cores except core0 and check the HTT bit again if it is cleared. However, this will not answer Your question whether it is hyperthreading or not. @soder10 everything up to you.

closed 2019-08-05 by soder10