pcengines/coreboot #323: Setting 'amd_iommu=off' causes stack trace in early boot

chr0mag, 2019-07-11:
I'm running an APU2C4 as a wireless router with the following devices attached:
- WLE200NX (PCIe slot 2)
- WLE900VX (PCIe slot 1)
- mSATA SSD (boot drive)
- USB 3.0 hard disk (non-boot)

I'm running an up-to-date Archlinux and this setup has been running well for over a year. 
_uname -a_
`Linux apu2c4 5.1.16-arch1-1-ARCH #1 SMP PREEMPT Wed Jul 3 20:23:07 UTC 2019 x86_64 GNU/Linux`

After upgrading to firmware` v4.9.0.5` I ran into issue #206 . I added `amd_iommu=off` to my kernel params to workaround this and I now see the following stack trace on boot. The system does recover though and seems to complete booting without further errors, and my WLE200NX can once again broadcast its SSID

```
 WARNING: CPU: 0 PID: 0 at kernel/workqueue.c:2771 flush_workqueue+0x3ca/0x420
Jul 10 21:33:42 apu2c4 kernel: Modules linked in:
Jul 10 21:33:42 apu2c4 kernel: CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.1.16-arch1-1-ARCH #1
Jul 10 21:33:42 apu2c4 kernel: Hardware name: PC Engines apu2/apu2, BIOS v4.9.0.5 05/09/2019
Jul 10 21:33:42 apu2c4 kernel: RIP: 0010:flush_workqueue+0x3ca/0x420
Jul 10 21:33:42 apu2c4 kernel: Code: ff ff ff 4c 89 e7 48 89 0c 24 e8 51 f1 ff ff 41 8b 44 24 44 41 8b 74 24 40 48 8b 0c 24 e9 f8 fe ff ff 4c 89 fb e9 77 fe ff ff <0f> >
Jul 10 21:33:42 apu2c4 kernel: RSP: 0000:ffffffff83603d60 EFLAGS: 00010246
Jul 10 21:33:42 apu2c4 kernel: RAX: ffffffff83603d98 RBX: ffff9aef6a439980 RCX: 0000000000000000
Jul 10 21:33:42 apu2c4 kernel: RDX: ffffffff83c7bc48 RSI: ffffffff83498042 RDI: ffffffff83603d90
Jul 10 21:33:42 apu2c4 kernel: RBP: ffffffff83603de0 R08: 0000000000000000 R09: 0000000000000000
Jul 10 21:33:42 apu2c4 kernel: R10: fffff4fcc4a95608 R11: 0000000000000000 R12: 0000000000000000
Jul 10 21:33:42 apu2c4 kernel: R13: ffffffff83603d70 R14: 0000000000000501 R15: 0000000000010c01
Jul 10 21:33:42 apu2c4 kernel: FS:  0000000000000000(0000) GS:ffff9aef6aa00000(0000) knlGS:0000000000000000
Jul 10 21:33:42 apu2c4 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 10 21:33:42 apu2c4 kernel: CR2: ffff9aef6efff000 CR3: 000000004740e000 CR4: 00000000000406b0
Jul 10 21:33:42 apu2c4 kernel: Call Trace:
Jul 10 21:33:42 apu2c4 kernel:  kmem_cache_destroy+0x5b/0x200
Jul 10 21:33:42 apu2c4 kernel:  ? __free_pages_ok+0x156/0x2c0
Jul 10 21:33:42 apu2c4 kernel:  free_iommu_resources+0x5a/0x191
Jul 10 21:33:42 apu2c4 kernel:  iommu_go_to_state+0xafd/0x12ef
Jul 10 21:33:42 apu2c4 kernel:  amd_iommu_prepare+0x16/0x2a
Jul 10 21:33:42 apu2c4 kernel:  irq_remapping_prepare+0x36/0x5f
Jul 10 21:33:42 apu2c4 kernel:  enable_IR_x2apic+0x21/0x188
Jul 10 21:33:42 apu2c4 kernel:  default_setup_apic_routing+0x12/0x6f
Jul 10 21:33:42 apu2c4 kernel:  apic_intr_mode_init+0x15c/0x197
Jul 10 21:33:42 apu2c4 kernel:  x86_late_time_init+0x17/0x1c
Jul 10 21:33:42 apu2c4 kernel:  start_kernel+0x47c/0x530
Jul 10 21:33:42 apu2c4 kernel:  secondary_startup_64+0xa4/0xb0
Jul 10 21:33:42 apu2c4 kernel: ---[ end trace 1d30ed0ae73a58ff ]---
```

I have yet to try the latest firmware (`v4.7.0.9` at the time of writing) or different kernels (LTS, Mainline). Let me know if I can provide more information.

miczyg1, 2019-07-12:
@chr0mag the information seem to be sufficient for now. @artur-rs let's try to reproduce it.

thillux, 2019-07-12:
@chr0mag you can use intremap=off as a temporary fix.

chr0mag, 2019-07-28:
Thanks @thillux - I no longer see the stack trace after adding that parameter.

FWIW, an upgrade to kernel 5.2 doesn't help. 

_uname -r_
`5.2.3-arch1-1-ARCH`

thillux, 2019-07-29:
Nice to hear that @chr0mag. Let me add some further explanations. Your WLE200NX card uses the AR9280 chip, which is served by the ath9k driver in the Linux kernel. ath9k uses legacy interrupts by default (Message Signaled Interrupts [MSI] are usable with a module parameter). These legacy interrupts cannot be remapped, as far as I know. By activating the IOMMU (which 4.8.0.4+ does), the kernel tries to remap interrupts with help of the IOMMU. This is only possible with MSI/MSI-X.

If you need interupt remapping, you can also try the use_msi=1 parameter of the ath9k kernel module, which enables remappable message signaled interrupts in the ath9k driver.

miczyg1, 2019-08-01:
@thillux this is useful information. We will add it to our documentation for WLE200NX module. Thank You

miczyg1, 2019-08-05:
Documentation updated: https://github.com/pcengines/apu2-documentation/pull/159
Can we close the issue?

chr0mag, 2019-09-15:
Apologies for the slow response. From my perspective this can be closed.

@thillux - thanks for the additional information about this.

closed 2019-09-15 by chr0mag