pcengines/coreboot #329: 4.10.0.0 FreeBSD apuled driver failed to initialize on APU2

alexpro, 2019-08-22:
With new 4.10.0.0 BIOS I lost LEDs and modeswitch devices on my APU2 in FreeBSD 11.2
With 4.9.0.5 all works fine.
Driver message with 4.9.0.5:
`apuled0: <APU2> at iomem 0xfed81610-0xfed8161b,0xfed81664-0xfed81667 on isa0`
Driver message with 4.10.0.0:
```
apuled0: <APU2> at iomem 0xfed81610-0xfed8161b,0xfed81664-0xfed81667 on isa0
apuled0: Unable to allocate memory region 0
device_attach: apuled0 attach returned 6
```
device_attach() returns ENXIO error (device not configured)
Is this some error in new BIOS, or is there a need to make changes in driver code according to changes in new BIOS?

miczyg1, 2019-08-26:
The v4.10.0.0 BIOS adds a GPIO controller device in the ACPI, which reserves the MMIO for the GPIOs. Most likely FreeBSD cannot reserve the memory for the native driver now. 

The GPIOs are now accessible in Linux, however it introduced a regression on BSD. IMO it would be better to make changes to the driver. There are few reasons:

1. ACPI GPIO controller allows to access all GPIOs (i.e. not only leds, mode switch, but also PCIe resets, WLAN disable signals on PCIe slots)
2. Certain GPIOs can be controlled by generic drivers like gpio-leds, gpio-keys in Linux. We will check whether there is any equivalent on BSD side.

It worked for Linux without issue when native platform drivers were in place along with the ACPI GPIO controller, but BSD is not so flexible.

alexpro, 2019-08-29:
> The v4.10.0.0 BIOS adds a GPIO controller device in the ACPI, which reserves the MMIO for the GPIOs.

Thanks for the hint, the problem is exactly here. I can't quickly fix the driver code unfortunately, since it seems I need to completely rewrite it. But fortunately I found a workaround which can solve this problem pretty easy. There is a need to add environment variable `debug.acpi.avoid="\_SB_.PCI0.GPIO"` 
to loader.conf, and apuled driver works fine. This parameter does not affect other ACPI functionality, it only prevents GPIO resource allocation by ACPI driver.

alexpro, 2019-09-13:
@Nimloth There are some kind of "pools" each of 32 pins. But gpioled does not support select particular "pool", it can use only first one.

avg-I, 2019-10-15:
@alexpro , I added "pin_list" hint a while ago to address that exact issue.
See gpiobus manual page.

FWIW, gpioled is a wrong place to document hints like "pins" and "pin_list", because -- even if not intuitive -- those are properties of the gpio bus and not any individual driver on it.

miczyg1, 2020-07-15:
The workaround is added to the documentation: https://github.com/pcengines/apu2-documentation/blob/master/docs/gpios.md#known-issues

closed 2020-07-15 by miczyg1