pcengines/coreboot #351: Thermal Sensors not available on 4.10.0.3

amichelf, 2019-11-08:
Hi,
after installing the latest 4.10.0.3 Firmware on two apu2c4 running opnsense the thermal sensors did not report the temperature.
Downgrading to 4.10.0.2 fiexed the issue.
amichelf

pietrushnic, 2019-11-22:
@amichelf sorry for not replying for long. We working on issue, it was reported here: https://github.com/pcengines/apu2-documentation/issues/168

miczyg1, 2019-12-10:
Fixed in v4.11.0.1. coreboot community changed the definitions of processors in ACPI to be compliant with ACPI specification, so now pfSense/OPNSense APCI compliance is under a big question mark. We will investigate the topic further, for now we kept the non-compliant ACPI definitions of processors.

closed 2019-12-10 by miczyg1