pcengines/coreboot #352: Duplicate sysfs entries for front LEDs

chr0mag, 2019-11-17:
After upgrading firmware from `v4.9.0.5` to `v4.10.0.3` I now see 2 sets of sysfs entries for the 3 front LEDs. The `apu2:green:ledX` are new, which makes sense from the [docs](https://github.com/pcengines/apu2-documentation/blob/master/docs/gpios.md). The `apu:green:Y` are the ones I've been using up until the firmware upgrade.

Both set of entries appear to be exposed by the `leds_gpio` module.
```
% ls -l /sys/class/leds 
total 0
lrwxrwxrwx 1 root root 0 Nov 16 21:03 apu2:green:led1 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led1
lrwxrwxrwx 1 root root 0 Nov 16 21:03 apu2:green:led2 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led2
lrwxrwxrwx 1 root root 0 Nov 16 21:03 apu2:green:led3 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led3
lrwxrwxrwx 1 root root 0 Nov 16 21:03 apu:green:1 -> ../../devices/platform/leds-gpio/leds/apu:green:1
lrwxrwxrwx 1 root root 0 Nov 16 21:03 apu:green:2 -> ../../devices/platform/leds-gpio/leds/apu:green:2
lrwxrwxrwx 1 root root 0 Nov 16 21:03 apu:green:3 -> ../../devices/platform/leds-gpio/leds/apu:green:3
lrwxrwxrwx 1 root root 0 Nov 16 20:13 ath9k-phy1 -> ../../devices/pci0000:00/0000:00:02.1/0000:01:00.0/leds/ath9k-phy1
lrwxrwxrwx 1 root root 0 Nov 16 20:08 mmc0:: -> ../../devices/pci0000:00/0000:00:14.7/leds/mmc0::
% sudo rmmod leds_gpio 
% ls -l /sys/class/leds
total 0
lrwxrwxrwx 1 root root 0 Nov 16 20:13 ath9k-phy1 -> ../../devices/pci0000:00/0000:00:02.1/0000:01:00.0/leds/ath9k-phy1
lrwxrwxrwx 1 root root 0 Nov 16 20:08 mmc0:: -> ../../devices/pci0000:00/0000:00:14.7/leds/mmc0::
% sudo modprobe leds_gpio                                 
% ls -l /sys/class/leds  
total 0
lrwxrwxrwx 1 root root 0 Nov 16 21:15 apu2:green:led1 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led1
lrwxrwxrwx 1 root root 0 Nov 16 21:15 apu2:green:led2 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led2
lrwxrwxrwx 1 root root 0 Nov 16 21:15 apu2:green:led3 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led3
lrwxrwxrwx 1 root root 0 Nov 16 21:15 apu:green:1 -> ../../devices/platform/leds-gpio/leds/apu:green:1
lrwxrwxrwx 1 root root 0 Nov 16 21:15 apu:green:2 -> ../../devices/platform/leds-gpio/leds/apu:green:2
lrwxrwxrwx 1 root root 0 Nov 16 21:15 apu:green:3 -> ../../devices/platform/leds-gpio/leds/apu:green:3
lrwxrwxrwx 1 root root 0 Nov 16 20:13 ath9k-phy1 -> ../../devices/pci0000:00/0000:00:02.1/0000:01:00.0/leds/ath9k-phy1
lrwxrwxrwx 1 root root 0 Nov 16 20:08 mmc0:: -> ../../devices/pci0000:00/0000:00:14.7/leds/mmc0::
```
I'm using the following configuration and can control the LEDs with either set of entries.
```
% cat /etc/tmpfiles.d/leds.conf 
#Power on/off
w /sys/class/leds/apu2:green:led1/brightness - - - - 1
#WAN activity
w /sys/class/leds/apu2:green:led2/trigger - - - - netdev
w /sys/class/leds/apu2:green:led2/device_name - - - - wan0
w /sys/class/leds/apu2:green:led2/tx - - - - 1
w /sys/class/leds/apu2:green:led2/rx - - - - 1
#LAN activity
w /sys/class/leds/apu2:green:led3/trigger - - - - netdev
w /sys/class/leds/apu2:green:led3/device_name - - - - br0
w /sys/class/leds/apu2:green:led3/tx - - - - 1
w /sys/class/leds/apu2:green:led3/rx - - - - 1
```
I've blacklisted the old `leds_apu` module. 
```
% lsmod | grep -E "(led|apu|amd|gpio)"           
leds_gpio              16384  0
amd64_edac_mod         36864  0
edac_mce_amd           32768  1 amd64_edac_mod
kvm_amd               106496  0
kvm                   770048  1 kvm_amd
pcengines_apuv2        16384  0
gpio_keys_polled       16384  0
input_polldev          20480  1 gpio_keys_polled
gpio_amd_fch           16384  3
ccp                    98304  1 kvm_amd
gpio_keys              20480  0
pinctrl_amd            32768  5
ledtrig_netdev         16384  0
```
```
% uname -a
Linux apu2c4 5.2.14-arch2-1-ARCH #1 SMP PREEMPT Thu Sep 12 10:42:38 UTC 2019 x86_64 GNU/Linux
```

pietrushnic, 2019-11-22:
@miczyg1 I assume this is because of ACPI changes?

miczyg1, 2020-01-07:
@chr0mag you are free to use either of the sysfs entries. SInce v4.10.0.0 we have added an ACPI support for GPIO controller which lets control all GPIOs, not only those from leds_apu. apu boards also have sim swaps, PCI Express reset etc.

The only problem you can experience is https://github.com/pcengines/coreboot/issues/329
which can be solved by https://github.com/pcengines/apu2-documentation/blob/master/docs/gpios.md#known-issue

Anyway, on Linux this is not a problem since anybody can blacklist the old apu driver. Added some documentation about that in the issue section of GPIO documentation linked above.

closed 2020-01-07 by miczyg1

chr0mag, 2020-01-12:
> ...on Linux this is not a problem since anybody can blacklist the old apu driver.

@miczyg1 - are you saying that by blacklisting the `leds_apu` module I shouldn't see the `apu:green:Y` sysfs entries? I still see both sets of sysfs entries even with this old driver blacklisted, but perhaps I've misunderstood?

miczyg1, 2020-01-13:
@chr0mag have you double-checked with `lsmod` that the `leds_apu` is not loaded?

chr0mag, 2020-01-14:
```
% lsmod | grep led             
gpio_keys_polled       16384  0
input_polldev          20480  1 gpio_keys_polled
leds_gpio              16384  0
ledtrig_netdev         16384  0
```
```
% cat /etc/modprobe.d/blacklist.conf 
blacklist bluetooth
blacklist kvm
blacklist pcspkr
blacklist leds_apu
```
Full `lsmod` output below.
```
Module                  Size  Used by
dm_crypt               53248  1
dm_mod                159744  2 dm_crypt
cmac                   16384  4
ccm                    20480  24
bridge                217088  0
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
nft_masq               16384  1
nft_chain_nat          16384  2
nf_nat                 49152  2 nft_masq,nft_chain_nat
nft_reject_ipv4        16384  3
nf_reject_ipv4         16384  1 nft_reject_ipv4
nft_reject             16384  1 nft_reject_ipv4
nft_counter            16384  11
nft_ct                 20480  6
nf_conntrack          163840  3 nf_nat,nft_ct,nft_masq
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
libcrc32c              16384  2 nf_conntrack,nf_nat
nf_tables_set          32768  2
nf_tables             167936  64 nft_reject_ipv4,nft_ct,nft_counter,nft_masq,nft_chain_nat,nf_tables_set,nft_reject
nfnetlink              16384  1 nf_tables
ath10k_pci             65536  0
ath10k_core           598016  1 ath10k_pci
ath9k                 180224  0
amd64_edac_mod         32768  0
edac_mce_amd           32768  1 amd64_edac_mod
kvm_amd               110592  0
ath9k_common           40960  1 ath9k
ath9k_hw              512000  2 ath9k_common,ath9k
kvm                   790528  1 kvm_amd
ath                    36864  4 ath9k_common,ath9k,ath9k_hw,ath10k_core
mac80211              991232  3 ath9k,ath9k_hw,ath10k_core
irqbypass              16384  1 kvm
crct10dif_pclmul       16384  1
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
aesni_intel           372736  22
pcengines_apuv2        16384  0
cfg80211              851968  5 ath9k_common,ath9k,ath,mac80211,ath10k_core
gpio_keys_polled       16384  0
crypto_simd            16384  1 aesni_intel
input_polldev          20480  1 gpio_keys_polled
gpio_amd_fch           16384  3
cryptd                 24576  3 crypto_simd,ghash_clmulni_intel
glue_helper            16384  1 aesni_intel
igb                   253952  0
sp5100_tco             20480  0
i2c_piix4              28672  0
fam15h_power           16384  0
k10temp                16384  0
ccp                    98304  1 kvm_amd
rfkill                 28672  4 cfg80211
i2c_algo_bit           16384  1 igb
libarc4                16384  1 mac80211
uio_pdrv_genirq        16384  0
dca                    16384  1 igb
rng_core               16384  1 ccp
evdev                  24576  2
uio                    20480  1 uio_pdrv_genirq
leds_gpio              16384  0
pinctrl_amd            32768  5
mac_hid                16384  0
acpi_cpufreq           28672  0
coreboot_table         16384  0
ledtrig_netdev         16384  0
ip_tables              36864  0
x_tables               49152  1 ip_tables
ext4                  782336  1
crc32c_generic         16384  0
crc16                  16384  1 ext4
mbcache                16384  1 ext4
jbd2                  135168  1 ext4
sd_mod                 57344  4
uas                    28672  1
usb_storage            77824  1 uas
ahci                   40960  1
sdhci_pci              57344  0
libahci                40960  1 ahci
cqhci                  32768  1 sdhci_pci
xhci_pci               20480  0
sdhci                  69632  1 sdhci_pci
libata                274432  2 libahci,ahci
crc32c_intel           24576  3
mmc_core              184320  3 sdhci,cqhci,sdhci_pci
xhci_hcd              282624  1 xhci_pci
scsi_mod              249856  4 sd_mod,usb_storage,uas,libata
ehci_pci               20480  0
ehci_hcd               98304  1 ehci_pci
gpio_keys              20480  0
```

miczyg1, 2020-01-14:
@chr0mag hmm, the outputs look good. How about the sysfs? Could you paste the double entries as well?

chr0mag, 2020-01-18:
```
% ls -l /sys/class/leds/            
total 0
lrwxrwxrwx 1 root root 0 Jan 12 13:12 apu2:green:led1 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led1
lrwxrwxrwx 1 root root 0 Jan 12 13:12 apu2:green:led2 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led2
lrwxrwxrwx 1 root root 0 Jan 12 13:12 apu2:green:led3 -> ../../devices/pci0000:00/PRP0001:02/leds/apu2:green:led3
lrwxrwxrwx 1 root root 0 Jan 12 13:13 apu:green:1 -> ../../devices/platform/leds-gpio/leds/apu:green:1
lrwxrwxrwx 1 root root 0 Jan 12 13:13 apu:green:2 -> ../../devices/platform/leds-gpio/leds/apu:green:2
lrwxrwxrwx 1 root root 0 Jan 12 13:13 apu:green:3 -> ../../devices/platform/leds-gpio/leds/apu:green:3
lrwxrwxrwx 1 root root 0 Jan 12 13:13 ath9k-phy0 -> ../../devices/pci0000:00/0000:00:02.1/0000:01:00.0/leds/ath9k-phy0
lrwxrwxrwx 1 root root 0 Jan 12 13:12 mmc0:: -> ../../devices/pci0000:00/0000:00:14.7/leds/mmc0::
```

jluebbe, 2020-09-05:
The apu:green:* LEDs are added by the pcengines_apuv2 platform driver: https://github.com/torvalds/linux/blob/master/drivers/platform/x86/pcengines-apuv2.c