pcengines/coreboot #356: Issue booting OpenBSD 6.6 (current) on v4.11.0.1

drduh, 2019-12-12:
I can't boot my APU4 after upgrading firmware to v4.11.0.1. Console output:

```console
pcib0 at pci0 dev 20 function 3 "AMD Hudson-2 LPC" rev 0x11
sdhc0 at pci0 dev 20 function 7 "AMD Bolton SD/MMC" rev 0x01: can't map interrupt
pchb2 at pci0 dev 24 function 0 "AMD AMD64 16h Link Cfg" rev 0x00
pchb3 at pci0 dev 24 function 1 "AMD AMD64 16h Address Map" rev 0x00
pchb4 at pci0 dev 24 function 2 "AMD AMD64 16h DRAM Cfg" rev 0x00
km0 at pci0 dev 24 function 3 "AMD AMD64 16h Misc Cfg" rev 0x00
pchb5 at pci0 dev 24 function 4 "AMD AMD64 16h CPU Power" rev 0x00
pchb6 at pci0 dev 24 function 5 "AMD AMD64 16h Misc Cfg" rev 0x00
isa0 at pcib0
isadma0 at isa0
com0 at isa0 port 0x3f8/8 irq 4: ns16550a, 16 byte fifo
com0: console
com1 at isa0 port 0x2f8/8 irq 3: ns16550a, 16 byte fifo
com2 at isa0 port 0x3e8/8 irq 5: ns16550a, 16 byte fifo
intr_establish: pic ioapic0 pin 5: can't share type 3 with 2
pcppi0 at isa0 port 0x61
spkr0 at pcppi0
lpt0 at isa0 port 0x378/4 irq 7
wbsio0 at isa0 port 0x2e/2: NCT5104D rev 0x53
vmm0 at mainbus0: SVM/RVI
ehci_sync_hc: tsleep() = 35
ehci_sync_hc: tsleep() = 35
uhub1: device problem, disabling port 1
vscsi0 at root
scsibus1 at vscsi0: 256 targets
softraid0 at root
scsibus2 at softraid0: 256 targets
root device:
use one of: exit em0 em1 em2 em3
root device:
use one of: exit em0 em1 em2 em3
root device:
use one of: exit em0 em1 em2 em3
root device:
use one of: exit em0 em1 em2 em3
root device: exit
syncing disks... done
```

Reverting back to v4.10.0.3 gets it back in working order:

```console
vmm0 at mainbus0: SVM/RVI
uhub2 at uhub1 port 1 configuration 1 interface 0 "Advanced Micro Devices product 0x7900" rev2
vscsi0 at root
scsibus2 at vscsi0: 256 targets
softraid0 at root
scsibus3 at softraid0: 256 targets
pf enabled
ddb.panic: 1 -> 0
net.inet.ip.forwarding: 0 -> 1
starting network
```

Is there anything I need to disable or toggle to use the latest firmware?

pietrushnic, 2019-12-16:
@miczyg1 any ideas?

miczyg1, 2019-12-16:
OPNSense seem to be complaining on USB and that it could not find rootfs... Is this USB installation?

Hard to tell without reproduction.

drduh, 2019-12-16:
OpenBSD is running on a single SSD, the model and mSATA slot hasn't changed in years.

miczyg1, 2019-12-19:
@drduh booted an OpenBSD 6.6 install image for USB and experiencing an issue where SD card and SATA disks are not detected. The problem seems to be lying in the interrupt configuration, see:

```
ahci0 at pci0 dev 17 function 0 "AMD Hudson-2 SATA" rev 0x40: unable to map interrupt
...
sdhc0 at pci0 dev 20 function 7 "AMD Bolton SD/MMC" rev 0x01: can't map interrupt
```

I am on the track and will resolve it till the end of year

miczyg1, 2019-12-19:
@drduh coreboot strives to be ACPI compliant and one commit introduced this bug. Pushed a fix to the coreboot repository: https://review.coreboot.org/c/coreboot/+/37835
so you should enjoy working v4.11.0.2 firmware by the end of year. Weird that we haven't catched that in validation with Linux and pfSense installed on SATA disk...

stillundecided, 2020-01-04:
@miczyg1 I can confirm, that 4.11.0.1 did not work with OpenBSD 6.6, but 4.11.0.2 is working fine again.
The .zip packages for 4.11.0.2 on https://pcengines.github.io are the wrong ones or at least named the wrong way.

sthen, 2020-01-06:
> Weird that we haven't catched that in validation with Linux and pfSense installed on SATA disk...

BTW: OpenBSD doesn't use Intel's acpica (like Linux/FreeBSD/NetBSD), rather it has its own ACPI implementation. It's fairly mature by now - it isn't bad, just different - so it might be useful to add an OpenBSD test if you're wanting to validate changes to ACPI.

miczyg1, 2020-01-07:
I see, thank you @sthen . We have a wide range of tests for BSD systems on the roadmap.

miczyg1, 2020-01-07:
@stillundecided indeed, links haven't been updated. Fixed.

Thank you all for the help and reports. Closing the issue.

closed 2020-01-07 by miczyg1