pcengines/coreboot #358: BIOS/Coreboot Upgrade fails

virse, 2019-12-13:
Hey there,

I tried to upgrade my APU2 with newest Mainline 4.11.0.1-ROM.

But the upgrade isn't working and I get this message:

```
# flashrom -w apu2_v4.11.0.1.rom -p internal
flashrom v1.0.1 on Linux 4.14.154-ipfire (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0x77fce000.
Found chipset "AMD FCH".
Enabling flash write... OK.
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
This coreboot image (PC Engines:apu2) does not appear to
be correct for the detected mainboard (PC Engines:PC Engines apu2).
Aborting. You can override this with -p internal:boardmismatch=force.
```
Any help is greatly appreciated. Thx.

virse

michaelsteinmann, 2019-12-13:
Hi,

Use the parameter ' -p internal:boardmismatch=force '

Best regards
Michael Steinmann

virse, 2019-12-13:
Hi Michael,

thx for your reply.

I am a little bit afraid to use the parameter. I don't want to break and brike  my apu2 due to the message. Is it safe to use the command with the parameter?

BR, virse

pietrushnic, 2019-12-15:
@virse there is no way of doing that from old firmware version, you should not be concerned if:
1. You know board model and you have correct firmware downloaded
2. You checked binary integrity before flashing
If anything will go really wrong (e.g. power fail during update) you can always order a quite cheap recovery dongle: https://www.pcengines.ch/spi1a.htm

franz, 2019-12-26:
@virse FWIW, i've just updated my APU 2c4 (running the original firmware from 2016) with the 4.10.0.3 mainline firmware. Had to use the "-p internal:boardmismatch=force" option. The 1st issue i ran into was that boot order was reset, and the board tried to boot from USB drive. The 2nd issue was [this](https://github.com/pcengines/coreboot/issues/206), but that's a Qualcomm bug, not an APU bug. But anyway i recommend an RS232 cable to fix boot order.

miczyg1, 2020-01-07:
@franz the bootorder is not preserved yet: https://github.com/pcengines/coreboot/issues/211

miczyg1, 2020-01-07:
The board mismatch warning is also explained here: https://github.com/pcengines/apu2-documentation/blob/ff3ac87e4e3ade914b42eac021e470356ef617c8/docs/firmware_flashing.md#motherboard-mismatch-warning  (also available on pcengines.github.io)
Closing.

closed 2020-01-07 by miczyg1