pcengines/coreboot #365: APU(x) does not boot after soft-restart(crash)

TimelessNL, 2020-01-14:
At home I've been testing OpenWrt v19.07.0 on an AP2 using bios v4.11.0.2. And due to unknown reasons (_no serial output collected yet_) OpenWrt reboots(_crashes_) after 1-2 days, this itself is not a direct problem but the APU does not recover during this reboot but instead freezes with only the middle LED faintly lit and no further progress.
[image]
Before I was running OpenWrt v18.06.0 and bios v4.9.0.3 just fine, stable and no reboots.
So I'm going to revert back to v4.9.03 to see if OpenWrt v19.07.0 remains stable. Maybe if OpenWrt v18.06.0 had rebooted(crashed) more often it could also trigger the reboot issue, but since it ran stable I had not seen any indication of it.

_Also at work I've seen quite some APU2/3's on both Legacy/Mainline BIOS versions that had similar soft-reboot(crash) problems like described above, after reboot the BIOS seems to be able to turn all LED off like it normally does and then should turn LED1 on, but instead all LEDs remain off and it stops doing anything meaningful. We're still investigating this issue, as it may be unrelated but I wanted to mention it anyway._

pietrushnic, 2020-01-14:
@artur-rs do we have any tests for OpenWRT? If not maybe those should be added?

TimelessNL, 2020-01-14:
I personally don't think this is a OpenWrt issue, but just a coincidence and that the real trigger is the reboot itself. At work we don't use OpenWrt and still have a similar outcome.

miczyg1, 2020-01-14:
@TimelessNL thank you for the report. All of our tests run on real hardware and we have a few continuous reboot tests that loop 50 times from firmware to operating system and reboot.

Typically we encountered situations where all LEDs were either on or off (no single dimmed LED yet). Anyway, we will try to reproduce and see what is going on.

TimelessNL, 2020-01-14:
@miczyg1 Ah so this issue has already been reproduced in your testframework?

If it helps I can also supply my specific OpenWrt build, if the v4.9.0.3 downgrade does not solve the OpenWrt crash/reboot.

miczyg1, 2020-01-14:
@TimelessNL that would be even better :) I could flash the image on disk and flash debug BIOS and start testing immediately to gather information.

As for watchdog, we noticed recently that we have been enabling the watchdog too early and it sometimes caused hangs too... Since v4.11.0.2 it should be allright.

TimelessNL, 2020-01-14:
> @TimelessNL that would be even better :) I could flash the image on disk and flash debug BIOS and start testing immediately to gather information.

Sure, I've created my custom images with the [OpenWrt imagebuilder](https://downloads.openwrt.org/releases/19.07.0/targets/x86/64/openwrt-imagebuilder-19.07.0-x86-64.Linux-x86_64.tar.xz) using:

`make image PROFILE=Generic PACKAGES="-hostapd-common -wpad-basic -dnsmasq -ppp -ppp-mod-pppoe -dnsmasq dnsmasq-full wpad-openssl libc uhttpd luci-mod-admin-full luci-theme-bootstrap luci-app-firewall libiwinfo-lua luci-proto-ipv6 rpcd-mod-rrdns libustream-mbedtls px5g luci-app-opkg luci-app-uhttpd htop nano block-mount kmod-usb-storage kmod-leds-apu2 kmod-gpio-nct5104d kmod-fs-ext4 kmod-scsi-core kmod-usb2 kmod-usb3 usbutils pciutils kmod-usb-ledtrig-usbport kmod-ledtrig-heartbeat fdisk e2fsprogs luci-app-statistics collectd-mod-disk collectd-mod-uptime collectd-mod-conntrack collectd-mod-df collectd-mod-disk collectd-mod-dns collectd-mod-sensors lm-sensors lm-sensors-detect collectd-mod-cpufreq screen ipset iputils-traceroute6 luci-app-adblock adblock luci-app-banip banip wget ca-certificates luci-app-nut nut-upsmon"`

Resulting in these files that I use:
[openwrt-19.07.0-x86-64-combined-squashfs.img.gz](https://github.com/pcengines/coreboot/files/4060147/openwrt-19.07.0-x86-64-combined-squashfs.img.gz)
[openwrt-19.07.0-x86-64-generic.manifest](https://github.com/pcengines/coreboot/files/4060151/openwrt-19.07.0-x86-64-generic.manifest.txt)

Heavy processing like adblock & banip (dns/ip blocks) are refreshed at 04:00 and 05:00 (_midnight_) though a cronjob and I did not experience any reboot/crash during that time. So I don't think those are related. I have collectd running that collects system data. But that's about it.

> As for watchdog, we noticed recently that we have been enabling the watchdog too early and it sometimes caused hangs too... Since v4.11.0.2 it should be allright.

Ah, but I was running v4.11.0.2 since my OpenWrt v19.07.0 update, so this should not apply I guess.

_Ps. I'm pleasantly surprised by the adequate and quick responses, thanks! And keep up the good work 😃_

TimelessNL, 2020-01-14:
Ok, I've been testing v4.9.0.3 and installed `kmod-sp5100_tco` to enable the watchdog. And at first it seems fine, the watchdog chip is detected and triggers after 60sec when `ubus call system watchdog '{ "stop": true }'` has been executed. However after some tries (_2_) the BIOS seemed to hang and all LEDs are ON, just like you mentioned @miczyg1. I tested 6 times more and 2 gave me similar results, all LEDs ON and BIOS stops doing anything. I also tried to reboot OpenWrt normally 10 times but it rebooted just fine.

Strange part is, when I tested v4.11.0.2 I did not have the watchdog enabled (installed). So I don't think my 19.07.0 reboots are watchdog trigger related. Because of this I still don't have any clue to why it still manages to freeze during reboot. I will keep this v4.9.0.3 installed the next couple of days to see how stable OpenWrt 19.07.0 actually is. Since this APU2 is also our main firewall I can't run continuous tests on it :P.

miczyg1, 2020-01-15:
@TimelessNL it is exactly as you say. It is not necessarily watchdog related. The typical watchdog-related hang caused all the LEDs on. The reboot hang has different indicators and possibly different reasons as well. Thank you for the image, I will proceed with some reboot tests on it.

TimelessNL, 2020-01-15:
Ok just got notified that my firewall was inaccessible again (all 3 LEDs on) while running v4.9.0.3. So this seems to indicate that 1. OpenWrt 19.07.0 with my installed packages is unstable. And 2. V4.9.0.3 did also have reboot issues while I was running OpenWrt v18.06.0 but since it never rebooted I didn't notice any stability issues.

TimelessNL, 2020-01-21:
@miczyg1 Any results from testing APU2 units?

**I've been testing these configurations so far:**

1. `OpenWrt 19.0.7.0 (custom) & BIOS v4.9.0.3` (_Sandisk 32GB SD_)
2. `OpenWrt 19.0.7.0 (custom) & BIOS v4.11.0.2` (_PC Engines pSLC 4GB SD_)
3. `OpenWrt 19.0.7.0 (vanilla) & BIOS v4.11.0.2` (_PC Engines pSLC 4GB SD_)
4. `OpenWrt SNAPSHOT  r12053-9e799f3eee (custom) & BIOS v4.11.0.2` (_PC Engines pSLC 4GB SD_)

OpenWrt 19.07.0 uses kernel 4.14 and OpenWrt SNAPSHOT uses 4.19 so I was hoping that may have magically fixed something. The newer kernel even shows this special APU2 log:
`[    1.130803] reboot: PC Engines apu2 series board detected. Selecting PCI-method for reboots.`

Neither of them is stable or successfully recovers after a reboot/crash, I tried to disable persistent writing to the SD card and run everything in RAM to rule out disk problems. I even switched SD cards to see if that may have anything to do with it.

I turned on persistent logging and attached a permanent serial cable to an external device where I may hopefully capture the reboot reason. **But other than that I'm out of options**.

miczyg1, 2020-01-28:
@TimelessNL sorry for the delay. As for now the apu2 survived 50 reboots with the following configuration:
- apu2 v4.11.0.2
- your OpenWRT 19.07 image flashed on SD card
- boot order 1st priority set to SD card

I will proceed with more testing with OpenWRT on mSATA. If that will not bring any results I will have to provide a debug binary for you. I hope the verbose serial output will not bother you.

TimelessNL, 2020-01-28:
Thanks for your response. But I think some sort of crash causes this reboot issue. After approx 6h-2day('s) the device will just freeze, with or without LEDs on (_sometimes only 1 on a random position_). When I manually reboot the APU2 it works just fine. I've been running some more tests since my last message.

**Test**:
5. `Openwrt 18.06.0 (custom) & BIOS v4.11.0.2` (_PC Engines pSLC 4GB SD_)
6. `Openwrt 18.06.0 (custom) & BIOS v4.9.0.3` (_Sandisk 32GB SD_) (Old PSU)
7. `Openwrt 18.06.0 (custom) & BIOS v4.9.0.3` (_Sandisk 32GB SD_) (New PSU)
8. `OpenWrt 19.0.7.0 (custom) & BIOS v4.11.0.2` (_PC Engines pSLC 4GB SD_) (_with serial cable and persistent logging_)

Tests **6/7** are especially interesting, I normally swap SD cards during OpenWrt updates (_this saves me time when an upgrade is unsuccessful and I want to revert back to the previous situation_). So I put the old card back, that SD card contained the exact (_stable_) data of my previous installation. and yet... for some unknown reason that install is not stable anymore. After 2day's it froze just like the others.

Test **8** did not show any interesting data, neither serial (_RS232_) nor persistent OpenWrt logging showed a crash or kernel panic:
```
[   18.920825] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   18.930393] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   19.050781] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   19.060642] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   39.518716] random: crng init done
--POWER CYCLE--
  PC Engines apu2
coreboot build 20193012
BIOS version v4.11.0.2
4080 MB ECC DRAM
```

**So my current test is**
1. `Replaced production APU2C4 with APU3C2.`
2. `Running APU2C4 on a test banch in situation` **7** `without network cables attached.`
_This to see if the APU2 will run stable outside of my live environment and if the APU3 shows the same symptoms in the live environment._

miczyg1, 2020-01-29:
@TimelessNL how many amps does your PSU give output? Is that PSU supplying only the APU?

TimelessNL, 2020-01-29:
Sorry, I should have mentioned that. I've been using the original PC Engines distributed PSU's. Both the First generation (bulky) and [Second](https://www.pcengines.ch/ac12veur3.htm) (smaller) 12v 2A power supplies directly from PC Engines themselves.

However I did notice (_like with any SMPS without GND/Earth connection_) that there is a slight electrical tingle while touching the metal case while the APU is powered, most likely due to leakage current from the Y-class caps in the SMPS. However because I've been using FTP (UTP with shielding) cable-ling this 'leakage' current is drained by my main switch that is properly grounded. Normally a device should be able to sustain that usage, and since the RJ45 outlets on the APU are metal I see no reason to really doubt that. But maybe this may have caused premature failure over time.

As for now both my APU2 and APU3 in their current test situation are still working fine, almost hit the 3-day uptime mark, which I have not been seeing in the last few weeks.

TimelessNL, 2020-02-08:
> As for now both my APU2 and APU3 in their current test situation are still working fine, almost hit the 3-day uptime mark, which I have not been seeing in the last few weeks.

Ok, my APU2 crashed some days ago (_uptime +/-9days_). This by just sitting on my desk without any cables other than power attached. So running idle and end result is the same all LEDs OFF. This while my new APU3 that replaced the APU2 as my firewall/gateway is still running strong and almost hitting 13days uptime