pcengines/coreboot #366: ath9k occasionally panics, fails

drduh, 2020-01-17:
About once per week, I experience an issue on my APU4C4 (Debian 9, latest firmware): wireless (hostapd) functionality completely stops working. dmesg output:

```
[702963.606524] irq 63: nobody cared (try booting with the "irqpoll" option)
[702963.613332] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.9.0-11-amd64 #1 Debian 4.9.189-3+deb9u2
[702963.622110] Hardware name: PC Engines apu4/apu4, BIOS v4.11.0.2 12/30/2019
[702963.629073]  0000000000000000 ffffffff93936404 ffff896a94484200 ffff896a944842d4
[702963.636660]  ffffffff936d9b10 ffff896a94484200 0000000000000000 ffffffff94319500
[702963.644245]  00000000000000c4 ffffffff936d9e9b ffff896a94484200 0000000000000000
[702963.651830] Call Trace:
[702963.654375]  <IRQ> 
[702963.656410]  [<ffffffff93936404>] ? dump_stack+0x5c/0x78
[702963.661821]  [<ffffffff936d9b10>] ? __report_bad_irq+0x30/0xc0
[702963.667748]  [<ffffffff936d9e9b>] ? note_interrupt+0x23b/0x280
[702963.673676]  [<ffffffff936d6fe1>] ? handle_irq_event_percpu+0x51/0x70
[702963.680208]  [<ffffffff936d7039>] ? handle_irq_event+0x39/0x60
[702963.686130]  [<ffffffff936da577>] ? handle_fasteoi_irq+0xa7/0x170
[702963.692319]  [<ffffffff93628fbf>] ? handle_irq+0x1f/0x30
[702963.697726]  [<ffffffff93c2112e>] ? do_IRQ+0x4e/0xe0
[702963.702782]  [<ffffffff93c1ecde>] ? common_interrupt+0x9e/0x9e
[702963.708705]  <EOI> 
[702963.710730]  [<ffffffff936fb8cf>] ? tick_nohz_idle_enter+0x3f/0x70
[702963.717010]  [<ffffffff936fb8c8>] ? tick_nohz_idle_enter+0x38/0x70
[702963.723280]  [<ffffffff936bf20e>] ? cpu_startup_entry+0x9e/0x240
[702963.729381]  [<ffffffff9364ac60>] ? start_secondary+0x170/0x1b0
[702963.735392] handlers:
[702963.737771] [<ffffffffc0794a20>] ath_isr [ath9k]
[702963.742507] Disabling IRQ #63
```

Rebooting APU gets it working again. Could it be a hardware problem? Thanks.

sp4rkie, 2020-01-30:
I've got similar problems with 'Qualcomm Atheros AR958x 802.11abgn', up-to-date Debian 10 (kernel 4.19.0-6-amd64).
In my case introduction of 'enabled IOMMU' appears to be responsible for that.
Since only firmwares <= apu2_v4.8.0.2.rom work without problems. All newer firmwares reveal the error.

drduh, 2020-05-08:
I have not; I'm running the latest and greatest firmware. Here's another trace that may be useful:

```
[58572.005837] irq 63: nobody cared (try booting with the "irqpoll" option)
[58572.012554] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.9.0-12-amd64 #1 Debian 4.9.210-1
[58572.020641] Hardware name: PC Engines apu4/apu4, BIOS v4.11.0.6 04/26/2020
[58572.027517]  0000000000000000 ffffffffa6736bfe ffff8c78250e7600 ffff8c78250e76d4
[58572.035010]  ffffffffa64d9bb0 ffff8c78250e7600 0000000000000000 ffffffffa7119400
[58572.042509]  00000000000000c4 ffffffffa64d9f3b ffff8c78250e7600 0000000000000000
[58572.050007] Call Trace:
[58572.052464]  <IRQ> 
[58572.054414]  [<ffffffffa6736bfe>] ? dump_stack+0x66/0x88
[58572.059737]  [<ffffffffa64d9bb0>] ? __report_bad_irq+0x30/0xc0
[58572.065576]  [<ffffffffa64d9f3b>] ? note_interrupt+0x23b/0x280
[58572.071418]  [<ffffffffa64d7081>] ? handle_irq_event_percpu+0x51/0x70
[58572.077866]  [<ffffffffa64d70d9>] ? handle_irq_event+0x39/0x60
[58572.083699]  [<ffffffffa64da617>] ? handle_fasteoi_irq+0xa7/0x170
[58572.089795]  [<ffffffffa6428fbf>] ? handle_irq+0x1f/0x30
[58572.095115]  [<ffffffffa6a20f2e>] ? do_IRQ+0x4e/0xe0
[58572.100090]  [<ffffffffa6a1eade>] ? common_interrupt+0x9e/0x9e
[58572.105935]  [<ffffffffa64b7fc1>] ? update_sd_lb_stats+0x171/0x4b0
[58572.112119]  [<ffffffffa64b7f36>] ? update_sd_lb_stats+0xe6/0x4b0
[58572.118213]  [<ffffffffa64b833e>] ? find_busiest_group+0x3e/0x4d0
[58572.124314]  [<ffffffffa64b899c>] ? load_balance+0x1cc/0xa00
[58572.129975]  [<ffffffffa6481f1e>] ? irq_exit+0x3e/0xd0
[58572.135121]  [<ffffffffa6a20f37>] ? do_IRQ+0x57/0xe0
[58572.140088]  [<ffffffffa64b9894>] ? rebalance_domains+0x234/0x2c0
[58572.146181]  [<ffffffffa6a21ead>] ? __do_softirq+0x10d/0x2b0
[58572.151843]  [<ffffffffa6481fa2>] ? irq_exit+0xc2/0xd0
[58572.156989]  [<ffffffffa6a2192c>] ? smp_apic_timer_interrupt+0x4c/0x60
[58572.163515]  [<ffffffffa6a2005e>] ? apic_timer_interrupt+0x9e/0xb0
[58572.169691]  <EOI> 
[58572.171633]  [<ffffffffa64fb95f>] ? tick_nohz_idle_enter+0x3f/0x70
[58572.177826]  [<ffffffffa64fb958>] ? tick_nohz_idle_enter+0x38/0x70
[58572.184006]  [<ffffffffa64bf44e>] ? cpu_startup_entry+0x9e/0x240
[58572.190013]  [<ffffffffa644ad90>] ? start_secondary+0x170/0x1b0
[58572.195940] handlers:
[58572.198230] [<ffffffffc087da30>] ath_isr [ath9k]
[58572.202888] Disabling IRQ #63
```