pcengines/coreboot #367: apu2 USB3 ports inexistent in OS when booted with USB-serial device plugged-in

ventosus, 2020-01-26:
I have this curious issue (100% reproducible) that when I boot (cold and warm does not matter) my apu2 with a USB serial device plugged-in in one of the external USB-3 ports. The USB-3 ports are just inexistent in the OS thereafter.

Plugging in the USB serial device after boot works as intended, but is not practical for obvious reasons.

I have the problem with both newest mainline (v4.11.0.2) and legacy(v4.0.29) coreboot from [1].

Not sure whether this is an issue with coreboot or seabios. Anyone having similar issues ?

[1] https://pcengines.github.io/

Case 1: boot -> plug in USB serial device

    [root@apu ~]# lsusb -t
    /:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
        |__ Port 1: Dev 2, If 0, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 1: Dev 2, If 1, Class=CDC Data, Driver=cdc_acm, 12M
    /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
        |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M

Case 2: plug-in USB serial device -> boot (cold or warm).

    [root@apu ~]# lsusb -t
    /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480
        |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M

ventosus, 2020-01-26:
Case 1: boot -> plug in USB serial device

    [root@apu ~]# dmesg | grep -i usb
    [    0.545677] usbcore: registered new interface driver usbfs
    [    0.545702] usbcore: registered new interface driver hub
    [    0.545743] usbcore: registered new device driver usb
    [    1.644748] usbcore: registered new interface driver usbserial_generic
    [    1.644802] usbserial: USB Serial support registered for generic
    [    2.123330] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
    [    2.139796] ehci-pci 0000:00:13.0: new USB bus registered, assigned bus number 1
    [    2.165475] ehci-pci 0000:00:13.0: USB 2.0 started, EHCI 1.00
    [    2.165880] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
    [    2.165887] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.165891] usb usb1: Product: EHCI Host Controller
    [    2.165896] usb usb1: Manufacturer: Linux 4.19.91-gnu-1-lts ehci_hcd
    [    2.165899] usb usb1: SerialNumber: 0000:00:13.0
    [    2.166456] hub 1-0:1.0: USB hub found
    [    2.209942] xhci_hcd 0000:00:10.0: new USB bus registered, assigned bus number 2
    [    2.218637] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
    [    2.218644] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.218648] usb usb2: Product: xHCI Host Controller
    [    2.218653] usb usb2: Manufacturer: Linux 4.19.91-gnu-1-lts xhci-hcd
    [    2.218657] usb usb2: SerialNumber: 0000:00:10.0
    [    2.219247] hub 2-0:1.0: USB hub found
    [    2.219767] xhci_hcd 0000:00:10.0: new USB bus registered, assigned bus number 3
    [    2.219805] xhci_hcd 0000:00:10.0: Host supports USB 3.0 SuperSpeed
    [    2.220236] usb usb3: We don't know the algorithms for LPM for this host, disabling LPM.
    [    2.220501] usb usb3: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 4.19
    [    2.220534] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.220538] usb usb3: Product: xHCI Host Controller
    [    2.220542] usb usb3: Manufacturer: Linux 4.19.91-gnu-1-lts xhci-hcd
    [    2.220546] usb usb3: SerialNumber: 0000:00:10.0
    [    2.221007] hub 3-0:1.0: USB hub found
    [    2.535264] usb 1-1: new high-speed USB device number 2 using ehci-pci
    [    2.735525] usb 1-1: New USB device found, idVendor=0438, idProduct=7900, bcdDevice= 0.18
    [    2.735533] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
    [    2.736987] hub 1-1:1.0: USB hub found

ventosus, 2020-01-26:
Case 2: plug-in USB serial device -> boot (cold or warm).

    [root@apu ~]# dmesg | grep -i usb
    [    0.548332] ACPI: bus type USB registered
    [    0.548479] usbcore: registered new interface driver usbfs
    [    0.548505] usbcore: registered new interface driver hub
    [    0.548546] usbcore: registered new device driver usb
    [    0.656960] pci 0000:00:10.0: quirk_usb_early_handoff+0x0/0x6cc took 16410 usecs
    [    1.664277] usbcore: registered new interface driver usbserial_generic
    [    1.664298] usbserial: USB Serial support registered for generic
    [    2.150758] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
    [    2.166513] ehci-pci 0000:00:13.0: new USB bus registered, assigned bus number 1
    [    2.194221] ehci-pci 0000:00:13.0: USB 2.0 started, EHCI 1.00
    [    2.194548] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
    [    2.194554] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.194559] usb usb1: Product: EHCI Host Controller
    [    2.194565] usb usb1: Manufacturer: Linux 4.19.91-gnu-1-lts ehci_hcd
    [    2.194594] usb usb1: SerialNumber: 0000:00:13.0
    [    2.195197] hub 1-0:1.0: USB hub found
    [    2.215920] xhci_hcd 0000:00:10.0: new USB bus registered, assigned bus number 2
    [    2.231998] xhci_hcd 0000:00:10.0: USB bus 2 deregistered
    [    2.557961] usb 1-1: new high-speed USB device number 2 using ehci-pci
    [    2.758354] usb 1-1: New USB device found, idVendor=0438, idProduct=7900, bcdDevice= 0.18
    [    2.758362] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
    [    2.759794] hub 1-1:1.0: USB hub found

ventosus, 2020-01-26:
Both USB2/3 show up in lspci in both cases.

    [root@apu 0000:00:10.0]# lspci | grep -i usb
    00:10.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB XHCI Controller (rev 11)
    00:13.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller (rev 39)

miczyg1, 2020-01-27:
@ventosus thank you for reporting the issue. What model is the USB serial device?

ventosus, 2020-01-30:
> Simple USB-TTL UART?

No, it was an STM32 uC implementing a CDC/ACM interface. As I have the firmware available for the latter, I've changed it to not try to enmerate before 30s into the boot process. Then it just works.

Hooking up simple USB-UART converter (FT232 from FTDI) works just fine.

So I guess the STM32 uC CDC/ACM firmware may be doing something strange (when trying to enumerate) early in the boot which lead to deactivation of the whole port.

I've circumvented the problem for know, thanks.

closed 2020-01-30 by ventosus