pcengines/coreboot #382: 4.11.0.3/4.11.0.4 stuck on low CPU frequency

310b8fb5, 2020-03-02:
I recently noticed that my apu2 running OpenBSD 6.6 seems to be stuck at the lowest frequency available to `apm` (600MHz). I have `apm` starting with the flag to autoscale the frequency. The initial behaviour seems to work as expected, but within 24 hours, it seems to be stuck at the lowest. I can roughly confirm that this isn't a display issue, as benchmarks are slowed down correspondingly. I do not see this behaviour on an apu1. When I configure `apm` to run at the highest frequency all the time (`-H`), it properly stays throttled to 1GHz.

This could be an OpenBSD issue. However, the frequency does scale properly on boot & I noticed that this was an issue with the firmware in the past.

I will be happy to file a bug with the OpenBSD guys as well, but wanted to post here first in case this was a regression. I'm also happy to provide any debugging output requested, but am unsure of what would be useful.

Thanks!

pietrushnic, 2020-03-02:
@310b8fb5 thanks for reporting. @lwcislo ideally if we would have this test in regression.

krystian-hebel, 2020-03-03:
@310b8fb5 just to be sure: did You power-cycled the platform after updating firmware? The bug we had in https://github.com/pcengines/coreboot/issues/196 required a full power cycle more than others.

310b8fb5, 2020-03-03:
@krystian-hebel I did power cycle the machine after the firmware update. If you would like, I can try to reproduce the issue again this afternoon to get a more exact idea of how long it takes for the frequency to become "stuck."

310b8fb5, 2020-03-04:
@krystian-hebel I just gave the machine a check after a reboot. The frequency is stuck low within 4 hours - ` 1:17AM  up  3:39, 1 user, load averages: 0.07, 0.02, 0.01`

310b8fb5, 2020-03-12:
@krystian-hebel Just a quick update - I downgraded to 4.10.0.3 and saw some weirdness, so disabled core performance boost and reflashed to 4.11.0.4. The system seems to be behaving better now with core performance boost disabled - I will report back if I see the problem again.

If I don't see the issue with boost disabled, when OpenBSD 6.7 comes out around April/May I will re-enable and see what happens.

310b8fb5, 2020-05-19:
6.7 was released today, so I'll test again and see if I can narrow down what's causing the issue.

krystian-hebel, 2020-05-19:
Sorry, I didn't notice those replies earlier. Seems that Github sent e-mail only for the message in which I was **not** mentioned (!?)

Knowing that it may be somehow connected with CPB may help with narrowing down further diagnostics. It may be caused by some problems around core power gating, which is enabled as a prerequisite to CPB. Could you run the benchmark on 1/2/3/4 cores (after the frequency gets stuck) and see if the results are linearly proportional to the number of active cores? Preferably highest to lowest number of cores, to minimalise impact of thermal throttling.

310b8fb5, 2020-06-03:
@krystian-hebel quick update - I'm on 4.12.0.1 now, and seeing some inconsistent behavior with time trials and `apm` (the OpenBSD tool I use to observe cpu frequency). For example, `apm` might show 600Mhz (even when `apmd` is forced to high-performance mode), but a time trial runs as expected for 1000Mhz or vice versa. I think `apm` logic may be getting confused with multiple cores.

I'm going to close this issue for now and raise the problem on the OpenBSD mailing list (I'll update with a link for tracking purposes).

closed 2020-06-03 by 310b8fb5