pcengines/coreboot #387: Significant reduction in network performance after upgrading to v4.11.0.5

jstevensen, 2020-04-01:
I upgraded the BIOS on my PCEngines APU4B4 from v4.11.0.4 to v4.11.0.5 and experienced a 60-70% reduction in network traffic through the device. I also noticed an increased number of interface errors that I hadn't noticed before. 

I'm running OpenBSD 6.6 (amd64) with all errata to date applied.
Interfaces are Intel i211AT.
Switch is a UniFi Switch 8 port switch running the latest stable version, 4.0.80.10875.

I ran multiple rounds of iperf3 tests on between two separate devices in separate VLANs. 
4.11.0.5 average forwarding was around 160 Mbits/sec 
4.11.0.4 average forwarding was around 500 Mbits/sec

pietrushnic, 2020-04-01:
@jstevensen thank you for this report it is very important to avoid this kind of performance drops. If you can elaborate a little bit on how you perform those test it would be great. @lwcislo, please make sure regression will be extended with those test, so we can inform if the release has such problems.

@miczyg1 maybe those drops in performance are power consumption related?

jstevensen, 2020-04-01:
@pietrushnic In an effort to test my network forwarding speeds, I ran iperf3 without any special options on two separate computers. Each computer was on a separate VLAN. The APU4 running OpenBSD had a very simple testing PF ruleset that allowed all traffic between the two VLANs. I didn't run anything on the PCEngines APU4 itself. I did check for interface errors with netstat. (netstat -e -I em1; netstat -s -s) on the PCEngines APU4.

pietrushnic, 2020-04-01:
@jstevensen ok, thanks for that description we will try to create performance test as close to your environment as possible.

jstevensen, 2020-04-01:
PS. I'm not endorsing iperf3 as an ideal bandwidth testing tool, as there are a number of concerns with what it is actually testing and what that means. You may find better ways to do this. I do think testing the forwarding performance of an APU is important since most people are using these devices as networking devices.

mjenny, 2020-04-04:
You will have a similar throughput reduction on Linux based systems (tried it with OpenWRT 19.07.1).

miczyg1, 2020-04-09:
@jstevensen @mjenny due to enablement of PCIe ASPM the performance dropped in favor of power savings. We will make this option runtime configurable for those who want more power efficient device and do not need high performance. In the meantime the ASPM can be controlled from Linux kernel command line with `pcie_aspm.policy=performance` which will instruct Linux to not use ASPM. I haven't found yet any equivalent option for FreeBSD and derivative systems.

mjenny, 2020-04-10:
I couldn't find anything equicalent for FreeBSD either. I will therefore stick to 4.11.0.4 for the time being.
However, I am currently trying to figure out what I should use on my APU2E4. If I install IPFire or something similar I will definately try 4.11.0.5 with said kernel arg.

alexandervdm, 2020-04-14:
I also encountered this problem and was about to share my findings when I found this bug. 
My test setup is quite similar to @jstevensen but hopefully the details below will be helpful:

I have two different models - both are affected:
* apu2c2 (2Gb, 3x i211AT)
* apu4d4 (4Gb, 4x i211AT)

The apu is configured as an OpenBSD router with the minimal PF config required
to allow routing iperf3 traffic between the two networks:
```
  [client]        [L2 switch]        [server]
10.10.1.1/24 --->    ^   ^    <--- 10.10.2.2/24
                     |   |
                     |   |
           10.10.1.254   10.10.2.254
                    [router]
```

Test results using `iperf3 -O 1 -c server -P 1`
```
                                P=1             P=4
apu4d4 with  v4.9.0.2:          933-933         928-931
apu4d4 with v4.11.0.2:          927-932         927-928
apu4d4 with v4.11.0.5:          292-302         290-294

apu2c2 with  v4.9.0.2:          816-852
apu2c2 with v4.11.0.2:          820-854
apu2c2 with v4.11.0.4:          806-830
apu2c2 with v4.11.0.5:          292
```
Comparing OpenBSD dmesg sections between BIOS versions gave some hints:

```
bios0: vendor coreboot version "v4.11.0.4" date 02/26/2020
...
ioapic0 at mainbus0: apid 4 pa 0xfec00000, version 21, 24 pins
ioapic1 at mainbus0: apid 5 pa 0xfec20000, version 21, 32 pins, remapped
acpihpet0 at acpi0: 14318180 Hz
acpiprt0 at acpi0: bus 0 (PCI0)
acpiprt1 at acpi0: bus -1 (PBR4)
acpiprt2 at acpi0: bus 1 (PBR5)
acpiprt3 at acpi0: bus 2 (PBR6)
acpiprt4 at acpi0: bus 3 (PBR7)
acpiprt5 at acpi0: bus -1 (PBR8)
acpicpu0 at acpi0: C2(0@400 io@0x1771), C1(@1 halt!), PSS
acpicpu1 at acpi0: C2(0@400 io@0x1771), C1(@1 halt!), PSS
acpicpu2 at acpi0: C2(0@400 io@0x1771), C1(@1 halt!), PSS
acpicpu3 at acpi0: C2(0@400 io@0x1771), C1(@1 halt!), PSS
acpibtn0 at acpi0: PWRB
acpipci0 at acpi0 PCI0: 0x00000000 0x00000011 0x00000001
acpicmos0 at acpi0
"AMD0030" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"BOOT0000" at acpi0 not configured
cpu0: 998 MHz: speeds: 1000 800 600 MHz
pci0 at mainbus0 bus 0
```

```
bios0: vendor coreboot version "v4.11.0.5" date 03/29/2020
....
ioapic0 at mainbus0: apid 4 pa 0xfec00000, version 21, 24 pins
ioapic1 at mainbus0: apid 5 pa 0xfec20000, version 21, 32 pins
acpihpet0 at acpi0: 14318180 Hz
acpiprt0 at acpi0: bus 0 (PCI0)
acpiprt1 at acpi0: bus -1 (PBR4)
acpiprt2 at acpi0: bus 1 (PBR5)
acpiprt3 at acpi0: bus 2 (PBR6)
acpiprt4 at acpi0: bus 3 (PBR7)
acpiprt5 at acpi0: bus -1 (PBR8)
acpitz0 at acpi0: critical temperature is 115 degC
"ACPI0007" at acpi0 not configured
"ACPI0007" at acpi0 not configured
"ACPI0007" at acpi0 not configured
"ACPI0007" at acpi0 not configured
"ACPI0007" at acpi0 not configured
"ACPI0007" at acpi0 not configured
"ACPI0007" at acpi0 not configured
"ACPI0007" at acpi0 not configured
acpibtn0 at acpi0: PWRB
acpipci0 at acpi0 PCI0: 0x00000000 0x00000011 0x00000001
acpicmos0 at acpi0
"AMD0030" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"PRP0001" at acpi0 not configured
"BOOT0000" at acpi0 not configured
pci0 at mainbus0 bus 0
```

Had a quick try at disabling APM or related ACPI drivers using User Kernel Configuration mode that mostly made my kernel crash and in none of the cases improved network throughput. 

@miczyg1 If new power optimalizations are to blame, then surely they shouldn't be enabled by default given the huge performance hit?

miczyg1, 2020-04-30:
@alexandervdm the power optimization is runtime configurable now since v4.11.0.6 with default state disabled.

We did some measurements of the power consumption when the optimizations are enabled and the overall average gain is 0.5W in both idle and stressed states:
https://github.com/pcengines/apu2-documentation/blob/master/docs/research/power-consumption-tests.md

If one does not need full 1Gb bandwidth, this option may be useful.

alexandervdm, 2020-05-07:
@miczyg1 Thanks for the interesting read and your work. I will continue to track the coreboot releases.

closed 2020-06-05 by miczyg1

Legogris, 2020-06-12:
@miczyg1 Does this issue being closed mean that the problem was resolved? Do not see anything regarding this in the release notes.

miczyg1, 2020-06-16:
@Legogris if you would like to obtain more details about the release, sign up for the mailing list (https://pcengines.github.io/) or track the blog: https://pcengines.github.io/blog/

The fix was explained in `Key changes` (point 2):
https://pcengines.github.io/firmware/2020/04/28/PC-Engines-Firmware-v4-11-0-6.html

Legogris, 2020-06-16:
> @Legogris if you would like to obtain more details about the release, sign up for the mailing list (https://pcengines.github.io/) or track the blog: https://pcengines.github.io/blog/
>
> The fix was [...]

Thank you, I missed the blog, looks like there's stuff there not in the release notes :+1:

miczyg1, 2020-06-16:
True, I missed the SD 3.0 mode fix in the changelog back then. But other items are there, including the `PCIe power management features runtime configuration`