pcengines/coreboot #388: Regression: Complex WLE200NX wifi module not found after upgrade to v4.11.0.5 [APU2]

silentcreek, 2020-04-01:
Hi, 

I have an APU2C4 with two wireless modules: WLE200NX and WLE600VX by Complex. Yesterday, I upgraded the firmware from v4.11.0.2 to v4.11.0.5 and performed a cold bout (shut down, pulled the plug, waited a few seconds, put it back in and booted). After rebooting the WLE200NX wifi module did not come up again. The second wifi module worked fine. There was no error during boot (other than hostapd complaining about the missing device). And a second reboot did not make any difference. The command 'iw dev' does not show the WLE200NX device, either.

I downgraded to v4.11.0.4 and now both wifi modules come up again just fine. So, there seems to be a clear regression in v4.11.0.5.

My OS is Debian 10.

Regards, 

Timo

pietrushnic, 2020-04-01:
@silentcreek thank you for this report. Obviously we don't have resources to cover all hardware configuration so it is great that you reported that regression. Definitely some of the new features introduced could cause that behavior.

@miczyg1 let's take a look at this. @lwcislo let's check out hardware resources available if we can put together such configuration.

riptidewave93, 2020-04-03:
I did some testing at https://github.com/pcengines/apu2-documentation/issues/115#issuecomment-608446225 which shows that the mPCIe Slot 2 does not work as intended in BIOS v4.11.0.5

ldelouw, 2020-04-10:
Same here with APU2 and 01:00.0 Network controller: Qualcomm Atheros QCA986x/988x 802.11ac Wireless Network Adapter.

The adapter is not visible, I tried several cold boot, no success.

I needed to revert to v4.11.0.4

Thanks,

Luc

lukasmrtvy, 2020-04-14:
Can confirm, ( apu2c4 ), after upgrading to v4.11.0.5 mPCIe2 does not detect Wifi card.

miczyg1, 2020-04-16:
@lukasmrtvy in this case FOrce mPCIe2 clock will not help.

silentcreek, 2020-04-29:
Just installed 4.11.0.6 and can confirm the regression has been fixed. Both wifi cards are detected again. So, this issue can be closed. I won't close it myself, as I have one final suggestion.

@pietrushnic You said, you can't test every possible configuration and I absolutely understand that. But maybe you could add a test to verify that the second mPCIe slot on the APU2 works at all. In this particular case, such a test would have caught this issue as the tests of @riptidewave93 have shown.

miczyg1, 2020-04-30:
@silentcreek we have already changed the configuration, yet we still need to switch to apu2d or newer (should happen this month). Currently tested was apu2c.

closed 2020-06-05 by miczyg1