pcengines/coreboot #389: CPU freqency

pfsenseuser, 2020-04-01:
Hi,

on pfsense 2.4.5 freebsd 11.3 after upgrade to Bios 4.11.0.5 the command sysctl dev.cpu.0.freq
returns: sysctl: unknown oid 'dev.cpu.0.freq'.
It seams the sysctl dev.cpu tree is missing.

Regards,
pfsenseuser

miczyg1, 2020-04-01:
Hi,

Thank you for the report, we will look at it.

miczyg1, 2020-04-01:
@pfsenseuser it looks like I rushed things... FreeBSD 12.x detects CPU on ACPI correctly with v4.11.0.5. If pfSense 2.5.0 based on FreeBSD 12.0 is out it will work I guess.

Sorry about that. Will revert it on next release.

dugeem, 2020-04-03:
Thanks for investigating.

FYI I believe the FreeBSD 12.x ACPI CPU change was introduced two years ago:

https://svnweb.freebsd.org/base?view=revision&revision=326956

miczyg1, 2020-04-03:
According to FreeBSD release dates https://www.freebsd.org/releases/, this change should be present in FreeBSD 11.2 already. I will investigate more versions of FreeBSD and pfSense to determine what is wrong then.

lc200, 2020-04-03:
Also noticed the same thing on APU2E4, this issue makes itself immediately visible in the pfSense Web UI dashboard under CPU Type where the CPU frequency is no longer shown and sysctl dev.cpu.0.freq is returning nothing.

dugeem, 2020-04-05:
> According to FreeBSD release dates https://www.freebsd.org/releases/, this change should be present in FreeBSD 11.2 already. I will investigate more versions of FreeBSD and pfSense to determine what is wrong then.

After further inspection of the FreeBSD 11.x ACPI source (/release/11.x/sys/dev/acpica/acpi_cpu.c) it appears that revision 326956 was never MFCed to FreeBSD 11.x. 

As fas as I can tell my test APU2 running FW v4.11.0.5 and pfSense 2.4.5 (ie FreeBSD 11.3-STABLE) is working fine. Performance is as expected and CPB is working. Are there any potential issues from not having ACPI functionality?

miczyg1, 2020-04-06:
@dugeem it is all about compliance with ACPI specification. Official coreboot wants to update build tools and this functionality was blocking the update process.

dugeem, 2020-04-29:
Fixed with v4.11.0.6. Thanks!

closed 2020-04-30 by miczyg1