pcengines/coreboot #392: PCI addresses change depending on use of mPCIe slot, breaking persistant network interface naming

silentcreek, 2020-04-02:
Hi,

the regression I reported yesterday (#388) had the side effect of exposing another issue that I actually experienced a long time ago once. Back then, I had a suspicion of what was going on, but I didn't follow up on it. Now, combing through the logs of before and after the firmware upgrade I attempted to do two days ago, I can see my suspicion was correct.

**Issue**
When you populate the mPCIe slots on an APU2 board, the PCI addresses of other PCIe devices change, in particular the ethernet interfaces, which in turn leads to new ethernet device names if you use systemd with its default persistent interface naming scheme. On a gateway or router, this may cause serious problems.

**History**
Over a year ago, I added two wireless mPCIe modules to my APU2 setup. After doing so, my network was dysfunctional and I was locked out of my device (at least via SSH). When I hooked up a serial cable, I was baffled to see that my ethernet interfaces got new names (all their indexes were 1 higher than before). Hence, my firewall rules were broken, my PPPoE connection was broken and all o the services that bind to an interface failed. When I removed the wireless cards, everything worked as before. My workaround back then was to write my own udev rules to make sure my interface names are persistent (based on the driver and MAC address).

**Manifestation**
Now, with the attempted upgrade to v4.11.0.5 two days ago and the effect that my WLE200NX wifi module did not work anymore after the upgrade, I can actually see this in my logs. Before the upgrade (with both of my wireless modules operational) my network devices had the following PCI addresses:
ath9k 0000:01:00.0 (Compex WLE200NX)
igb 0000:02:00.0 (Intel i210AT NIC no. 1)
igb 0000:03:00.0 (Intel i210AT NIC no. 2)
igb 0000:04:00.0 (Intel i210AT NIC no. 3)
ath10k_pci 0000:05:00.0 (Compex WLE600VX)

Now, after booting with firmware v4.11.0.5 and my WLE200NX card dysfunctional, the addresses became these:
igb 0000:01:00.0
igb 0000:02:00.0
igb 0000:03:00.0
ath10k_pci 0000:04:00.0

Lucky me, this time it caused no issues for me because I still have my custom udev rules in place, making sure my interface names are fixed.

**Notes**
I'm not really sure whether this technically is a bug, i.e. whether PCI addresses may change depending on whether a slot is populated or not. But it is against user expectation (and apparently the expectations of systemd developers) that adding a PCIe module will change the interface names of other network devices. After all, it's not that we're adding new slots rather than merely populating one that was there before.

In addition, this issue is not related to v4.11.0.5 in any way other that it showed it as a side effect. When I first ran into that problem more than a year ago, I was obviously running a completely different version. And I'm not even sure, whether back than I was running mainline or legacy.

Regards,

Timo

miczyg1, 2020-04-02:
Hi @silentcreek .
I know this "issue" and the worst thing we cannot do anything about that. It also causes the wake-on-lan capable NIC to be the last enumerated instead of the first enumerated (and as a result iPXE doesn't try to boot from the WoL capable interface). For example in your case:

```
ath9k 0000:01:00.0 (Compex WLE200NX)         ----- PCIe Bridge 0000:00:02.1 (mPCIe 1 slot)
igb 0000:02:00.0 (Intel i210AT NIC no. 1)           ----- PCIe Bridge 0000:00:02.2 (NIC 2 according to schematics)
igb 0000:03:00.0 (Intel i210AT NIC no. 2)           ----- PCIe Bridge 0000:00:02.3 (NIC 1 according to schematics)
igb 0000:04:00.0 (Intel i210AT NIC no. 3)           ----- PCIe Bridge 0000:00:02.4 (NIC 0 with WoL according to schematics)
ath10k_pci 0000:05:00.0 (Compex WLE600VX) ----- PCIe Bridge 0000:00:02.5 (mPCIe 2 slot)
```
Linux is enumerating devices according the PCIe Bridge order, starting from 2.1.

We can reorder the numbering of bridges in the firmware to cause the desired enumeration, however, imagine all other apu owners and their confusion they would experience as you had experienced and described it here. I have talked to PC Engines team about it, but they are too afraid to change it after all these years of product lifetime on the market. And I completely understand them. The order of PCIe Bridges was already badly implemented by the initial authors of coreboot firmware in 2016.

The desired order would indeed look like:
```
igb 0000:01:00.0 (Intel i210AT NIC no. 1)           ----- PCIe Bridge 2.1 (NIC 0 with WoL according to schematics)
igb 0000:02:00.0 (Intel i210AT NIC no. 2)           ----- PCIe Bridge 2.2 (NIC 1 according to schematics)
igb 0000:03:00.0 (Intel i210AT NIC no. 3)           ----- PCIe Bridge 2.3 (NIC 2 according to schematics)
ath9k 0000:04:00.0 (Compex WLE200NX)         ----- PCIe Bridge 2.4 (mPCIe 1 slot)
ath10k_pci 0000:05:00.0 (Compex WLE600VX) ----- PCIe Bridge 2.5 (mPCIe 2 slot)
```

miczyg1, 2020-04-02:
Additionally, the apu2 is a bit unique, because of 2 PCIe slots. If the mPCIe module on PCIe Bridge 2.4 is not detected it will cause different enumeration for mPCIe module on PCIe Bridge 2.5. This configuration would be really benefitable only for apu3 and apu4 which have only 1 mPCIe slot (with PCIe lanes).

silentcreek, 2020-04-02:
Hi @miczyg1,

I see. In this case I guess the best thing to do would be to document this behavior and warn users so they are aware of the issues this may cause. Personally, I think it would be ok to change this behavior with a firmware upgrade. After all, upgrading the firmware is a manual process and if you were to put a big fat warning on the download page, I'd assume people could take precautions. Nevertheless, I also do understand the viewpoint of a manufacturer not to change such behavior. So, I guess documenting it, is the only way.

Thanks,

Timo

michaelsteinmann, 2020-04-02:
Perhaps this could be an option in the settings?

miczyg1, 2020-04-03:
@michaelsteinmann if you are fine with that then no problem.

silentcreek, 2020-08-19:
So, I installed coreboot v4.12.0.3 today and see the new configuration option to change the PCI device enumeration is now implemented. Thanks for that! I see that the three Intel NICs are now the first three PCI devices, followed by the two the mPCIe slots. I haven't tested yet if the PCI device number of the two wifi cards will change if I remove either of them, but even if not, it's already a welcome improvement if the ethernet devices will now have persistent names.

Btw. @miczyg1 I noticed that my new PCI layout doesn't exactly match your "desired" layout from above. You wrote that what was igb 0000:04:00.0 in my setup would become 0000:01:00.0 in the desired order. And that my mPCIe 1 device ath9k 0000:01:00.0 would become 0000:04:00.0, both of which is not the case. With the new order my previously named igb 0000:02:00.0 became 0000:01:00.0 and the ath9k 0000:01:00.0 device became 0000:05:00.0. To me, this doesn't really matter as long as the addressing is persistent. I'm just not sure if it really is what you intended.

miczyg1, 2020-08-20:
@silentcreek no that is not what I intended. @Sakoram you will have to investigate the change.

silentcreek, 2020-08-20:
@miczyg1 @Sakoram
Ok, just to be sure, I checked my ethernet ports again. The RJ45 port next to the serial port is what I use for my WAN connection. According to this [PCEngines board image](https://www.pcengines.ch/pic/apu2c4_1.jpg), it's labeled LAN1. Before the firmware upgrade this port was 0000:02:00.0. Now, it has the address 0000:01:00.0.

As for the mPCIe modules: I didn't take the time to open up the device and confirm which of my wireless modules populates which slot. But from the fact that v4.11.0.5 broke the mPCIe 2 slot  (#388 ), I can infer that my WLE200NX (ath9k) card is installed in mPCIE 2 (cause that was the one that wasn't detected anymore). This means we mixed up the attribution of my ath9k and ath10k devices before. So, here are the full before and after layouts again:

Before v4.12.0.3:

> ath9k 0000:01:00.0 (Compex WLE200NX)          ----- mPCIe 2 slot
> igb 0000:02:00.0 (LAN 1)                                    ----- next to serial port
> igb 0000:03:00.0 (LAN 2)                                    ----- middle port
> igb 0000:04:00.0 (LAN 3)                                    ----- next to USB port
> ath10k_pci 0000:05:00.0 (Compex WLE600VX)  ----- mPCIe 1 slot

With v4.12.0.3 and reverse PCI addressing order:

> igb 0000:01:00.0 (LAN 1)                                    ----- next to serial port
> igb 0000:02:00.0 (LAN 2)                                    ----- middle port
> igb 0000:03:00.0 (LAN 3)                                    ----- next to USB port
> ath10k_pci 0000:04:00.0 (Compex WLE600VX)  ----- mPCIe 1 slot
> ath9k 0000:05:00.0 (Compex WLE200NX)          ----- mPCIe 2 slot`

I can't tell which PCIe bridge addresses these would translate to since the schematics seem to be too complicated for me to find that piece of information in there, but I hope this list is detailed enough for you to know what I mean.

silentcreek, 2020-11-04:
@miczyg1 @Sakoram

I flashed v4.12.0.6 today and revisited the PCI addressing order. It looks good now, the order of the ethernet devices is reversed:
igb 0000:01:00.0 (LAN 3) ----- next to USB port
igb 0000:02:00.0 (LAN 2) ----- middle port
igb 0000:03:00.0 (LAN 1) ----- next to serial port
ath10k_pci 0000:04:00.0 (Compex WLE600VX) ----- mPCIe 1 slot
ath9k 0000:05:00.0 (Compex WLE200NX) ----- mPCIe 2 slot

Thanks! Now, I just need to wait for the coreboot feature to be able to make this configuration permanent across coreboot updates ;)

silentcreek, 2021-09-28:
Now that persistant runtime configuration has been integrated in coreboot, I think we can really close this issue. Thanks!

closed 2021-09-28 by silentcreek