pcengines/coreboot #394: Random boot loader hangup on apu1d4 with v4.11.0.5, no serial output in about 19 of 20 power cycles

floxo, 2020-04-04:
I tried to upgrade my apu1d4 firmware to the latest release (v4.11.0.5).
After successful flashing through flashrom on the installed debian, the board only boots randomly about every 20 power cycles. 
It seems to hang, with no output on the serial console for at least 30 minutes.
Once it boots, it does so without error and boots to OS. Rebooting from the OS works stable.

I tried to downgrade and get the same issue with versions v4.9.0.5 and v4.6.8.
If I downgrade to apu140405.rom ([https://pcengines.ch/file/apu140405.zip](https://pcengines.ch/file/apu140405.zip) on [PC Engines APU1D4 page](https://pcengines.ch/apu1d4.htm), the boot is successful every time.

Serial output directly after flashing + soft reboot:
```
SeaBIOS (version rel-1.12.1.3-0-g300e8b7)

Press F10 key now for boot menu

Select boot device:

1. Payload [setup]
2. Payload [memtest]
```
I then powered off and cleared CMOS.
Then, in this case, two power cycles later, next success 5 more cycles:
```
SeaBIOS (version rel-1.12.1.3-0-g300e8b7)

Press F10 key now for boot menu

Select boot device:

1. AHCI/0: KINGSTON RBU-SMS100S360GD ATA-8 Hard-Disk (57241 MiBytes)
2. Payload [setup]
3. Payload [memtest]
```
When unsuccessful, no output at all, blank console.

miczyg1, 2020-04-06:
@floxo thank you for your report. We will investigate the issue.

maxammann, 2020-04-14:
I had similar issues about 1-2 months ago. I tried it today again and had no problems.

Then I flashed the latest firmware v4.11.0.5. And I still don't have problems.

Even stress tests did not cause a problem.

miczyg1, 2020-04-26:
@floxo could you please test whether shorting SYS_RST pin to ground for a second helps?
SYS_RST pin is available near COM2 header (page 14 https://www.pcengines.ch/schema/apu1d.pdf).
1. Plug the power cable (baord probably don't POST at this point - but LEDs are on).
2. Short the pins for 1 second.
3. Observe the serial output after unshorting the pins. 2 of 3 LEDs should go off after a while.

This is a temporary workaround for now, while we are looking for a fix. Warm rebooting from the operating system should not cause this issue. If you encounter any problems, let me know.

maxammann, 2020-04-27:
I'll try, thanks! :)

maxammann, 2020-04-30:
@floxo can you still reproduce this?

Irihapeti, 2020-05-03:
I ran into this problem after upgrading my APU1D4 to v4.11.0.5 The unit simply would not boot after powering on again. It also refused to boot after I had to force poweroff because memtest locked up, and on another occasion after the OS on a USB stick misbehaved.

Fortunately, shorting the RST pins fixed it in all cases. The normal serial output returned immediately. (I didn't observe the LEDs.)

My assumption was that it had to do with improper shutdown. However, it's just done it again, this time after upgrading to v4.11.0.6 and shutting down properly. (The initial boot after upgrade was fine.) 

Again, shorting the RST pins fixed it, but having to take the lid off to get at the pins does get a little tiresome after a few times.

jakedaytona, 2020-05-25:
I've the same issue, with an apu1c. Shorting the RST pins fixes it. This is the same on 4.11.0.2 and now on 4.11.0.6 - didn't try the releases between and the unit was running the apu140908.zip rom before that (it was fine on that).
Soft reboots work, the issue is only when powering on from rest.
Proper or improper shutdowns don't make a difference to the issue.
LED's are: 
Whilst powered on and no serial output: D10 & D11 are the only two lit. 
When RST is pressed, all the other LEDs come on as normal: D5,D6,D7 & the mSATA one.

miczyg1, 2020-05-25:
We have found the culprit and the v4.12.0.1 release (which will be available soon) will contain a fix.

pimvanpelt, 2020-05-25:
Nice going @miczyg1 ! Thank you. Having recently updated a few apu1d and seeing several (but not all?) of them hang at boot, this fix is a big relief as my devices are in colos.

floxo, 2020-05-28:
Thanks miczyg1! I'll try v4.12.0.1 when it's ready and report.
And sorry for the delay. I'm available again though and happy to test.

Irihapeti, 2020-06-03:
I've installed v4.12.0.1 and started three times from poweroff. All were successful.

One of these was a forced poweroff. (I tried to power down using "grub halt", but it didn't work, and I didn't really expect it to.) Booting after this was as uneventful as the other two. As the APU1 is my LAN router, I'd prefer just to leave it running for now, but if any other tests are needed, I can take part.

Many thanks to all for fixing this.

jakedaytona, 2020-06-03:
Just tested 4.12.0.1 and my boot issue is resolved. Many thanks.

miczyg1, 2020-06-03:
Great. @floxo if it also works for you, feel free to close the issue

floxo, 2020-06-07:
Just flashed apu1_v4.12.0.1.rom and did 10 successful reboots, seems all right.
Thanks a lot @miczyg1!

closed 2020-06-07 by floxo