pcengines/coreboot #395: apu1d with v4.11.0.5: memtest hangs at start

maxammann, 2020-04-14:
I tried to run a memtest on my apu1d. The memtest hangs at the start though.

Maybe there is again a problem with SPD which was fixed in 1.6.2: https://github.com/pcengines/coreboot/blob/develop/CHANGELOG.md#added-20

Here is the screen that I see where it hangs:

[image]

miczyg1, 2020-04-19:
Issue confirmed.
@maxammann did previous releases cause this problem for you?

maxammann, 2020-04-20:
First time flashing a rom, so I don't know. I can try to flash another ROM if you need more information.

miczyg1, 2020-04-30:
@maxammann confirmed the issue on v4.11.0.5. Applied a fix in v4.11.0.6, you may try it out.

maxammann, 2020-04-30:
Thank you, I think I'll find some time in the evening today. 
Then I can also test whether my APU is still capable for production :)

maxammann, 2020-05-01:
memtest is working :) thank you!

maxammann, 2020-05-01:
Or maybe also not, the memtest freezes after the beep and 5 seconds.
[image]

maxammann, 2020-05-01:
I'm able to select different tests, which run for more than 5 seconds. Maybe my pcengine is damaged?

michaelsteinmann, 2020-05-01:
Does it boot an OS without crashing?
Can you downgrade the BIOS?
Check power supply.

maxammann, 2020-05-01:
Yeah the APU itself works just fine, I could also downgrade but that would not fix memtest.

miczyg1, 2020-05-01:
@maxammann apu1 in 3mdeb office worked for more than 10 minutes without hang, also 4GB variant. I have not interfered with the tests, just launched the memtest and left it working.

maxammann, 2020-05-01:
Alright thank you, in that case I will no longer use this APU for production :) RIP

miczyg1, 2020-05-02:
@maxammann it may be worth to dig into your particular problem on apu1. I will prepare a memtest with debug version to see what is going wrong on your hardware. There may still be hope :) Let me assist you a little bit longer.

miczyg1, 2020-05-19:
@maxammann could you please try using this binary and memtest? 
[apu1_v4.11.0.6_memtest_debug]9https://cloud.3mdeb.com/index.php/s/FrHKtoDZXF9DD5E/download)
It will print debug information on COM2. Do you have a USB-TTL converter? If yes, please connect it to the COM2 header and gather output from it when launching and running memtest. Example command for  minicom:
`minicom -D /dev/ttyUSB0 -b 115200 -o -C /tmp/logfile`

maxammann, 2020-05-19:
I'm already connected via the COM port on the back of my APU. Do you mean that port or is there another internal one?

miczyg1, 2020-05-20:
@maxammann  yes there is another one. Have a look at the page 14: https://www.pcengines.ch/schema/apu1d.pdf
There is COM2 header which uses TTL logic level (3.3V/5V).

maxammann, 2020-05-21:
Alright, thank you! I have the hardware needed here. I'll test this.

maxammann, 2020-05-24:
I created two test logs by running different tests:

[logfile2.txt](https://github.com/pcengines/coreboot/files/4673806/logfile2.txt)
[logfile.txt](https://github.com/pcengines/coreboot/files/4673807/logfile.txt)

The error seems to be happening inside the coretemp function: https://github.com/Distrotech/memtest86/blob/distrotech-memtest86-5.01/controller.c#L76

Not sure which line though as the lines seem to differ.

maxammann, 2020-06-16:
@miczyg1 any update on this? What do you think about the log?

miczyg1, 2020-06-16:
@maxammann I am sorry, I missed your comment. Will look into it.

Sakoram, 2020-07-02:
@maxammann I've looked at your results, but it's hard to deduce the reason for the problem from them, so I prepared another binary, this time with more debug prints in functions called last in the logs.
[apu1_v4.12.0.2_memtest_debug](https://cloud.3mdeb.com/index.php/s/2fab7aGRtCre6Wk/download)
It will also print debug information on COM2. Could I ask you to make another log?

maxammann, 2020-07-07:
@Sakoram here are two more logfiles with the new debugging rom. Thanks for investigating!

[logfile.txt](https://github.com/pcengines/coreboot/files/4887131/logfile.txt)
[logfile1.txt](https://github.com/pcengines/coreboot/files/4887133/logfile1.txt)

Sakoram, 2020-07-13:
@maxammann Thanks for logs. It looks like it isn't a problem with the coretemp function as we assumed after first logs. I suspect some problem with memory ranges being tested, because this time the last thing in logs is the start of writing test values to the block of memory. I prepared yet another binary, this time It logs a lot of memory addresses. Could I ask you to make logs again?
[apu1_v4.12.0.2_memtest_debug-2](https://cloud.3mdeb.com/index.php/s/QLpNYDX3AjHJpC6/download)

maxammann, 2020-07-19:
Letting the first test run. The APU freezes at 80% like described before:
[logfile.txt](https://github.com/pcengines/coreboot/files/4943783/logfile.txt)
[logfile4.txt](https://github.com/pcengines/coreboot/files/4943787/logfile4.txt)

Running test 3 by skipping the first two. The APU just crashes and reboots:
[logfile1.txt](https://github.com/pcengines/coreboot/files/4943785/logfile1.txt)

Running test 4. The APU just crashes and reboots:
[logfile3.txt](https://github.com/pcengines/coreboot/files/4943786/logfile3.txt)

Sakoram, 2020-07-21:
@maxammann Thanks again. I think that I found the cause of the problem. In all tests is an if statement which checks if a pointer to tested memory overflowed, but it didn't work, because of optimization done by the compiler. I corrected it and I believe that it can solve the problem. Could you test it and send logs of the first few minutes of memtest working? I'm still not sure why your APU in particular has this problem. It is a apu1d, right? or is it a 4GB version, apu1d4?
Additionally if you have Linux or FreeBSD installed, could you send me the output of this command? `dmesg | grep BIOS-e820`
[apu1_v4.12.0.2_memtest_fix](https://cloud.3mdeb.com/index.php/s/snopJiDbo7DmZHw/download)

maxammann, 2020-07-24:
```
[2.4.5-RELEASE][root@pfSense.localdomain]/root: dmidecode | grep BIOS
SMBIOS 2.8 present.
BIOS Information
                BIOS is upgradeable
        BIOS Revision: 4.12
```

Wow, seems like it is working! Etched on the board is "apu1d", but it has 4GB actually.

The memtest passed and ran for about 7hours:
[logfile.txt](https://github.com/pcengines/coreboot/files/4971246/logfile.txt)

I'm quite interrested in what exactly cased the overflow check to fail and maybe want to do a small writeup if the problem is interesting. Can you maybe give me some more insights on the fix?

Sakoram, 2020-07-24:
Funny thing is that we just found that this bug was already found and fixed by the coreboot community, but we don't always use the most recent memtest for stability sake. It seems like it's the time for an update.  ^^
https://review.coreboot.org/plugins/gitiles/memtest86plus/+/2182c5bbf65be2ab416588d9b35f2e84e05a8866
As you can read in this commit:
> Recent GCC optimizes away pointer comparisons used by overflow checks in test.c implementation.
This causes memtest to calculate wrong memory addresses which ends up in endless loops, freezes or crashes.
Disable strict pointer overflow check optimization via -fno-strict-overflow flag until code is amended to comply.

https://review.coreboot.org/plugins/gitiles/memtest86plus/+/refs/heads/master/test.c#198
Expression `pe + SPINSZ > pe` was the cause of the problem. ( where `pe` is the pointer to the end of a block of memory being modified and `SPINSZ` is a defined value which moves this pointer by 256MB. ) Compiler assumed that this expression is always TRUE, because it would be if overflow had never happened, but in this case the whole point is to test if overflow happens so it was an obvious bug. In your case because of lack of overflow detection a loop hadn't stopped when it should and the entire ram, including memory used by memtest itself, was overwritten by test values.

Sakoram, 2020-07-24:
>It seems like it's the time for an update. ^^

Hmm, actually... We updated memtest in v4.12.0.1 so it should work for you from this version. 
I did send you v4.12.0.2, but sadly it contained debug version of memtest based on older commit.

maxammann, 2020-07-24:
Ouch! Alright, thank you for your patience and for the great support!

I'll test out the latest firmware!

maxammann, 2020-08-06:
Thank you very much! The latest firmware is working.

That means I can put the APU back into production :)

closed 2020-08-06 by maxammann