pcengines/coreboot #397: agesa errors on apu1 when booting 4.11.0.5

310b8fb5, 2020-04-18:
After upgrading to 4.11.0.5 from 4.11.0.4 on an apu1 (including a power cycle), some errors are present in the console.

[image]

miczyg1, 2020-04-18:
@310b8fb5 thank you for the report. We will investigate it.

maxammann, 2020-05-01:
I can confirm this:

```
PC Engines apu1
coreboot build 20202604
BIOS version v4.11.0.6
ASSERTION ERROR: file 'src/vendorcode/amd/agesa/f14/Proc/Common/CommonReturns.c', line 174
ASSERTION ERROR: file 'src/vendorcode/amd/agesa/f14/Proc/CPU/cpuGeneralServices.c', line 817
ASSERTION ERROR: file 'src/vendorcode/amd/agesa/f14/Proc/Common/CommonReturns.c', line 174
ASSERTION ERROR: file 'src/vendorcode/amd/agesa/f14/Proc/CPU/cpuGeneralServices.c', line 817

```

miczyg1, 2020-06-05:
This should be fixed in v4.12.0.1. Any udpates?

310b8fb5, 2020-06-08:
@miczyg1 this seems to be fixed. Thanks!

closed 2020-06-08 by 310b8fb5