pcengines/coreboot #400: PCI resources - I/O port space issue

dugeem, 2020-04-30:
Up to and including v4.11.0.4 the following warning for failing to allocate I/O port window would appear in FreeBSD 11.x dmesg:
`pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0`
`pcib0: could not evaluate _ADR - AE_NOT_FOUND`
`pci0: <ACPI PCI bus> on pcib0`
`pci0: <base peripheral, IOMMU> at device 0.2 (no driver attached)`
`pcib1: <ACPI PCI-PCI bridge> irq 25 at device 2.2 on pci0`
**`pcib1: failed to allocate initial I/O port window: 0x1000-0x1fff`**
`pci1: <ACPI PCI bus> on pcib1`

In v4.11.0.5 the warning disappeared - presumably resulting from GNB IOAPIC change:
`pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0`
`pcib0: could not evaluate _ADR - AE_NOT_FOUND`
`pci0: <ACPI PCI bus> on pcib0`
`pci0: <base peripheral, IOMMU> at device 0.2 (no driver attached)`
`pcib1: <ACPI PCI-PCI bridge> irq 25 at device 2.2 on pci0`
`pci1: <ACPI PCI bus> on pcib1`

Unfortunately the same message is now back in v4.11.0.6:
`pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0`
`pcib0: could not evaluate _ADR - AE_NOT_FOUND`
`pci0: <ACPI PCI bus> on pcib0`
`pci0: <base peripheral, IOMMU> at device 0.2 (no driver attached)`
`pcib1: <ACPI PCI-PCI bridge> irq 25 at device 2.2 on pci0`
**`pcib1: failed to allocate initial I/O port window: 0x1000-0x1fff`**
`pci1: <ACPI PCI bus> on pcib1`

NB fortunately there is no impact as the FreeBSD igb driver is able to use memory access instead.

Is this a possible regression?

miczyg1, 2020-04-30:
@dugeem I think it is rather caused by the ACPI CPU declarations somehow. In v4.11.0.5 the CPUs were described correctly, however, not all BSD systems implement the ACPI specification requirements for CPUs yet, which caused the system to not detect CPUs on ACPI (see https://github.com/pcengines/coreboot/issues/389). As far as I know, since FreeBSD 12.x it is done correctly. I will check what is going on either way.

miczyg1, 2020-07-15:
@dugeem v4.11.0.5 seems to be the only working version for unknown reason. Around v4.11.0.5 and v4.11.0.6 we have been doing the ACPi transitions to be compliant to the newer ACPI specification and I suspect that the kernel gave up relying on ACPI at some point in v4.11.0.5 allocating resources by itself.

We have tested also the newer FreeBSD 12.1 and the error still persists across all firmware versions (even v4.11.0.5). To me it looks like a FreeBSD issue rather than BIOS issue, because coreboot and Linux doesn't show problems with allocating IO ports for this bridge.

Sakoram, 2020-07-15:
I checked almost every version of coreboot since v4.0.10 and this warning appears in all of them in FreeBSD 12.1. I don't think it's a problem with coreboot, but rather with FreeBSD implementation of ACPI. It's true that it doesn't appear in FreeBSD 11.x with coreboot 4.11.0.5 though. If I have to guess the reason I would suspect changes in ACPI tables which were done in this version, but honestly I'm not sure why it doesn't appear in this case.