pcengines/coreboot #401: UEFI build not working anymore since 4.11.0.5

bn82, 2020-05-02:
Hello,

FYI, it's not possible to build an UEFI build of 4.11.0.5 and 4.11.0.6 with tyanocore.
It fails with : 

Cloning into '/home/coreboot/coreboot/payloads/external/tianocore/tianocore'...
remote: Enumerating objects: 341677, done.
remote: Total 341677 (delta 0), reused 0 (delta 0), pack-reused 341677
Receiving objects: 100% (341677/341677), 285.86 MiB | 9.30 MiB/s, done.
Resolving deltas: 100% (246201/246201), done.
Checking out files: 100% (15552/15552), done.
    Fetching new commits from the Tianocore repo
Fetching origin
Fetching upstream
     is not a valid git reference
make[1]: *** [Makefile:63: update] Error 1
make: *** [payloads/external/Makefile.inc:177: payloads/external/tianocore/tianocore/Build/UEFIPAYLOAD.fd] Error 2

Regards, Bertrand

miczyg1, 2020-05-02:
@bn82 probably they broke something in upstream coreboot. Which flavor are you trying to build? UefiPayloadPkg or CorebootPayloadPkg? I will check the build. 

May I know what is your particular interest/use-case in UEFI payload? Maybe we can help?

bn82, 2020-05-02:
@miczyg1 I'm not sure about which flavor i use, i'm following your guide : 
https://github.com/pcengines/apu2-documentation/blob/master/docs/tianocore_build.md

I'm testing it just to play with intel clearlinux and containers, which only support UEFI boot, and this work well so far.

pietrushnic, 2020-05-02:
@miczyg1 maybe we should consider having another build target?
@lwcislo this is not first time when support for UEFI-aware OS is needed on apu2, maybe we can extend infrastructure to support that in the long run (e.g. end of Q2'20)?

miczyg1, 2020-07-06:
@bn82 we have updated the instructions. I hope it helps: https://github.com/pcengines/pce-fw-builder/blob/master/README.md#building-coreboot-with-tianocore-payload

bn82, 2020-07-10:
Tested and work :)

closed 2020-07-10 by miczyg1