pcengines/coreboot #412: Random reboots of pfSense 2.4.x on APU2C2

giox069, 2020-06-25:
I have an APU2C2 with latest pfSense installed (2.4.5-p1).

Today I upgraded coreboot to v4.12.0.1. pfSense started to reboot randomly every 2-10 minutes.

I reinstalled pfSense 2.4.5-p1 from scratch on the internal SSD and reimpoted the old configuration, but it still reboot.

I downgraded to coreboot v4.9.0.7, but pfSense still randomly reboot.

Only after downgrading to coreboot v4.6.10 pfSense seems to be stable again.

Any ideas? Unfortunately I'm unable to do other testing, because pfSense is in production environment at a customer site.

miczyg1, 2020-06-26:
@giox069 please try v4.9.0.1 and v4.9.0.2. I suspect that perhaps CPU boost makes it unstable. Expected result is that v4.9.0.1 should be stable. In order to disable boost in v4.9.0.3 and later, enter the setup menu and disable Core Performance Boost (runtime configuration of boost is not available yet in v4.9.0.2)

giox069, 2020-06-26:
Hi @miczyg1. As mentioned above, I cannot do a lot of testing, because the firewall is in user at a customer site and is 25Km from my office. Maybe this saturday they will have less internet usage and I could try to go there.

Just to avoid extra tests: if I will try directly with the latest 4.12.X and disable boost (can it be done in 4.12. ?), do you think it could work? But if you still need a test on 4.9, I will do it.

miczyg1, 2020-06-26:
@giox069 simply do what is easier for you. Since v4.9.0.3, all versions have the runtime configuration, so you may as well use v4.12.0.1 and just disable the option in setup menu.

giox069, 2020-06-27:
It works!
I installed v4.12.0.1, disabled "Core performance boost", and now pfSense is running without reboots for 3 hours.
You can close this issue.

closed 2020-06-27 by miczyg1