pcengines/coreboot #436: Wrong serial number in dmidecode on APU1 is back

pcenginetest, 2020-08-26:
coreboot image : apu1_v4.12.0.3.rom

i can't re open the Ticket #402 ?

miczyg1, 2020-08-27:
Yes, please. Could you additionally paste the bad output and expected serial number?

pcenginetest, 2020-08-28:
Handle 0x0002, DMI type 2, 14 bytes
Base Board Information
	Manufacturer: PC Engines
	Product Name: apu1
	Version: 1.0
	Serial Number: -64
	Asset Tag: Not Specified
	Features: None
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Unknown

zokl, 2020-12-03:
4.13.0.1 have the same issue.

miczyg1, 2020-12-03:
The patch has been merged that fixes it: https://review.coreboot.org/c/coreboot/+/42512 
We will re-investigate the issue, since the test still fails.

miczyg1, 2021-06-23:
Since v4.13.0.2 it is definitely fixed.
https://github.com/pcengines/coreboot/blob/release/CHANGELOG.md#v41302---2020-12-28

Also, our tests don't show errors and the serial number is calculated correctly from the MAC address. Closing

closed 2021-06-23 by miczyg1