pcengines/coreboot #439: Allow disabling TPM

Legogris, 2020-08-30:
Some of use might explicitly not want to use TPM due to trust tissues.

In absence of allowing configuring this in BIOS (which would be awesome!), is there any ways we can get pointers on how to build without TPM?

miczyg1, 2020-08-31:
@Legogris we may add an option to disable TPM.

If you are confident enough, you may build your own image. When in coreboot configuration menu, go to
`Security -> Trusted Platform Module` then change the `Trusted Platform Module (TPM 2.0)` to `Trusted Platform Module (disabled)`.

TPM is designed to provide trust. What kind of trust issues you have in mind?

miczyg1, 2020-08-31:
Also why don't you just unplug the TPM?

Legogris, 2020-08-31:
Thanks for the reply, much appreciated. 

Some other perspective on trust and openness when it comes to firmware:

* https://www.fsf.org/blogs/sysadmin/the-management-engine-an-attack-on-computer-users-freedom
* https://libreboot.org/amd-libre.html
* https://libreboot.org/faq.html

macpijan, 2021-07-06:
This is a discrete TPM (a physical device you can attach to the board or not). If there is no desire to use that, one can not attach the TPM device. Unless some specific use-case for this feature is presented, we are not in favor of adding enable/disable TPM runtime configuration option. Especially, that it would not disable the ability to communicate with the TPM completely, as explained by @miczyg1  above.
Please feel free to reopen if you have some more detailed description of what the goals of such an option would be. Thanks.

macpijan, 2021-07-06:
To clarify: we are talking about the discrete TPM module like this one: https://pcengines.ch/tpm4a.htm

The fTMP (feature of the CPU) was never supported for the APU boards if that was your concern.

closed 2021-07-06 by macpijan