pcengines/coreboot #453: AMD CCP Iinitialization - Nonextistent entropy from /dev/hwrng

4oo4, 2020-11-30:
I've been working on installing Debian Buster on my apu2e4, and am running into significant issues with getting proper entropy when booting into the OS.

I did a bunch of research and the issue seems to be with the usage of the AMD CCP for random number generation. I've made sure that the ccp kernel module is loaded, and this is what the kernel is trying to use as an rng.

```
# cat /sys/class/misc/hw_random/rng_available
ccp-1-rng

# cat /sys/class/misc/hw_random/rng_current
ccp-1-rng
```

However, the ccp does not appear to be giving me any randomness at all, and this is confirmed by rng-tools.

```
# hexdump -n 512 -C /dev/hwrng
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000200

# dd if=/dev/hwrng status=none | rngtest -c 100
rngtest 5
Copyright (c) 2004 by Henrique de Moraes Holschuh
This is free software; see the source for copying conditions.  There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rngtest: starting FIPS tests...
rngtest: bits received from input: 2000032
rngtest: FIPS 140-2 successes: 0
rngtest: FIPS 140-2 failures: 100
rngtest: FIPS 140-2(2001-10-10) Monobit: 100
rngtest: FIPS 140-2(2001-10-10) Poker: 100
rngtest: FIPS 140-2(2001-10-10) Runs: 100
rngtest: FIPS 140-2(2001-10-10) Long run: 100
rngtest: FIPS 140-2(2001-10-10) Continuous run: 100
rngtest: input channel speed: (min=21.798; avg=26.945; max=38.377)Mibits/s
rngtest: FIPS tests speed: (min=57.624; avg=92.756; max=122.266)Mibits/s
rngtest: Program run time: 93000 microseconds
```

However, according to the kernel, it still thinks it enables the ccp OK:
```
# dmesg | grep -i ccp
[   12.821952] ccp 0000:00:08.0: 5 command queues available
[   12.983496] ccp 0000:00:08.0: ccp enabled
```

Because of this poor/nonexistent entropy, it takes ~90-120 seconds for the kernel to initialize crng, and this causes many issues with services that depend on proper entry (cryptmounts, ssh, etc.)

```
# dmesg | grep -i crng
[    0.159387] random: get_random_u64 called from __kmem_cache_create+0x3e/0x520 with crng_init=0
[  103.325998] random: crng init done
```
After doing research about how CCP works in the Linux kernel, I keep coming back to potential issues with how coreboot is initializing the CCP part of the CPU. I did notice this commit that seems extremely relevant to the issue I'm having:

https://github.com/coreboot/coreboot/commit/6b303d54aa4e0e19a9a773ff6f4cf77ac4528d6a#diff-7da6dd7c9f2fc35c36a12f29cf87fbe1179281ad994fe45118751563161d2fa7

If I'm following PCEngine's GitLab CI builds correctly, is this something that will be included in the v4.13.x builds I see? And if I'm understanding what I'm seeing in mainline coreboot, AMD CCP/PSP support seems to be a work in progress?

https://gitlab.com/pcengines/coreboot/-/commit/90d697891f1f67f90475acd15585701166e20fa6

From the sounds of it others have used a hardware rng, I'm trying to avoid going down that route and use the rng that's already provided by the CPU (unless there's a good security reason not to), assuming that coreboot supports this.
https://kula.tproa.net/lnt/computers/random/randomness-on-an-apu2.html

I've also looked at userspace entropy generators, those don't seem to help, since I'm wondering if the presence of a non-functioning `/dev/hwrng` throws them off.

miczyg1, 2020-11-30:
Hi @4oo4 

> After doing research about how CCP works in the Linux kernel, I keep coming back to potential issues with how coreboot is initializing the CCP part of the CPU. I did notice this commit that seems extremely relevant to the issue I'm having:
> 
> [coreboot@6b303d5#diff-7da6dd7c9f2fc35c36a12f29cf87fbe1179281ad994fe45118751563161d2fa7](https://github.com/coreboot/coreboot/commit/6b303d54aa4e0e19a9a773ff6f4cf77ac4528d6a#diff-7da6dd7c9f2fc35c36a12f29cf87fbe1179281ad994fe45118751563161d2fa7)
> 

Unfortunately, this is a miss. This commit refers to a newer family of AMD CPUs and is not relevant to apu2.

> If I'm following PCEngine's GitLab CI builds correctly, is this something that will be included in the v4.13.x builds I see? And if I'm understanding what I'm seeing in mainline coreboot, AMD CCP/PSP support seems to be a work in progress?
> 
> https://gitlab.com/pcengines/coreboot/-/commit/90d697891f1f67f90475acd15585701166e20fa6
> 
> From the sounds of it others have used a separate hardware rng instead, I'm trying to avoid going down that route and use the rng that's already provided by the CPU (unless there's a good security reason not to), assuming that coreboot supports this.
> https://kula.tproa.net/lnt/computers/random/randomness-on-an-apu2.html
> 
> I've also looked at userspace entropy generators, those don't seem to help, since I'm wondering if the presence of a non-functioning `/dev/hwrng` throws them off.
> 
> Cheers

The current status (to be honest it didn't change for a few years) is that the CCP RNG is not working and probably will never work. There are a few reasons for that:
1. AMD AGESA code probably does not initialize PSP correctly (although release note say it does send correct mailbox messages: https://github.com/coreboot/blobs/blob/master/pi/amd/00730F01/FT3b/AGESA_Release_Notes.txt#L94)
2. coreboot doesn't perform any PSP initialization, the resource management is also limited. There are probably many more mailbox commands to be sent. PSP fTPM also isn't supported by coreboot (which is probably the source of RNG). 
3. Some time ago many PSP firmware binaries have been removed from the build: https://github.com/coreboot/coreboot/commit/bfb5c807e720761f4457d5106bb919f2aacb5535#diff-a64d01e465c7a6d0d96f57e132a45b9a0d5e020f0a5f59301d946c97144c412c
   fTPM and PSP Secure OS is no longer present in the firmware.

In light of the above reasons, the CCP RNG has no chance to work.

Have you tried to use TPM2 as the source of entropy? Maybe it could solve your problems.

4oo4, 2020-11-30:
@miczyg1 Thanks, that's really good to know. Usually TPM is my preference for entropy, but I don't think the apu2 board I have has a TPM, unless I'm missing something? I don't have a TPM option in coreboot, nor do I have a TPM device in the OS.

I did see something about a TPM1A module that I can still track down from PCEngines resellers, so maybe that's the route I need to go?

miczyg1, 2020-11-30:
@4oo4 TPM1A, yes, that's it. You can simply plug it into the mainboard near mSATA slot, 20pin header. If you don't see TPM in the OS, then probably you don't have one. TPM2 is enabled by default in PC Engines firmware, so you don't have to worry about the settings in coreboot. `dmesg` should say something that it enables TPM bypass because of lack of the TPM. I would recommend going TPM way, because PSP is rather a lost option. With TPM you may get something that works, just like that.

4oo4, 2020-11-30:
@miczyg1 Awesome, I actually had just bought one to test with, so once that arrives I should be good. Thanks!

closed 2020-11-30 by 4oo4

Rombobeorn, 2021-03-05:
For others reading this: On SystemD-based systems, one way to ensure that random numbers are taken from the TPM1A instead of the CCP is to write a file named /etc/systemd/system/rngd.service.d/tpm-rng.conf with these contents:

`[Service]
ExecStartPre=sh -c 'echo tpm-rng-0 >/sys/class/misc/hw_random/rng_current'`

That selects the TPM for /dev/hwrng right before RNGD is started.

If the cryptographic functions of the CCP are equally nonfunctional, then it may be better to blacklist the kernel module as suggested in the linked blog post.

Rombobeorn, 2021-03-05:
In the previous comment there was supposed to be a linebreak immediately after "[Service]". Thank you so very much, Github, for messing that up!

4oo4, 2021-03-06:
@Rombobeorn FYI, you can edit your GitHub comments. There are multiple ways to set values in `sysfs` at boot, but given that it's very easy to show that CCP provides absolutely no entropy whatsoever due to the issues that @miczyg1 pointed out, it's reckless to try to use CCP as your source of entropy by setting that. So like you said, blacklisting is the only way to go and you need to find another entropy source.

miczyg1, 2021-03-08:
```
[Service]
ExecStartPre=sh -c 'echo tpm-rng-0 >/sys/class/misc/hw_random/rng_current'
```

Use 3-backtick blocks when pasting code. It assures the formating stays intact.

miczyg1, 2021-03-08:
We will add these hints to the documentation